doe300/VC4CL #1: Test Compile Error

nomaddo, 2017-11-09:
I have troubled with the following errors when compiling test cases.
What I did is:

- cross-compile LLVM-VSPHIR and copy them to rasberrypi
- build VC4C
- build VC4CL 
- then, build VC4CL/test

Any suggestion?

```bash
[  6%] Building CXX object CMakeFiles/TestVC4CL.dir/test.o                                                                                                                                                                                                                                                                                                                           
/usr/bin/c++   -DTEST_OUTPUT_CONSOLE=1 -I/home/pi/work/VC4CL/test/.. -I/home/pi/work/VC4CL/test/../include -I/home/pi/work/VC4CL/test/lib/cpptest-lite/include   -o CMakeFiles/TestVC4CL.dir/test.o -c /home/pi/work/VC4CL/test/test.cpp                                                                                                                                             
In file included from /home/pi/work/VC4CL/test/../src/extensions.h:10:0,                                                                                                                                                                                                                                                                                                             
                 from /home/pi/work/VC4CL/test/TestExtension.h:11,                                                                                                                                                                                                                                                                                                                   
                 from /home/pi/work/VC4CL/test/test.cpp:20:                                                                                                                                                                                                                                                                                                                          
/home/pi/work/VC4CL/test/../src/common.h:31:106: warning: ignoring attributes on template argument ‘cl_int {aka int}’ [-Wignored-attributes]                                                                                                                                                                                                                                         
     CHECK_RETURN typename std::enable_if<std::is_arithmetic<T>::value | std::is_pointer<T>::value, cl_int>::type                                                                                                                                                                                                                                                                    
                                                                                                          ^                                                                                                                                                                                                                                                                          
In file included from /home/pi/work/VC4CL/test/../src/types.h:12:0,                                                                                                                                                                                                                                                                                                                  
                 from /home/pi/work/VC4CL/test/../src/common.h:15,                                                                                                                                                                                                                                                                                                                   
                 from /home/pi/work/VC4CL/test/../src/extensions.h:10,                                                                                                                                                                                                                                                                                                               
                 from /home/pi/work/VC4CL/test/TestExtension.h:11,                                                                                                                                                                                                                                                                                                                   
                 from /home/pi/work/VC4CL/test/test.cpp:20:                                                                                                                                                                                                                                                                                                                          
/home/pi/work/VC4CL/test/test.cpp: In function ‘int main(int, char**)’:                                                                                                                                                                                                                                                                                                              
/home/pi/work/VC4CL/test/test.cpp:45:34: error: ‘struct _cl_context’ has no member named ‘dispatch’                                                                                                                                                                                                                                                                                  
     assert(offsetof(_cl_context, dispatch) == 0);                                                                                                                                                                                                                                                                                                                                   
                                  ^                                                                                                                                                                                                                                                                                                                                                  
CMakeFiles/TestVC4CL.dir/build.make:374: recipe for target 'CMakeFiles/TestVC4CL.dir/test.o' failed                                                                                                                                                                                                                                                                                  
```

doe300, 2017-11-09:
You probably have compiled VC4CL without ICD support (Option `BUILD_ICD` in CMakeLists.txt).

If so, then this is a bug, where I forgot to check whether the ICD support is enabled before running this assertion. I will push a fix in a few moments.

doe300, 2017-11-09:
Should now be fixed

nomaddo, 2017-11-10:
Thanks! I have checked that it is fixed.
I found more problems:

- When executing `cmake`,  got an error
- `VC4CL_LIBRARY_VERSION` is undefine
- `HAS_COMPILER` is undefined

My cmake is `cmake 3.7.2`. VC4C is build without `BUILD_ICD` support.
And, after adhoc-fix, Test cases seem fail.
I will make an issue about that...

```
pi@nomaddo:~/work/VC4CL/test$ cmake .                                                                                                                                                                                                                                                                                                                                                
-- The C compiler identification is GNU 6.3.0                                                                                                                                                                                                                                                                                                                                        
-- The CXX compiler identification is GNU 6.3.0                                                                                                                                                                                                                                                                                                                                      
-- Check for working C compiler: /usr/bin/cc                                                                                                                                                                                                                                                                                                                                         
-- Check for working C compiler: /usr/bin/cc -- works                                                                                                                                                                                                                                                                                                                                
-- Detecting C compiler ABI info                                                                                                                                                                                                                                                                                                                                                     
-- Detecting C compiler ABI info - done                                                                                                                                                                                                                                                                                                                                              
-- Detecting C compile features                                                                                                                                                                                                                                                                                                                                                      
-- Detecting C compile features - done                                                                                                                                                                                                                                                                                                                                               
-- Check for working CXX compiler: /usr/bin/c++                                                                                                                                                                                                                                                                                                                                      
-- Check for working CXX compiler: /usr/bin/c++ -- works                                                                                                                                                                                                                                                                                                                             
-- Detecting CXX compiler ABI info                                                                                                                                                                                                                                                                                                                                                   
-- Detecting CXX compiler ABI info - done                                                                                                                                                                                                                                                                                                                                            
-- Detecting CXX compile features                                                                                                                                                                                                                                                                                      [... cut ...]                                                                                                 
Makefile:83: recipe for target 'all' failed                                                                                                                                                                                                                                                                                                                                          
make: *** [all] Error 2  
```

doe300, 2017-11-10:
See #2 for fix

nomaddo, 2017-11-13:
This is solved, Thank you so much!

closed 2017-11-13 by nomaddo