doe300/VC4CL #2: Test case failed

nomaddo, 2017-11-10:
I am not sure what happned, but test cases seem failed.
Thanks to #1, I have compiled test cases by adhoc-way.

- Comment out line 18 to avoid `Unknown CMake command "ExternalProject_Get_Property". `
  And I downloaded libcpptest-lite by-mand, and link `TestVC4CL` by man-hand
- Add `#define HAS_COMPILER 1`
- Modify `VC4CL_VERSION` as constant string to avoid compilation-error

Can you teach me what happened? 
Or, Can you tell me the way to construct a right env (my env may be wrong...)

```
pi@nomaddo:~/work/VC4CL/test$ sudo ./TestVC4CL
Running suite 'TestSystem' with 1 tests...
Suite 'TestSystem' finished, 1/1 successful (100%) in 48 microseconds (0.048 ms).
Running suite 'TestPlatform' with 2 tests...
Test 'TestPlatform::testGetPlatformInfo()' failed!
	Suite: TestPlatform
	File: TestPlatform.cpp
	Line: 56
	Failure: Assertion 'platform_config::VERSION.compare(buffer) == 0' failed
Test-method 'TestPlatform::testGetPlatformInfo()' finished with errors!
Suite 'TestPlatform' finished, 1/2 successful (50%) in 401 microseconds (0.401 ms).
Running suite 'TestDevice' with 5 tests...
Suite 'TestDevice' finished, 5/5 successful (100%) in 1665 microseconds (1.665 ms).
Running suite 'TestContext' with 5 tests...
Suite 'TestContext' finished, 5/5 successful (100%) in 198 microseconds (0.198 ms).
Running suite 'TestCommandQueue' with 4 tests...
Suite 'TestCommandQueue' finished, 4/4 successful (100%) in 568 microseconds (0.568 ms).
Running suite 'TestBuffer' with 17 tests...
Suite 'TestBuffer' finished, 17/17 successful (100%) in 1536 microseconds (1.536 ms).
Running suite '' with 0 tests...
Suite '' finished, 0/0 successful (0%) in 0 microseconds (0 ms).
Running suite 'TestProgram' with 11 tests...
Test 'TestProgram::testCompileProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 109
	Failure: Got -3, expected 0
Test-method 'TestProgram::testCompileProgram()' finished with errors!
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 118
	Failure: Got -59, expected 0
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 119
	Failure: Got 0, expected 0x138a06c
Test-method 'TestProgram::testLinkProgram()' finished with errors!
Test 'TestProgram::testUnloadPlatformCompiler()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 129
	Failure: Got 5, expected 13
Test-method 'TestProgram::testUnloadPlatformCompiler()' finished with errors!
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 180
	Failure: Got 0, expected -30
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 199
	Failure: Got -45, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 200
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 201
	Failure: Got 0, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 204
	Failure: Got -45, expected 0
Test-method 'TestProgram::testGetProgramInfo()' finished with errors!
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 139
	Failure: Got -1, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 152
	Failure: Got 0, expected 4
Test-method 'TestProgram::testGetProgramBuildInfo()' finished with errors!
Suite 'TestProgram' finished, 6/11 successful (54.54%) in 5657030 microseconds (5657.03 ms).
Running suite 'TestKernel' with 13 tests...
Test 'TestKernel::testCreateKernel()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 62
	Failure: Got -45, expected 0
Test 'TestKernel::testCreateKernel()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 63
	Failure: Assertion 'kernel != NULL' failed
Test-method 'TestKernel::testCreateKernel()' finished with errors!
Test 'TestKernel::testCreateKernelsInProgram()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 71
	Failure: Got -45, expected 0
Test 'TestKernel::testCreateKernelsInProgram()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 72
	Failure: Got 0, expected 1
Segmentation fault
```

doe300, 2017-11-10:
The cpptest-lite (like any other dependency) should be downloaded automatically via ExternalProject_Add. The fact that your CMake doesn't know the ExternalProject_Get_Property confuses me. 
In ticket #1 you stated, that you use CMake 3.7.2 which should support this feature (since it was introduces before 3.1). Same with `VC4CL_VERSION` from #1, which is definitively set by CMake.

The `#define HAS_COMPILER` is added via CMake, when the [VC4C](https://github.com/doe300/VC4C) compiler is found. By default, it searches in the relative paths `./lib/vc4c/` and `../VC4C/`. If your VC4C project resides in another relative path, you might need to modify the lines 64 and 70 of the root `CmakeLists.txt` file in the VC4CL project or manually set the CMake variables `VC4C_HEADER_PATH` and `VC4CC_LIBRARY` to point to the correct files (the `include/VC4C.h` header-file and the `libVC4CC.so` library-file). 
A message starting with `VC4C compiler header found at:` should indicate, if the compiler was found correctly, otherwise, CMake will print `No VC4C compiler header found!` instead.

The test-cases fail, because ou specified `HAS_COMPILER`, but the compiler could not be found.

Do you have the CMake configuration `BUILD_TESTING` set to `ON` or `OFF`?

doe300, 2017-11-10:
I just saw, that you ran cmake in the `./test` directory. You need to use the root `CMakeLists.txt`-file in the root directory!

To build the test-program `TestVC4CL`:

1. set the option `BUILD_TESTING` in the root `CMakeLists.txt`-file to `ON`
2. set the correct paths for the VC4C compiler (if necessary)
3. run cmake in the root-directory (or in an extra build-directory, but using the root `CMakeLists.txt`!)
4. now you can `make TestVC4CL` which will download and build `cpptest-lite`, build `VC4CL` and the `TestVC4CL` test-program.

nomaddo, 2017-11-13:
Yes, `BUILD_TESTING` was `FALSE` and I tried to compile test cases in `VC4CL/test`, not the root directory.
In the root directory, I have succeed in compilation of `TestVC4CL`. 
I closed #1. Thank you so much!

After that, only one test case seemed to fail.
Is it a problem?

```
Running suite 'TestSystem' with 1 tests...                                                                                                                                                                                                                                                                                                                                           
Suite 'TestSystem' finished, 1/1 successful (100%) in 29 microseconds (0.029 ms).                                                                                                                                                                                                                                                                                                    
Running suite 'TestPlatform' with 2 tests...                                                                                                                                                                                                                                                                                                                                         
Test 'TestPlatform::testGetPlatformInfo()' failed!                                                                                                                                                                                                                                                                                                                                   
        Suite: TestPlatform                                                                                                                                                                                                                                                                                                                                                          
        File: TestPlatform.cpp                                                                                                                                                                                                                                                                                                                                                       
        Line: 56                                                                                                                                                                                                                                                                                                                                                                     
        Failure: Assertion 'platform_config::VERSION.compare(buffer) == 0' failed                                                                                                                                                                                                                                                                                                    
Test-method 'TestPlatform::testGetPlatformInfo()' finished with errors!                                                                                                                                                                                                                                                                                                              
Suite 'TestPlatform' finished, 1/2 successful (50%) in 228 microseconds (0.228 ms).                                                                                                                                                                                                                                                                                                  
Running suite 'TestDevice' with 5 tests...                                                                                                                                                                                                                                                                                                                                           
Suite 'TestDevice' finished, 5/5 successful (100%) in 5666 microseconds (5.666 ms).                                                                                                                                                                                                                                                                                                  
Running suite 'TestContext' with 5 tests...                                                                                                                                                                                                                                                                                                                                          
Suite 'TestContext' finished, 5/5 successful (100%) in 110 microseconds (0.11 ms).                                                                                                                                                                                                                                                                                     [... cut ...]                                                                                             
Test-method 'TestKernel::testSetKernelArg()' finished with errors!                                                                                                                                                                                                                                                                                                                   
Segmentation fault           
```

nomaddo, 2017-11-13:
Sorry for post repeatly, after updating latest source code by `git pull`, the status has changed.
After all, test cases has failed like following:

```
pi@nomaddo:~/work/VC4CL$ cmake . -DBUILD_TESTING=True
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
-- Found OpenCL: /usr/local/lib/libOpenCL.so (found version "2.0")
-- Found headers for OpenCL up to version 2.0 located in /usr/include
-- VC4C library found: /home/pi/work/VC4C/build/libVC4CC.so
-- VC4C compiler header found at: /home/pi/work/VC4C/include/VC4C.h
-- Building with ICD support
-- Checking for one of the modules 'ocl-icd>=1.3'
-- Found Khronos ICD Loader in version 2.2.11 in /usr/local/lib
-- Enabling register-poking to run kernels
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/work/VC4CL
pi@nomaddo:~/work/VC4CL$ make -j3
Scanning dependencies of target cpptest-lite-project-build
[  2%] Performing update step for 'cpptest-lite-project'
[ 47%] Built target VC4CL
[ 52%] Built target v3d_profile
[ 56%] Built target v3d_info
From https://github.com/doe300/cpptest-lite
 * branch            HEAD       -> FETCH_HEAD
Already up-to-date.
[ 58%] Performing configure step for 'cpptest-lite-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/work/VC4CL/build/cpptest-lite/src/cpptest-lite-project-build
[ 60%] Performing build step for 'cpptest-lite-project'
[100%] Built target cpptest-lite
[ 67%] Built target cpptest-lite-project-build
[100%] Built target TestVC4CL
pi@nomaddo:~/work/VC4CL$ sudo ./test/TestVC4CL
Running suite 'TestSystem' with 1 tests...
Suite 'TestSystem' finished, 1/1 successful (100%) in 28 microseconds (0.028 ms).
Running suite 'TestPlatform' with 2 tests...
Test 'TestPlatform::testGetPlatformInfo()' failed!
	Suite: TestPlatform
	File: TestPlatform.cpp
	Line: 56
	Failure: Assertion 'platform_config::VERSION.compare(buffer) == 0' failed
Test-method 'TestPlatform::testGetPlatformInfo()' finished with errors!
Suite 'TestPlatform' finished, 1/2 successful (50%) in 299 microseconds (0.299 ms).
Running suite 'TestDevice' with 5 tests...
Suite 'TestDevice' finished, 5/5 successful (100%) in 945 microseconds (0.945 ms).
Running suite 'TestContext' with 5 tests...
Suite 'TestContext' finished, 5/5 successful (100%) in 112 microseconds (0.112 ms).
Running suite 'TestCommandQueue' with 4 tests...
Suite 'TestCommandQueue' finished, 4/4 successful (100%) in 390 microseconds (0.39 ms).
Running suite 'TestBuffer' with 17 tests...
Suite 'TestBuffer' finished, 17/17 successful (100%) in 892 microseconds (0.892 ms).
Running suite '' with 0 tests...
Suite '' finished, 0/0 successful (0%) in 0 microseconds (0 ms).
Running suite 'TestProgram' with 11 tests...
Test 'TestProgram::testCompileProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 109
	Failure: Got -3, expected 0
Test-method 'TestProgram::testCompileProgram()' finished with errors!
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 118
	Failure: Got -59, expected 0
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 119
	Failure: Got 0, expected 0x20ba754
Test-method 'TestProgram::testLinkProgram()' finished with errors!
Test 'TestProgram::testUnloadPlatformCompiler()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 129
	Failure: Got 5, expected 13
Test-method 'TestProgram::testUnloadPlatformCompiler()' finished with errors!
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 199
	Failure: Got -45, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 200
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 201
	Failure: Got 0, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 204
	Failure: Got -45, expected 0
Test-method 'TestProgram::testGetProgramInfo()' finished with errors!
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 139
	Failure: Got -1, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 152
	Failure: Got 0, expected 4
Test-method 'TestProgram::testGetProgramBuildInfo()' finished with errors!
Suite 'TestProgram' finished, 6/11 successful (54.54%) in 79490 microseconds (79.49 ms).
Running suite 'TestKernel' with 13 tests...
Test 'TestKernel::testCreateKernel()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 62
	Failure: Got -45, expected 0
Test 'TestKernel::testCreateKernel()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 63
	Failure: Assertion 'kernel != NULL' failed
Test-method 'TestKernel::testCreateKernel()' finished with errors!
Test 'TestKernel::testCreateKernelsInProgram()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 71
	Failure: Got -45, expected 0
Test 'TestKernel::testCreateKernelsInProgram()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 72
	Failure: Got 0, expected 1
Segmentation fault
```

doe300, 2017-11-13:
The error in TestProgram (-3 = CL_COMPILER_NOT_AVAILABLE) indicates, that the configuration for the use of the compiler was not correct, but your CMake log shows, that the header and library have been found.

Also, the error with mis-matching platform-versions looks like the macro you specified manually (see #1) doesn't match the automatically generated one.

One thing you could try, is cleaning and re-building the whole VC4CL project by deleting `CMakeCache.txt`, `CMakeFiles/` and `build/` and then re-configuring with `cmake . -DBUILD_TESTING=True`. This should at least fix the errors with the mis-matching platform-version and probably the ones with the compiler not found too.

nomaddo, 2017-11-13:
I tried to compile `VC4CL` as fresh (`rm -rf VC4CL` and `git pull` again).
But tests failed again...

```
Running suite 'TestSystem' with 1 tests...
Suite 'TestSystem' finished, 1/1 successful (100%) in 17 microseconds (0.017 ms).
Running suite 'TestPlatform' with 2 tests...
Suite 'TestPlatform' finished, 2/2 successful (100%) in 229 microseconds (0.229 ms).
Running suite 'TestDevice' with 5 tests...
Suite 'TestDevice' finished, 5/5 successful (100%) in 910 microseconds (0.91 ms).
Running suite 'TestContext' with 5 tests...
Suite 'TestContext' finished, 5/5 successful (100%) in 88 microseconds (0.088 ms).
Running suite 'TestCommandQueue' with 4 tests...
Suite 'TestCommandQueue' finished, 4/4 successful (100%) in 326 microseconds (0.326 ms).
Running suite 'TestBuffer' with 17 tests...
Suite 'TestBuffer' finished, 17/17 successful (100%) in 685 microseconds (0.685 ms).
Running suite '' with 0 tests...
Suite '' finished, 0/0 successful (0%) in 0 microseconds (0 ms).
Running suite 'TestProgram' with 11 tests...
Test 'TestProgram::testCompileProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 109
	Failure: Got -3, expected 0
Test-method 'TestProgram::testCompileProgram()' finished with errors!
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 118
	Failure: Got -59, expected 0
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 119
	Failure: Got 0, expected 0x52c4d4
Test-method 'TestProgram::testLinkProgram()' finished with errors!
Test 'TestProgram::testUnloadPlatformCompiler()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 129
	Failure: Got 5, expected 13
Test-method 'TestProgram::testUnloadPlatformCompiler()' finished with errors!
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 199
	Failure: Got -45, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 200
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 201
	Failure: Got 0, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 204
	Failure: Got -45, expected 0
Test-method 'TestProgram::testGetProgramInfo()' finished with errors!
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 139
	Failure: Got -1, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
[... 46 lines cut ...]
                                                                                                                                                                                          
Aborted           
```

With `sudo`, it failed.

```
pi@nomaddo:~/work/VC4C$ sudo build/VC4C -o ../test.s ../test.cl                                                                                                                           
Compiling '../test.cl' into '../test.s' with options '' ...                                                                                                                               
[I] Mon Nov 13 19:43:56 2017: Compiling OpenCL to LLVM-IR with :/opt/VC4C/bin/clang -cc1 -triple spir-unknown-unknown  -I .. -O3 -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-c\
onstant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /home/pi/work/VC4C/include/VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /dev/s\
tdout ../test.cl                                                                                                                                                                          
[I] Mon Nov 13 19:43:59 2017: Converting LLVM-IR to SPIR-V with :/opt/VC4C/bin/llvm-spirv -o /dev/stdout /dev/stdin                                                                       
[E] Mon Nov 13 19:43:59 2017: Errors in precompilation:                                                                                                                                   
[E] Mon Nov 13 19:43:59 2017: error: Invalid bitcode signature                                                                                                                            
                                                                                                                                                                                          
[W] Mon Nov 13 19:43:59 2017: LLVM-IR to SPIR-V failed, trying to compile with the LLVM-IR front-end...                                                                                   
[I] Mon Nov 13 19:43:59 2017: Compiling OpenCL to LLVM-IR with :/opt/VC4C/bin/clang -cc1 -triple spir-unknown-unknown  -I .. -O3 -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-c\
onstant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /home/pi/work/VC4C/include/VC4CLStdLib.h.pch -x cl -S -emit-llvm -o /dev/stdo\
ut ../test.cl                                                                                                                                                                             
[I] Mon Nov 13 19:44:01 2017: Compilation complete!                                                                                                                                       
terminate called after throwing an instance of 'vc4c::CompilationError'                                                                                                                   
  what():  General: Unrecognized source code type!                                                                                                                                        
Aborted       
```

doe300, 2017-11-13:
Yes, the test-cases in VC4CL TestProgram, TestKernel and TestExecutions fail if VC4C cannot compile the OpenCL kernel code.

The error compiling `test.cl` with/without sudo looks like the issue listed [here](https://github.com/doe300/VC4C/blob/master/Readme.md#known-issues): Somehow sometimes the pre-compilation in VC4C deletes `/dev/stdout` when failing to compile and then any successive compilation (maybe even any successive access to `stdout` from any process) fails.

My current workaround is to restart the Raspbery Pi, which will re-create the `dev/stdout` symbolic link.

I also created a "fix", which lowers the rights for the pre-compilation process (from su to user `pi`), who should be unable to delete `dev/stdout`. Since I cannot reliably reproduce the issue, I don't know if the fix works. To enable it, configure VC4C with `PRECOMPILER_DROP_RIGHTS` enabled. I've created an [issue](https://github.com/doe300/VC4C/issues/3) to track this particular problem.

nomaddo, 2017-11-13:
To me, `/dev/stdout` exists correctly (symlink to `/proc/self/fd/1`).
But as you said, somehow, `VC4C` can compile /tmp/hoge.cl now.....

Why don't you change `/dev/stdout` to a temporary file?
Temporary files can be easily created by `mktemp`, which create a tempoprary file and return its name without considering file name conflicts.
In general, using `/dev/stdout` is not recommended.

doe300, 2017-11-13:
I used temporary files before, but removed them, so I don't need to access the "hard drive" (read: SD card), e.g. if `/tmp/` is not located in RAM.

But if I cannot find a fix for using `stdout` and `stdin`, then I will have to revert to temporary files.

nomaddo, 2017-11-14:
It seems that you modified to use temporary files to compile!
Now, on my env, `TestVC4CL` still go wrong.
`TestProgram::testCompileProgram()` returns correctly (before that, it return -3), so the situation may be a little bit successful (?)

Do you have any idea why it failed?
Now, I write vc4 assembly directly. I miss the day I can use VC4CL......

```
Running suite 'TestSystem' with 1 tests...
Suite 'TestSystem' finished, 1/1 successful (100%) in 29 microseconds (0.029 ms).
Running suite 'TestPlatform' with 2 tests...
Suite 'TestPlatform' finished, 2/2 successful (100%) in 223 microseconds (0.223 ms).
Running suite 'TestDevice' with 5 tests...
Suite 'TestDevice' finished, 5/5 successful (100%) in 1104 microseconds (1.104 ms).
Running suite 'TestContext' with 5 tests...
Suite 'TestContext' finished, 5/5 successful (100%) in 148 microseconds (0.148 ms).
Running suite 'TestCommandQueue' with 4 tests...
Suite 'TestCommandQueue' finished, 4/4 successful (100%) in 470 microseconds (0.47 ms).
Running suite 'TestBuffer' with 17 tests...
Suite 'TestBuffer' finished, 17/17 successful (100%) in 1148 microseconds (1.148 ms).
Running suite '' with 0 tests...
Suite '' finished, 0/0 successful (0%) in 0 microseconds (0 ms).
Running suite 'TestProgram' with 11 tests...
Suite 'TestProgram' finished, 11/11 successful (100%) in 2904132 microseconds (2904.13 ms).
Running suite 'TestKernel' with 13 tests...
Test 'TestKernel::testSetKernelArg()' failed!
	Suite: TestKernel
	File: TestKernel.cpp
	Line: 82
	Failure: Got 0, expected -51
Test-method 'TestKernel::testSetKernelArg()' finished with errors!
Suite 'TestKernel' finished, 12/13 successful (92.3%) in 796 microseconds (0.796 ms).
Running suite 'TestEvent' with 12 tests...
Suite 'TestEvent' finished, 12/12 successful (100%) in 153 microseconds (0.153 ms).
Running suite 'TestExtension' with 5 tests...
Suite 'TestExtension' finished, 0/5 successful (0%) in 0 microseconds (0 ms).
Running suite 'TestExecutions' with 2 tests...
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 60
	Failure: Got -50, expected 0
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 65
	Failure: Got -52, expected 0
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 66
	Failure: Assertion 'event != NULL' failed
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 68
	Failure: Got -58, expected 0
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 72
	Failure: Got 0, expected 2
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 73
	Failure: Got 0, expected 3
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 74
	Failure: Got 0, expected 5
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 75
	Failure: Got 0, expected 8
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 76
	Failure: Got 0, expected 13
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 77
	Failure: Got 0, expected 21
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 78
	Failure: Got 0, expected 34
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 79
	Failure: Got 0, expected 55
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 80
	Failure: Got 0, expected 89
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 81
	Failure: Got 0, expected 144
Test 'TestExecutions::testFibonacci()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 85
	Failure: Got -58, expected 0
Test-method 'TestExecutions::testFibonacci()' finished with errors!
Suite 'TestExecutions' finished, 1/2 successful (50%) in 6689339 microseconds (6689.34 ms).
```

doe300, 2017-11-14:
TestExecutions currently fails, I don't know why, but these failures are "expected".

Why the `TestKernel::testSetKernelArg()` fails - or actually doesn't fail, even it is supposed to - I have to look into.

doe300, 2017-11-19:
Fixed with the latest commit

closed 2017-11-19 by doe300