doe300/VC4CL #4: Raise exception after `main`?

nomaddo, 2017-11-20:
In my OpenCL program (just call `test_mul24` in `test_int.cl`),  the exception is raised as follows:

```
terminate called without an active exception
Aborted
```

I am not sure what happened. But I see `~thread` is called in `__run_exit_handlers` in `_start`.
Some threads are finished (related to #3?).

My program is as follows. It is straightforward as OpenCL program.
Probably other program raise the same exception.....

```
#include <stdio.h>
#include <stdlib.h>
#include <time.h>
#include <assert.h>
#include <math.h>

#ifdef __APPLE__
#include <OpenCL/opencl.h>
#else
#include <CL/cl.h>
#endif

#define MAX_SOURCE_SIZE (0x100000)

#define N 1024

cl_int a[N];
cl_int b[N];

int main() {
  char fileName[] = "./test_int.cl";

  FILE *fp = fopen(fileName, "r");
  if (!fp) {
    fprintf(stderr, "Failed to load kernel\n");
    exit(1);
  }

  char * source_str = (char*)malloc(MAX_SOURCE_SIZE);
  size_t source_size = fread(source_str, 1, MAX_SOURCE_SIZE, fp);
  fclose(fp);

  cl_platform_id platform_id = NULL;
  cl_uint ret_num_platforms;
  cl_int ret = clGetPlatformIDs(1, &platform_id, &ret_num_platforms);
  cl_device_id device_id = NULL;
  cl_uint ret_num_devices;
  ret = clGetDeviceIDs(platform_id, CL_DEVICE_TYPE_DEFAULT, 1, &device_id, &ret_num_devices);

  cl_context context = clCreateContext(NULL, 1, &device_id, NULL, NULL, &ret);
  cl_command_queue command_queue = clCreateCommandQueue(context, device_id, 0, &ret);

  for (int i = 0; i < N; i++)
    a[i] = N;

  cl_mem memobj = clCreateBuffer(context, CL_MEM_READ_WRITE, sizeof(cl_float) * N, NULL, &ret);
  clEnqueueWriteBuffer(command_queue, memobj, CL_TRUE, 0, sizeof(cl_float) * N, a, 0, NULL, NULL);

  cl_program program = clCreateProgramWithSource(context, 1, (const char **)&source_str, (const size_t *)&source_size, &ret);
  ret = clBuildProgram(program, 1, &device_id, NULL, NULL, NULL);

  cl_kernel kernel = clCreateKernel(program, "test_mul24", &ret);
  cl_int arg0 = 2;

  ret = clSetKernelArg(kernel, 0, sizeof(cl_int), (void*)&arg0);
  ret = clSetKernelArg(kernel, 1, sizeof(cl_mem), (void*)&memobj);
  size_t global_item_size[] = { N };
  ret = clEnqueueNDRangeKernel (command_queue, kernel, 1, NULL,
				global_item_size, NULL, 0, NULL, NULL);
  ret = clFinish(command_queue);
  ret = clFlush(command_queue);
  ret = clEnqueueReadBuffer(command_queue, memobj, CL_TRUE, 0, sizeof(cl_float) * N, b, 0, NULL, NULL);
  ret = clReleaseKernel(kernel);
  ret = clReleaseProgram(program);
  ret = clReleaseMemObject(memobj);
  ret = clReleaseCommandQueue(command_queue);
  ret = clReleaseContext(context);
  free(source_str);

  return 0;
}
```

doe300, 2017-11-20:
The exception is thrown, because the background-thread actually handling the communication with the GPU is not correctly shut down: It is aborted by the program being stopped instead of stopping itself by returning from the thread-function.

In theory, the thread should be shut-down when the last command-queue is released, but it seems that doesn't happen.

doe300, 2017-11-20:
Testing with e.g. [boost-compute](https://github.com/boostorg/compute) test-case `test_adjacent_difference` does not throw this exception, but a few other test-programs do.

BTW: You have an error in your code, you don't set all 3 kernel arguments (there are 3: int num, int in and int* out), so `clEnqueueNDRangeKernel` fails.

Running your code throws, even with the above error fixed. Since your code seems to release all objects properly, there seems to be a bug in VC4CL.

nomaddo, 2017-11-20:
Sorry for confusing you because of my mistake. 
I have also checked that the same exception is thrown after fixing my bug.

doe300, 2017-11-20:
I found the issue:
I was expecting the caller of all the `clEnqueue*` functions to release the `event` passed to the functions once, after they are done handling the event. But if there is no `event` given, the caller cannot release it...

The latests commit now fixes the bug in your code as well as `TestVC4CL`.

nomaddo, 2017-11-20:
Thanks to your commits, my program can finish successfully.
But `TestVC4CL` make segmentation fail in `__cxa_finalize`.
It seems that other resource leaks may remind.

nomaddo, 2017-11-21:
As follows: 

 ```
(gdb) backtrace
#0  0x76f5b0d0 in std::default_delete<vc4cl::BaseObject>::operator() (this=<optimized out>, __ptr=0x1112f0)
    at /usr/include/c++/6/bits/unique_ptr.h:76
#1  std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >::~unique_ptr (this=<optimized out>, 
    __in_chrg=<optimized out>) at /usr/include/c++/6/bits/unique_ptr.h:239
#2  __gnu_cxx::new_allocator<std::_Rb_tree_node<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::destroy<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > (this=<optimized out>, __p=<optimized out>)
    at /usr/include/c++/6/ext/new_allocator.h:124
#3  std::allocator_traits<std::allocator<std::_Rb_tree_node<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > > >::destroy<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > (__a=..., __p=<optimized out>)
    at /usr/include/c++/6/bits/alloc_traits.h:487
#4  std::_Rb_tree<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::_Identity<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::less<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::allocator<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::_M_destroy_node (this=<optimized out>, __p=0x10f3b8) at /usr/include/c++/6/bits/stl_tree.h:567
#5  std::_Rb_tree<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::_Identity<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::less<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::allocator<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::_M_drop_node (this=0x76f8a378 <liveObjectsTracker>, __p=0x10f3b8)
    at /usr/include/c++/6/bits/stl_tree.h:575
#6  std::_Rb_tree<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::_Identity<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::less<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::allocator<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::_M_erase (__x=0x10f3b8, this=0x76f8a378 <liveObjectsTracker>)
    at /usr/include/c++/6/bits/stl_tree.h:1640
#7  std::_Rb_tree<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::_Identity<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::less<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::allocator<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::~_Rb_tree (this=0x76f8a378 <liveObjectsTracker>, __in_chrg=<optimized out>)
    at /usr/include/c++/6/bits/stl_tree.h:873
#8  std::set<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> >, std::less<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > >, std::allocator<std::unique_ptr<vc4cl::BaseObject, std::default_delete<vc4cl::BaseObject> > > >::~set (this=0x76f8a378 <liveObjectsTracker>, __in_chrg=<optimized out>) at /usr/include/c++/6/bits/stl_set.h:90
#9  vc4cl::ObjectTracker::~ObjectTracker (this=0x76f8a378 <liveObjectsTracker>, __in_chrg=<optimized out>)
    at /home/pi/work/VC4CL/src/ObjectTracker.cpp:18
#10 0x762231b0 in __cxa_finalize (d=0x76f896b8) at cxa_finalize.c:56
#11 0x76f351ac in __do_global_dtors_aux () from /home/pi/work/VC4CL/build/libVC4CL.so.1.2
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

closed 2017-11-23 by doe300

nomaddo, 2017-11-24:
In my env, this problem is reproduced during executing `TestVC4CL`.
The stack trace is as follows.

```
(gdb) bt                                                                                                                                                                                 
#0  0x76f5b8e4 in VC4CL_clReleasePerformanceCounterVC4CL (                                                                                                                               
    counter=<optimized out>)                                                                                                                                                             
    at /home/pi/work/VC4CL/src/PerformanceCounter.cpp:79                                                                                                                                 
#1  0x00084f3c in TestExtension::testReleasePerformanceCounters (                                                                                                                        
    this=0x7efffad0) at /home/pi/work/VC4CL/test/TestExtension.cpp:87                                                                                                                    
#2  0x76fa1684 in Test::Suite::runTestMethod(Test::Suite::TestMethod const&)                                                                                                             
    ()                                                                                                                                                                                   
   from /home/pi/work/VC4CL/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.1.1.2                                                                                   
#3  0x76fa1930 in Test::Suite::run(Test::Output&, bool) ()                                                                                                                               
   from /home/pi/work/VC4CL/build/cpptest-lite/src/cpptest-lite-project-build/libcpptest-lite.so.1.1.2                                                                                   
#4  0x0001c270 in main (argc=<optimized out>, argv=<optimized out>)                                                                                                                      
    at /home/pi/work/VC4CL/test/test.cpp:84  
```

commit 27533bf5 (2017-11-23): Fixes segfault in releasing leaked objects and errors in
performance-counters, fixes #4