doe300/VC4CL #5: Cmake configuration failed for cross-compilation

nomaddo, 2017-11-21:
Cmake configuration failed when I tried cross-compilation, because the execution `find_package(OpenCL REQUIRED)` failed. 
In this execution, cmake tries to compile some c program using `CMAKE_C_COMPILER`.
The problem is, this compiler run without any compiler-option to add include directories.
So, even though these exists `${CROSS_COMPILE}/include/CL`, finding OpenCL fails.

Probably `CMakeLists.txt` need to be modified.

My configuration command is as follows. Do I miss some specification?
Of course, `/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/include/CL` exists.
`/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc -I/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/include` works correctly.

```
cmake -DBUILD_TESTING=ON -DCROSS_COMPILE=ON -DCROSS_COMPILER_PATH=/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf
```

nomaddo, 2017-11-21:
Sorry, it seems my mistake.

nomaddo, 2017-11-21:
Sorry for post rapidly, but I seems the problem, at last.

nomaddo, 2017-11-21:
My mistake: not in `/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/include/CL`, but `/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include/CL`

 ```
$ :~/VC4CL$ rm -rf CMakeCache.txt  CMakeFiles; cmake -DBUILD_TESTING=ON -DCROSS_COMPILE=ON -DCROSS_COMPILER_PATH=/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Cross compiling for Raspbian with compiler: /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf
-- The C compiler identification is GNU 7.1.1
-- The CXX compiler identification is GNU 7.1.1
-- Check for working C compiler: /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
CMake Error at /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenCL (missing: OpenCL_LIBRARY) (found version "2.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.9/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.9/Modules/FindOpenCL.cmake:132 (find_package_handle_standard_args)
  CMakeLists.txt:58 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nomaddo/work/VC4CL/CMakeFiles/CMakeOutput.log".
```

doe300, 2017-11-21:
For the cross-compiler to find OpenCL headers, you need to have the OpenCL headers in the include-directory of the cross-compiler. I simply created a symlink in `/opt/rasperrypi/tools/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64/arm-linux-gnueabihf/include` (where the include-directory of my cross-compiler resides) to `/usr/include/CL` (where the OpenCL headers reside).

Similar for the OpenCL library: For the cross-compiler to be able to link it, you'll need to copy the OpenCL library (the one from your Raspberry Pi!) into the library-folder of the cross-compiler or make them available in the include path some other way.

nomaddo, 2017-11-21:
Yes, there exists `CL` in `${MY_CROSS_COMPILER_ROOT}/arm-linux-gnueabihf/include`.
The library too.

```
$ ls /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include
CL  c++
$ ls /opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/lib/libOpenCL*
/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/lib/libOpenCL.so
/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/lib/libOpenCL.so.1
/opt/gcc-linaro-7.1.1-2017.08-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/lib/libOpenCL.so.1.0.0
```

Upper execution log says, the headers found, but an error occurs.
And, `CMakeFiles/CMakeOutput.log` is created, but `CMakeFiles/CMakeError.log` not.
Is the cross-compilation successful in your env?

doe300, 2017-11-21:
No, VC4CL fails too on my machine. I fixed the inclusion of OpenCL-library by adding the command-line parameter `-DOpenCL_LIBRARY:PATH=<path-to-OpenCL-library>/libOpenCL.so` to CMake, now it errors on BCMHOST_LIBRARY not found. So I guess, you'll have to manually specify both OpenCL and bcm-host library.

I'd be happy, if you have any suggestion as to why CMake doesn't find the OpenCL library or how to automatize the inclusion of those libraries for cross-compilation.

nomaddo, 2017-11-21:
Thanks, I will try to solve this.
And, I want to add CI service (using Docker) to the project to standardize develop environment.
Suffering from the difference of build environment is hard.....

doe300, 2017-11-21:
I never worked with docker, but if it supports native Raspbian builds (incl. bcmhost_library already included), that would be great.

PR #9 (2017-11-28): Fix CMakeLists.txt for cross-compilation
I fixed #5 and add missing compilation flags of external projects.
In previous config, `CMAKE_FIND_ROOT_PATH` include `${CROSS_COMPILER_PATH}`, in addition to  `${CROSS_COMPILER_PATH}/arm-linux-gnueabihf`.
Somehow this two directories make `cmake` confused and show unexpected behaviors.
Fundamentally, `${CROSS_COMPILER_PATH}` is host x86 world. 
Specification to host world as client arm world is strange.

And, I add cross-compiler path or so on to external projects same as the previous pullreq for VC4C.
https://github.com/doe300/VC4C/pull/4
closed 2017-11-28 by doe300