doe300/VC4CL #7: Set milestones: support CLBlast?

nomaddo, 2017-11-24:
I think supporting practical libraries using OpenCL is one of the big milestone.
Probably [clblast](https://github.com/CNugteren/CLBlast) or [clblas](https://github.com/clMathLibraries/clBLAS) may be suitable for showing benchmark results and its utility.

Are you interested in such projects and setting milestones?
I want to contribute for such practical uses.

doe300, 2017-11-24:
True, passing all OpenCL CTS tests is of no use, if no library is supported. I shortly tested both libraries before (see [here](https://github.com/doe300/VC4C/blob/master/doc/Libraries.md)) with some errors.

I will check again the current progress of supporting these libraries and whether a full support is even possible (e.g. no fixed too larger work-group sizes, no use of `long` or `double`). If you want to contribute to supporting these libraries in any way, you are very welcome.

doe300, 2017-11-24:
Status for [clBLAS](https://github.com/clMathLibraries/clBLAS):
- some examples don't compile with Raspbian's gcc-4.9 (see [bug](https://github.com/clMathLibraries/clBLAS/issues/319))
- `libclBLAS` and `test-correctness` compile sucessfully

abhiTronix, 2019-02-19:
@doe300  can you try my Latest Raspbian's GCC cross compilers(Both 8.2.0 & 6.3.0 gcc binaries available) for this project?
Take a look, 
Project homepage: https://github.com/abhiTronix/raspberry-pi-cross-compilers
Project Binaries: https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/

I would see the project develop further.