doe300/VC4CL #8: v3d_info linking fails (symbols from libbcm_host not found)

senshu, 2017-11-27:
When building VC4CL, I get:

```
Linking CXX executable v3d_info
../libVC4CL.so.0.4: undefined reference to `bcm_host_get_peripheral_address'
../libVC4CL.so.0.4: undefined reference to `bcm_host_init'
../libVC4CL.so.0.4: undefined reference to `bcm_host_deinit'
```

Linking succeeds if I modify `tools/CMakeLists.txt`, adding `libbcm_host` to the libraries for `v3d_info` as follows:

```cmake
target_link_libraries(v3d_info VC4CL ${BCMHOST_LIBRARY})
```

doe300, 2017-11-27:
Strange, `libbcm_host` is already linked in the shared library `libVC4CL`, so the symbols should be available for `v3d_info`.

Have you done anything special with the build-configuration (e.g. cross compilation, out-of-source build)?

senshu, 2017-11-27:
I have tried to rebuild from scratch to confirm the issue but I cannot reproduce it...

> Have you done anything special with the build-configuration (e.g. cross compilation, out-of-source build)?

I'm building in a VM with a customized Raspbian Stretch, so I assume that the problem is in my setup.

senshu, 2017-11-27:
Just tried again on an official Raspbian Stretch image.
No linking error.

closed 2017-11-27 by senshu