doe300/VC4CL #12: Test debian packages

doe300, 2017-12-07:
(On a freshly installed Raspbian distribution) test installation and usage of the three created debian packages built with CircleCI.

Things to test:
- [ ] does VC4C find SPIRV-LLVM correctly in `/opt/SPIRV-LLVM/bin`?
- [ ] does VC4C build the PCH correctly on installation?
- [ ] do simple compilation tests run with VC4C?
- [ ] is VC4CL correctly configured to use VC4C? Does it find the `libVC4CC.so` in `/usr/local/lib`?
- [ ] is VC4CL correctly configured with ICD loader?
- [ ] does e.g. `clinfo` print correct results?
- [ ] do the tools `v3d_info` and `v3d_profile` work?

ThijsWithaar, 2017-12-08:
In case it's useful, my notes on compiling and installing on raspbian jessie (clinfo runs successfully)

Setting up cmake:
```
(cd VC4CLStdLib; mkdir build; cd build; cmake ..)
(cd VC4C; mkdir build; cd build; cmake -DSPIRV_FRONTEND=false ..)
(cd VC4CL; mkdir build; cd build; cmake -DVC4CC_LIBRARY=$(pwd)/../../VC4C/build/build/libVC4CC.so ..)
```

Build it all with 'make package', in each of the build directories

install:
```
sudo dpkg -i *.deb
sudo ldconfig -v
```

fangq, 2018-01-24:
@ThijsWithaar, thanks for sharing the commands, when I run the 3rd line (generating makefile for vc4cl), I got an error:

`none of the required 'ocl-icd>=1.3' found.
`
I do have ocl-icd-opencl-dev 2.2.11-1 installed, did I miss anything?

I also tried to type "make" inside VC4C/build after the Makefile is created, I got another error: 

`VC4C/src/Precompiler.cpp:125:38: error "CLANG_PATH" was not declared in this scope.
`
all codes were pulled from git today. any comment is appreciated.

fangq, 2018-01-25:
an update, I manually edited Precompiler.cpp and defined CLANG_PATH as a string "/usr/bin/clang", then make proceeds without further error. 

regarding the ocl-icd error, I solved it by installing ocl-icd-dev (README only mentioned about ocl-icd-opencl-dev).

In the end, I managed to create all deb  packages and install them. however, when I use "apt-get install clinfo" and then run clinfo, I got 0 available platforms. curious what additional setting I need to make in order to run clinfo.

doe300, 2018-01-27:
@fangq see [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1264810#p1264810) for the problem with`clinfo` seeing no platforms and a circumvention for it.

fangq, 2018-01-28:
thanks for the link, yes, I did resolve the clinfo issue by defining the LD_LIBRARYPATH.

maleroy, 2018-03-23:
I was following @doe300's instructions on how to get VC4CL, then @ThijsWithaar's on how to build, but when I try to "sudo dpkg -i *.deb" within VC4C's build directory, I get the error that vc4c-0.4-Linux.deb couldn't be processed because the "package architecture (amd64) does not match system (armhf)"

Additionally, I can't "make package" from within VC4CL's build directory because "file INSTALL cannot find "[my_dir]/VC4CL/VC4CL.icd"."

Any help would be greatly appreciated :) (I'm trying this from an RPi 3 model B)

doe300, 2018-03-24:
@maleroy , which instruction did you use? The ones from [here](https://github.com/doe300/VC4CL/wiki/How-to-get) should download the correct version for the ARM architecture.

I cannot reproduce the problem you encounter with `make package` though.