doe300/VC4CL #15: Clpeak compute-sp freezes with mailbox

doe300, 2017-12-20:
When executed via the mailbox-interface, the clpeak `--compute-sp` benchmark freezes (the kernel?) in the second execution. The `--global-bandwidth` benchmark runs fine, so do both benchmarks when run via register-poking.

More exact error description:
- according to htop, the `VC4CL Queue Handler` thread moves into status D, which means `uninterruptible sleep (usually IO)` (see [here](https://stackoverflow.com/questions/18470215/what-does-a-c-process-status-mean-in-htop)), which looks like it hangs in the mailbox-syscall
- forcibly stopping the process is unable to kill the `VC4CL Queue Handler` thread, even `sudo reboot` does nothing.

Side note: When this error is fixed, re-examine the delay in `src/executor.cpp:108`, since at least the `--global-bandwidth` benchmark seems not to require it (anymore).

ThijsWithaar, 2017-12-21:
Maybe related: since two weeks or so, `clinfo` freezes after:
`
  Max work group size                             12
`
This is with raspbian stretch, so gcc 6.3.0 and clang 3.9 installed. It used to work.

doe300, 2017-12-21:
@ThijsWithaar, do you use clinfo from the repository or a manually compiled version?
The version in the repository has an error, because of which I started using the [github version](https://github.com/Oblomov/clinfo). I just currently don't remember what the error was.

ThijsWithaar, 2017-12-21:
I use the one from raspbian (stretch)

doe300, 2017-12-23:
On my Raspberry Pi, the `clinfo` from the repository fails (before any output) with

> terminate called after throwing an instance of `unsigned int`

The manually compiled version runs fine, though I am not sure, this is related.