doe300/VC4CL #16: clinfo throws an exception

DV0000, 2018-01-08:
I am begginer in the use of OpenCL, but when I execute the `clinfo` command, I can check I have one platform with the following information:
 ```
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU

[...]

terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to enable QPUs!
Aborted
```
However, it fails and throws an exception before finishing. Could I solve this?

doe300, 2018-01-08:
Did you run `clinfo` as root? Since it maps the memory, it needs to be run with admin-rights.

DV0000, 2018-01-08:
Yes, I type `sudo clinfo` on the terminal, after a correct installation of VC4CL. 
My Raspberry Pi is running the Raspbian GNU/Linux 9 version of Raspbian

doe300, 2018-01-08:
Which Raspberry Pi are you using?
From the error-message, it seems the `/dev/vcio` file could be opened successfully, but the VideoCore IV hardware could not be enabled (`Mailbox.cpp:81-84`), but without throwing errors (so the syscall was successful too, but not with the expected results).

Can you reboot your Raspberry and try to run `sudo clinfo` again?

DV0000, 2018-01-08:
The model is Raspberry Pi 3 Model B Rev1.2.
I get the same result even if I reboot the Raspberry

doe300, 2018-01-08:
As mentioned in #15 , the "default" `clinfo` throws errors (though different errors than you reported) on my machine too. it is also ancient (I am running the "latest" clinfo version from 2013).

Can you download and compile https://github.com/Oblomov/clinfo manually and see whether it works?

DV0000, 2018-01-09:
I have checked the enabled GL Driver in the `/boot/config.txt` file. "OpenGL desktop driver with full KMS" was enabled, as selected in the `config.txt` line: `dtoverlay=vc4-kms-v3d`

By using the `raspi-config` tool, I disabled the OpenGL drivers. Now, both lines become comments:
```
#dtoverlay=vc4-kms-v3d
#dtoverlay=vc4-fkms-v3d
```
With this configuration, `clinfo` does work, showing the following information and errors (it takes a long time after the 'Max work group size 12' line):

```
  Platform Version                                OpenCL 1.2 VC4CL 0.4
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Version                                  OpenCL 1.2 VC4CL 0.4
  Driver Version                                  0.4
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Max compute units                               1
  Max clock frequency                             300MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
  Max work item dimensions                        3
  Max work item sizes                             12x12x12
  Max work group size                             12
  Preferred work group size multiple              <getWGsizes:498: build program : error -2>
  Preferred / native vector sizes                 
    char                                                16 / 16      
    short                                               16 / 16      
    int                                                 16 / 16      
    long                                                 0 / 0       
    half                                                 0 / 0        (n/a)
    float                                               16 / 16      
    double                                               0 / 0        (n/a)
  Half-precision Floating-point support           (n/a)
  Single-precision Floating-point support         (core)
    Denormals                                     No
    Infinity and NANs                             No
    Round to nearest                              No
    Round to zero                                 Yes
    Round to infinity                             No
    IEEE754-2008 fused multiply-add               No
    Support is emulated in software               No
    Correctly-rounded divide and sqrt operations  No
  Double-precision Floating-point support         (n/a)
  Address bits                                    32, Little-Endian
  Global memory size                              39845888 (38MiB)
  Error Correction support                        No
  Max memory allocation                           39845888 (38MiB)
  Unified memory for Host and Device              Yes
  Minimum alignment for any data type             64 bytes
  Alignment of base address                       512 bits (64 bytes)
  Global Memory cache type                        Read/Write
  Global Memory cache size                        <printDeviceInfo:89: get CL_DEVICE_GLOBAL_MEM_CACHE_SIZE : error -30>
  Global Memory cache line                        64 bytes
  Image support                                   No
  Local memory type                               Global
  Local memory size                               39845888 (38MiB)
  Max constant buffer size                        39845888 (38MiB)
  Max number of constant args                     64
  Max size of kernel argument                     256
  Queue properties                                
    Out-of-order execution                        No
[... 8 lines cut ...]
  Device Available                                Yes
  Compiler Available                              Yes
  Linker Available                                Yes
  Device Extensions                               cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_byte_addressable_store cl_nv_pragma_unroll cl_arm_core_id cl_ext_atomic_counters_32 cl_khr_initialize_memory

NULL platform behavior
  clGetPlatformInfo(NULL, CL_PLATFORM_NAME, ...)  OpenCL for the Raspberry Pi VideoCore IV GPU
  clGetDeviceIDs(NULL, CL_DEVICE_TYPE_ALL, ...)   Success [VC4CL]
  clCreateContext(NULL, ...) [default]            Success [VC4CL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1
```

doe300, 2018-01-09:
That sound plausible, that this is a conflict with the OpenGL driver. My development Raspberry Pi runs headless, so I never tested compatibility with the OpenGL driver.

> With this configuration, `clinfo` does work, showing the following information and errors (it takes a long time after the 'Max work group size 12' line):

At this point, `clinfo` compiles a test-program, which could take a few seconds. In your output, the compilation seem to fail ("`<getWGsizes:498: build program : error -2>`"). Have you built VC4CL with or without the VC4C compiler?

DV0000, 2018-01-09:
> In your output, the compilation seem to fail ("<getWGsizes:498: build program : error -2>"). Have you built VC4CL with or without the VC4C compiler?

I cloned and built the VC4C and VC4CLStdLib repositories before building VC4CL with the default options in CMake, so I used the VC4C compiler

doe300, 2018-01-09:
Can you re-build VC4CL with CMake option `BUILD_DEBUG` set to `ON`, re-run `clinfo`with the OpenGL driver enabled and post the output? If the output is too large, posting only the lines starting with `[VC4CL]...` will suffice.

This will print the actual contents of the sys-call (request and response) and should hint to what the issue is. My current theory is something along the lines of `the GPU cannot be "enabled" while it is in use`...

Another thing you could try is to remove the exception in `src/Mailbox.cpp:84` and see if the rest of the `clinfo` call runs correctly.

DV0000, 2018-01-10:
I re-built it using `cmake .. -DBUILD_DEBUG=ON` and run `clinfo` again, getting this output message:
`==968==ASan runtime does not come first in initial library list; you should either link runtime to your application or manually preload it with LD_PRELOAD.`

This `clinfo` output does not change with OpenGL enabled or disabled

doe300, 2018-01-11:
I disabled the check which failed with OpenGL driver enabled and fixed the error you mentioned above.

Can you re-build with `BUILD_DEBUG` enabled and post the results of `clinfo`, so I can see whether the fix I applied actually matches problem?

DV0000, 2018-01-11:
Of course. Thanks!

By rebuilding it with that CMake option, `clinfo` shows:

```
[VC4CL] get extension function address: clIcdGetPlatformIDsKHR
[VC4CL] get extension function address: clGetPlatformInfo
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Vendor ID                                0xa5c
  Device Version                                  OpenCL 1.2 VC4CL 0.4
  Driver Version                                  0.4
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Max compute units                               1
[VC4CL] Mailbox file descriptor opened: 3
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000001
[VC4CL] 0018: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x80000000
[VC4CL] 0018: 0x00000000

[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030004
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000005
[VC4CL] 0018: 0x00000000
[VC4CL] 001c: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030004
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000008
[VC4CL] 0014: 0x00000005
[VC4CL] 0018: 0x11e1a300
[VC4CL] 001c: 0x00000000

[VC4CL] Mailbox request: succeeded
  Max clock frequency                             300MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
  Max work item dimensions                        3
[VC4CL] base=0x3fc00000, mem=0x76f0d000
[... 190 lines cut ...]
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1

[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000000
[VC4CL] 0018: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x80000000
[VC4CL] 0018: 0x00000000

[VC4CL] Error in '/home/pi/VC4CL-GPU/VC4CL/src/Mailbox.cpp:91', returning status -5
[VC4CL] Mailbox file descriptor closed: 3

```

doe300, 2018-01-11:
It looks like the VC4 has some kind of reference counter of how often the mailbox was opened/closed and it returns 0 only for the first open and last close and 0x80000000 otherwise. I updated the code to reflect that, which now should also fix the error in the last 2 lines.

To fix the second error you get (in the compiler), you will have to delete `/usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch` and re-build it by making the target `vc4cl-stdlib` in the VC4C project. This needs currently be done every time the source-files in the VC4CLStdLib project change.

DV0000, 2018-01-12:
I removed `VC4CLStdLib.h.pch` and built VC4C (target `vc4cl-stdlib`) and VC4CL again. However, I get some errors:

```
[VC4CL] get extension function address: clIcdGetPlatformIDsKHR
[VC4CL] get extension function address: clGetPlatformInfo
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Vendor ID                                0xa5c
  Device Version                                  OpenCL 1.2 VC4CL 0.4
  Driver Version                                  0.4
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Max compute units                               1
[VC4CL] Mailbox file descriptor opened: 3
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000001
[VC4CL] 0018: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x80000000
[VC4CL] 0018: 0x00000000

[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030004
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000005
[VC4CL] 0018: 0x00000000
[VC4CL] 001c: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030004
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000008
[VC4CL] 0014: 0x00000005
[VC4CL] 0018: 0x11e1a300
[VC4CL] 001c: 0x00000000

[VC4CL] Mailbox request: succeeded
  Max clock frequency                             300MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
  Max work item dimensions                        3
[VC4CL] base=0x3fc00000, mem=0x76f05000
[VC4CL] V3D base: 0x76f05000
  Max work item sizes                             12x12x12
[... 188 lines cut ...]
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1

[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000000
[VC4CL] 0018: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x0000001c
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000004
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x80000000
[VC4CL] 0018: 0x00000000

[VC4CL] Mailbox file descriptor closed: 3
```
(It takes many time to continue after the` [VC4CL] Compiling source with: ` line)

Have I done something wrong?

doe300, 2018-01-12:
> (It takes many time to continue after the[VC4CL] Compiling source with: line)

At this point, the compilation is run, which takes a few seconds with the SPIR-V or LLVM library front-end and much longer with the LLVM IR front-end you currently use (see [here](https://github.com/doe300/VC4C/issues/7)).

The errors of this form (`[VC4CL] Error in '/home/pi/VC4CL-GPU/VC4CL/src/Device.cpp:400', returning status -1:No device for the given criteria: platform 0x17d9940`) are okay. `clinfo` queries here devices of device-types (e.g. CPU, OpenCL accelerator) which are not supported by the implementation, which is why the error is printed and an error-code is returned. `clinfo` handles the error-code correctly, as you can see in the following line:
> clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform

DV0000, 2018-01-17:
Thank you for all your feedback. Assuming that my compilation of VC4CL is operative now, would you have any example beyond clinfo, clpeak etc. proving that the GPU improves the performance when compared to using the CPU?

doe300, 2018-01-17:
No I don't. Beside clinfo and clpeak, most of the [boost-compute](https://github.com/boostorg/compute) test-cases run successfully as well as a some OpenCL CTS tests (see [here](https://github.com/doe300/VC4CL/wiki/OpenCL-CTS)). Other than that I have not really run any programs and especially no benchmarks that can be compared with CPU performance.

doe300, 2018-03-17:
@DV0000, can this be closed?

closed 2018-03-17 by DV0000