doe300/VC4CL #17: Switch to gcc 6.3 as minimum requirement?

doe300, 2018-01-23:
Currently, VC4C and VC4CL are held compatible with the ancient GCC 4.8 to support older Raspbian systems. GCC 4.8 barely supports C++11 [1] and hence lacks a lot of useful new features provided in C++14 and C++17. The latest Raspian Stretch ships with GCC 6.3, which completely supports C++14 as well as a lot of C++17 features [1].

Some useful C++14/C++17 features we could use when switching [1]:

- Full support for variadic templates
- Extended constexpr functions
- `__has_include` e.g. for distinguishing LLVM versions
- New library features: `string_view`, file-system support, `optional` and `any` in experimental namespace

Since the official raspberrypi-tools repository has not yet been upgraded to GCC 6.3 (see [here](https://github.com/raspberrypi/tools/issues/81)), we would need to use another tool-chain for our CircleCI cross-compilation such as [this one](https://hub.docker.com/r/idein/cross-rpi/ ) (see also [here](https://github.com/doe300/VC4C/issues/20)).

So the question is:
**Should we drop support for GCC 4.8 and therefore for older Raspbian systems?**

[1] https://gcc.gnu.org/projects/cxx-status.html

fighne, 2018-02-16:
I think the ease of access to the Raspbian image, and the supported hardware ( I have an original Raspberry Pi which it runs on ) I personally think it's a yes.

doe300, 2018-02-21:
Just a quick questions to anyone who has more experience with gcc-versions and targets than I have.
For the CircleCI build, can't we just switch to `g++-arm-linux-gnueabihf` as cross-compiler? Or do we have to use a custom-build compiler for the raspberry pi (as [idein](https://github.com/Terminus-IMRC/cross-rpi/blob/master/Dockerfile) does)? If so, why?
@nomaddo, any ideas?

nomaddo, 2018-02-21:
@doe300 
Why we use custom-build compiler is that, other choices has cons as following:

- linaro cross compilers: armv6 compiler lacks support for hardfp
- `g++-arm-linux-gnueabihf`: it lacks support for NEON

In this case, we care support for march `armv6` and `armv7` with hardfp, not care for NEON acceleration.
OK, I will create new Dockerfile.

Terminus-IMRC, 2018-02-21:
My cross-rpi repo is obsolete, so please use our new one: https://github.com/Idein/docker-cross-rpi

Terminus-IMRC, 2018-02-21:
@nomaddo linaro's arm toolchain lacks support for *armv6* itself, not vfp...

nomaddo, 2018-02-21:
@Terminus-IMRC very thanks for the correction :)

Terminus-IMRC, 2018-02-21:
After some investigations, it turned out that cross toolchain of Linaro and Debian both support armv6 (VFP) and armv7 (NEON), which means that you can just switch to use `g++-arm-linux-gnueabihf` as cross compiler. Sorry for misleading comments.

closed 2018-02-23 by doe300