doe300/VC4CL #19: VC4CL clCompileProgram failed with "No matching precompiler available" error

fangq, 2018-01-27:
I compiled and installed VC4CL on a RPI v3 with Debian stretch. Running clinfo under root works. 

When I run my own OpenCL code (mcxcl: http://github.com/fangq/mcxcl) using vc4cl, I got a "**No matching precompiler available**" error, see log below:

```
==============================================================================
- code name: [Vanilla MCXCL] compiled with OpenCL version [1]
- compiled with: [RNG] Logistic-Lattice [Seed Length] 5
initializing streams ...	terminate called without an active exception
init complete : 0 ms
Building kernel with option: -cl-mad-enable 
Kernel build log:
[E] Sat Jan 27 00:08:04 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xac [0x7668e23c]
[E] Sat Jan 27 00:08:04 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x1e0 [0x76690608]
[E] Sat Jan 27 00:08:04 2018:  (3) /usr/local/lib/libVC4CL.so : +0x4a630 [0x76983630]
[E] Sat Jan 27 00:08:04 2018:  (4) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&, void (*)(_cl_program*, void*), void*)+0xd8 [0x76984bc4]
[E] Sat Jan 27 00:08:04 2018:  (5) /usr/local/lib/libVC4CL.so : VC4CL_clCompileProgram+0xe88 [0x76987830]
[E] Sat Jan 27 00:08:04 2018:  (6) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x1a8 [0x76988468]
[E] Sat Jan 27 00:08:04 2018:  (7) ../../bin/mcxcl() [0x13834]
[E] Sat Jan 27 00:08:04 2018: Compiler threw exception: Pre-compilation: No matching precompiler available!

MCX ERROR(2):Error: Failed to build program executable! in unit mcx_host.cpp:450
Aborted
```

my opencl kernel was tested on nvidia/intel/amd ocl implementations, I am wondering if the above error is related to my setting or compatibility of the kernel to vc4cl. 

please let me know if you have any suggestions how to solve this issue. thanks

To reproduce this issue, you may follow the below commands as root

```
git clone https://github.com/fangq/mcxcl.git 
cd mcxcl/src
make clean all
cd ../example/benchmark
LD_LIBRARY_PATH=/usr/local/lib ./run_benchmark1.sh -n 1e5 
```

doe300, 2018-01-27:
> Running clinfo under root works.

What is the value or error `clinfo` prints for the line `Preferred work group size multiple`?

If that is an error, do you have CLang or SPIRV-LLVM installed and configured VC4C to use them?

If that line outputs no error (usually the value 0 or 1), can you try to compile without `-cl-mad-enable` set?

fangq, 2018-01-27:
@doe300, thanks for your reply. I do see an error

Preferred work group size multiple              <getWGsizes:498: build program : error -2>

I installed clang 3.8.1-24+rpi1, and followed the compilation steps from [ThijsWithaar's post](https://github.com/doe300/VC4CL/issues/12#issuecomment-350241471). The only tweak I made was to replace the CLANG_PATH macro the /usr/bin/clang string on https://github.com/doe300/VC4C/blob/master/src/Precompiler.cpp#L254 

I believe in ThijsWithaar's commands, there is a SPIRV_FRONTEND=false flag.

can you suggest a command to compile VC4C?

doe300, 2018-01-28:
I would suggest using the LLVM library front-end, which also depends on `llvm-3.9-dev`, but is a lot faster and more stable than the fall-back LLVM IR front-end.

You can set the following flags to compile with "default" CLang and the LLVM library front-end:

    -DSPIRV_FRONTEND=false -DLLVMLIB_FRONTEND=true

If the installed clang binary is not found automatically, it can be set via `-DCLANG_FOUND=/usr/bin/clang-3.9`, same with the llvm-config program, which can be set via `-DLLVM_CONFIG_PATH=/usr/bin/llvm-config-3.9`.

fangq, 2018-01-28:
thanks, I searched for the "No matching precompiler available!" in VC4C/src, and found it in the Compiler.cpp unit

https://github.com/doe300/VC4C/blob/master/src/Compiler.cpp#L187-L195

it looks like this error was triggered by the missing of the CLANG_PATH macro, same as the one I [encountered earlier](https://github.com/doe300/VC4CL/issues/12#issuecomment-360371213) in the Precompiler.cpp.

I have llvm-3.9-dev installed, and when I tried "cmake -DSPIRV_FRONTEND=false -DLLVMLIB_FRONTEND=true", I saw 

`-- Compiling LLVM library front-end with LLVM in version 3.9.1 located in '/usr/lib/llvm-3.9/lib'`

however, when I make, I still ended up with the same error.  

In CMakeCache.txt, I see 

```
CLANG_FOUND:FILEPATH=/usr/bin/clang
CLANG_PATH:UNINITIALIZED=/usr/bin/clang
```

I am not sure why the CLANG_PATH has "UNINITIALIZED" after it. If I append "-DCLANG_PATH=/usr/bin/clang" to cmake, it gives me a warning: 

```
CMake Warning:
  Manually-specified variables were not used by the project:

    CLANG_PATH
```

I guess the key here is how to let cmake to generate a makefile that defines the -DCLANG_PATH=/usr/bin/clang macro (or should it be another macro pointing to llvm-3.9 instead?).

doe300, 2018-01-29:
CMake should set the `CLANG_PATH` macro correctly when it finds the clang executable. Try removing `CMakeCache.txt` and `CMakeFiles` and then re-running CMake with the options I listed above.

fangq, 2018-01-29:
@doe300, I tried what you suggested (deleting the cmake cache files) and more (appending -DCLANG_PATH in make command, or cmake command, or defining environment variables etc), all of those failed with the same error. 

However, I was able to get around that with an ugly but effective hack, which is to modify the compilation flags directly using the following command (after cmake ...)

```
cd VC4C
find . -name "flags.make" -exec sed -i 's/Wsuggest-attribute=const/Wsuggest-attribute=const -DCLANG_PATH=\"\\\"\/usr\/bin\/clang\\\"\"/g' '{}' \;
```
the /usr/bin/clang is a link to clang-3.9.

after this change, I was able to compile VC4C without an error. However, I am still unable to compile my kernel. Now it gives me a different error.

```
root@raspberrypi:~/space/mcxcl/example/benchmark# ./run_benchmark1.sh -n 1e5
...
==============================================================================
$Rev::6e839e $ Last $Date::2017-07-20 12:46:23 -04$ by $Author::Qianqian Fang$
==============================================================================
- code name: [Vanilla MCXCL] compiled with OpenCL version [1]
- compiled with: [RNG] Logistic-Lattice [Seed Length] 5
initializing streams ...	init complete : 0 ms
Building kernel with option: 
Kernel build log:
[E] Mon Jan 29 20:47:58 2018: Errors in precompilation:
[E] Mon Jan 29 20:47:58 2018: Stack dump:
0.	Program arguments: /usr/bin/clang -cc1 -triple spir-unknown-unknown  -O3 -ffp-contract=off  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /home/pi/space/ocl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /tmp/vc4c-78YYTi - 

[E] Mon Jan 29 20:47:58 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x7665f5ec]
[E] Mon Jan 29 20:47:58 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x2d77b8 [0x766947b8]
[E] Mon Jan 29 20:47:58 2018:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0x2d7af0 [0x76694af0]
[E] Mon Jan 29 20:47:58 2018:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x368 [0x7669535c]
[E] Mon Jan 29 20:47:58 2018:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x1bc [0x7666162c]
[E] Mon Jan 29 20:47:58 2018:  (6) /usr/local/lib/libVC4CL.so : +0x4a630 [0x76981630]
[E] Mon Jan 29 20:47:58 2018:  (7) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&, void (*)(_cl_program*, void*), void*)+0xd8 [0x76982bc4]
[E] Mon Jan 29 20:47:58 2018:  (8) /usr/local/lib/libVC4CL.so : VC4CL_clCompileProgram+0xe88 [0x76985830]
[E] Mon Jan 29 20:47:58 2018:  (9) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x1a8 [0x76986468]
[E] Mon Jan 29 20:47:58 2018:  (10) ../../bin/mcxcl() [0x1381c]
[E] Mon Jan 29 20:47:58 2018: Compiler threw exception: Pre-compilation: Error in precompilation: Stack dump:
0.	Program arguments: /usr/bin/clang -cc1 -triple spir-unknown-unknown  -O3 -ffp-contract=off  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /home/pi/space/ocl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /tmp/vc4c-78YYTi - 

MCX ERROR(2):Error: Failed to build program executable! in unit mcx_host.cpp:450
terminate called without an active exception
Aborted
```

from the above log, it looks like clang was found and used, but the precompilation failed.

My kernel (*.cl) can be found here
https://github.com/fangq/mcxcl/blob/master/src/mcx_core.cl

in your opinion, is this error due to a bug in processing this kernel or still my VC4C compilation settings?

doe300, 2018-01-30:
I don't understand, why CMake fails to auto-configure CLang on your platform.

> I tried what you suggested (deleting the cmake cache files) and more (appending -DCLANG_PATH in make command, [...]

You need to set `-DCLANG_FOUND`, the macro `CLANG_PATH` is generated from it.

> Now it gives me a different error.

This looks like an error in CLang, maybe the PCH is built for the wrong version. Try rebuilding it by deleting `../VC4CLStdLib/include/VC4CLStdLib.h.pch` and making the target `make vc4cl-stdlib`.

> in your opinion, is this error due to a bug in processing this kernel or still my VC4C compilation settings?

On my setup, compiling this kernel passes the pre-compilation but throws an error somewhere inside the VC4C compiler. So the error in pre-compilation is probably due to wrong configuration.

> I still see "error -2" for the "Preferred work group size multiple" output in clinfo

Since the compilation process does not work yet on your setup, this is to be expected

> The difference is the stock clinfo also show another error

Yeah, that was an error, where `clinfo` expected the wrong type size on 32-bit machines, which has been fixes some time ago in clinfo.

fangq, 2018-02-01:
> You need to set -DCLANG_FOUND, the macro CLANG_PATH is generated from it.

I tried, did not solve the Precompiler.cpp error though. I will try a fresh installation on another pi3 and see if this happens again.

> This looks like an error in CLang, maybe the PCH is built for the wrong version. Try rebuilding it by deleting ../VC4CLStdLib/include/VC4CLStdLib.h.pch and making the target make vc4cl-stdlib.

tried this too, deleted the .pch, recompiled both VC4C and VC4CStdLib, and reinstalled the packages, but got the same precompiler error when running the benchmark.

josiahw, 2018-02-04:
I think this might be linked to the project not updating correctly when remaking over an old pull. 

My failure CLANG_PATH was fixed with:
`git clean -fdx && cmake . && make clean && make -j2`
This deletes all the files that aren't in original git repository and starts from scratch pulling in subrepositories.

fangq, 2018-02-20:
An update, I set up a new rpi zero w with a fresh installation of raspbian (2017-11-29-raspbian-stretch). After booting into rpi, I installed the needed dependencies (cmake, ocl-icd-opencl-dev, ocl-icd-dev, ocl-opencl-dev) and then run the compilation commands as in https://github.com/doe300/VC4CL/issues/12#issuecomment-350241471 and I was able to compile/install  everything without the CLANG error. 

After a reboot and setting the LD_LIBRARY_PATH, and switching to root, I was able to run clinfo without the previous error. I did notice two things:

1. clinfo paused for nearly 20-30 seconds to print out the "Preferred work group size multiple" value, which is 1; I did a top and saw clang-3.9 was busy running during this period of time
2. I tried "-DSPIRV_FRONTEND=false -DLLVMLIB_FRONTEND=true" in the cmake of VC4C, but it still ended up using clang compiler. I do have the llvm-3.9 installed.

However, running my kernel (mcxcl) still aborted with 2 warnings and an error (Compiler threw exception: Parser: Constant value is out of valid range: 4607182418800017408), after a nearly 1 min compilation by clang. The error log is attached below.

I am wondering if there is a fix to this issue?

```
root@raspberrypi:/home/pi/temp/mcxcl/example/benchmark# ./run_benchmark1.sh -n 1e5
...
Building kernel with option: -cl-mad-enable -DMCX_USE_NATIVE -DMCX_SIMPLIFY_BRANCH -DMCX_VECTOR_INDEX -DMCX_SRC_PENCIL 
Kernel build log:
[W] Tue Feb 20 17:05:21 2018: Warnings in precompilation:
[W] Tue Feb 20 17:05:21 2018: <stdin>:61:9: warning: 'NULL' macro redefined
#define NULL               0
        ^
/home/pi/temp/VC4C/../VC4CLStdLib/include/_config.h:93:9: note: previous definition is here
#define NULL ((void *)0)
        ^
<stdin>:1040:1: warning: null character ignored
<U+0000>
^
2 warnings generated.

[E] Tue Feb 20 17:05:28 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0xb665977c]
[E] Tue Feb 20 17:05:28 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::toConstant(vc4c::Module&, llvm::Value const*)+0x174 [0xb67bb6e8]
[E] Tue Feb 20 17:05:28 2018:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::toValue(vc4c::Method&, llvm::Value const*)+0x534 [0xb67bb274]
[E] Tue Feb 20 17:05:28 2018:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseInstruction(vc4c::Module&, vc4c::Method&, vc4c::PerformanceList<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> >, (vc4c::AccessType)2, (vc4c::InsertRemoveType)3>&, llvm::Instruction const&)+0x940 [0xb67b7c8c]
[E] Tue Feb 20 17:05:28 2018:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseFunctionBody(vc4c::Module&, vc4c::Method&, vc4c::PerformanceList<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> >, (vc4c::AccessType)2, (vc4c::InsertRemoveType)3>&, llvm::Function const&)+0x140 [0xb67b6ae0]
[E] Tue Feb 20 17:05:28 2018:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseFunction(vc4c::Module&, llvm::Function const&)+0x5ac [0xb67b68e4]
[E] Tue Feb 20 17:05:28 2018:  (7) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parse(vc4c::Module&)+0x14c [0xb67b5064]
[E] Tue Feb 20 17:05:28 2018:  (8) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::convert()+0xfc [0xb665b1d0]
[E] Tue Feb 20 17:05:28 2018:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x244 [0xb665b844]
[E] Tue Feb 20 17:05:28 2018:  (10) /usr/local/lib/libVC4CL.so : +0x49378 [0xb69bd378]
[E] Tue Feb 20 17:05:28 2018:  (11) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&, void (*)(_cl_program*, void*), void*)+0xd8 [0xb69be910]
[E] Tue Feb 20 17:05:28 2018:  (12) /usr/local/lib/libVC4CL.so : VC4CL_clCompileProgram+0xe64 [0xb69c1558]
[E] Tue Feb 20 17:05:28 2018:  (13) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x1a8 [0xb69c214c]
[E] Tue Feb 20 17:05:28 2018:  (14) ../../bin/mcxcl() [0x13994]
[E] Tue Feb 20 17:05:28 2018: Compiler threw exception: Parser: Constant value is out of valid range: 4607182418800017408

MCX ERROR(2):Error: Failed to build program executable! in unit mcx_host.cpp:475
terminate called without an active exception
Aborted
```

doe300, 2018-02-21:
> [...] and I was able to compile/install everything without the CLANG error.

I think the CLANG not found error only occurs when re-building an already configured cmake-project, but I haven't figured it out, why.

> clinfo paused for nearly 20-30 seconds to print out the "Preferred work group size multiple" value

At this point clang compiles an example kernel, which takes some time. SPIRV-LLVM/clang is faster at compiling than the default clang, but takes more effort to set up.

> [..] but it still ended up using clang compiler

There is no way around using clang to compile OpenCL -> LLVM IR. This can only be fixed, if someone can point me to another OpenCL -> something compiler where the intermediate output can be read.
What you can select, is which tool is used to read the output of clang, e.g. the LLVM library or SPIRV-Tools.

> However, running my kernel (mcxcl) still aborted with 2 warnings and an error

The problem is, that your compiler uses 64-bit values (`ulong` and `double`) which are not supported by VC4C. Since the VideoCore IV architecture is 32-bit, VC4C only supports up to 32-bit data-types.
I have to admit, the error-message could be improved in such cases.

fangq, 2018-02-23:
@doe300, your comments on the involvement of 64bit data is very helpful. In my code, the 64bit data can be disabled by appending the -J "-DUSE_LL5_RAND" flag. Now I am getting a new error. I posted the log for the new error in a separate issue (https://github.com/doe300/VC4CL/issues/24) for easy management.

Feel free to close this issue, I believe it is now fixed after your fix to https://github.com/doe300/VC4C/issues/34

closed 2018-03-17 by doe300