doe300/VC4CL #21: Could you provide a Image on Raspberry PI 3?

LeeHaoCN, 2018-02-12:
I clone and build the CV4C and SPIRV-LLVM but can't pass the test.Could you provide a Image on Raspberry PI 3?

doe300, 2018-02-12:
Can you please provide more details?
What test can't you pass?
Do you run a 64-bit OS on your Raspberry Pi 3?

LeeHaoCN, 2018-02-13:
I'm run 32-bit OS on my Pi 3.My OS  distribution information:
```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch
```
And when I running `TestVC4C`,I get the following message:
```
Test-method 'RegressionTest::testPending("./example/md5.cl", "", 0)' failed with exception!
        Exception: Instruction Mapping: Bit-casts across different vector-sizes are not yet supported!
        Errno: 0
        Error: Success

Test-method 'RegressionTest::testPending("./example/SHA-256.cl", "", 0)' failed with exception!
        Exception: Instruction Mapping: Operand needs to refer to a memory location or a parameter containing one: i8* %4294967295
        Errno: 0
        Error: Success

Test-method 'RegressionTest::testPending("./example/test_cl.cl", "", 0)' failed with exception!
        Exception: General: Shuffling vectors with non-constant mask-layout is not supported yet: <2 x i32> %372
        Errno: 0
        Error: Success
```
I'm build the `SPIRV-LLVM` fllow [http://llvm.org/docs/HowToBuildOnARM.html](http://llvm.org/docs/HowToBuildOnARM.html).But I'm not sure if it is the right way to build the llvm.

doe300, 2018-02-13:
These error messages are okay. These tests contain kernels which are not yet supported. In total, the `RegressionTest` test-suite fails about 80 of the ~280 tests currently available.

closed 2018-03-17 by doe300