doe300/VC4CL #22: Installation Instructions

fighne, 2018-02-16:
Using the instructions here 'https://github.com/doe300/VC4CL/wiki/How-to-get'
I'm stumbling at the first stage of installing, there doesn't appear to be a working link or on the nightly build either.

I must be missing something obvious could you point me in the right direction please.

doe300, 2018-02-16:
Look again at the [wiki](https://github.com/doe300/VC4CL/wiki/How-to-get). I had the instructions lying around for some time, just forgot to push them, sorry.

A side-note: The debian packages have not been tested (see #12) and they need (until https://github.com/doe300/VC4C/issues/29 is finished) SPIR-LLVM to be installed at the correct location (see [here](https://github.com/KhronosGroup/SPIRV-LLVM#build-instructions) for instructions).

fighne, 2018-02-16:
Many thanks,  can I suggest that to reduce the confusion specifying the Clang v3.9+ as the C++11 compiler. I have to say the less people have to do to install the less support issues you'll get. :)
But great work and many thanks - Pete

closed 2018-03-17 by doe300