doe300/VC4CL #23: Migrate to new dockeimage

nomaddo, 2018-02-23:
Currently, the problem is

- (SOLVED) missing wget and curl
  https://github.com/nomaddo/cross-rpi/pull/4
- Failed to find OpenCL
```
nomaddo@nomaddo-AS:~/idein/VC4CL$ circleci build

====>> Spin up Environment
Build-agent version 0.0.4704-deba4df (2018-02-21T21:01:29+0000)
Starting container nomaddo/cross-rpi:0.1

Using build environment variables:
  BASH_ENV=/tmp/.bash_env-localbuild-1519377596
  CI=true
  CIRCLECI=true
  CIRCLE_BRANCH=master
  CIRCLE_BUILD_NUM=
  CIRCLE_JOB=build
  CIRCLE_NODE_INDEX=0
  CIRCLE_NODE_TOTAL=1
  CIRCLE_REPOSITORY_URL=http://github.com/doe300/VC4CL.git
  CIRCLE_SHA1=68be86b590b85eb37862b01f50753a9ff5af3b48
  CIRCLE_SHELL_ENV=/tmp/.bash_env-localbuild-1519377596
  CIRCLE_WORKING_DIRECTORY=~/project

====>> Checkout code
  #!/bin/sh
mkdir -p /root/project && cp -r /tmp/_circleci_local_build_repo/. /root/project
====>> get vc4c url
  #!/bin/bash -eo pipefail
curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C/latest/artifacts?branch=master&filter=successful" --output /tmp/dump
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1089  100  1089    0     0   1289      0 --:--:-- --:--:-- --:--:--  1288
====>> get vc4cl-stdlib package
  #!/bin/bash -eo pipefail
wget -O /tmp/vc4cl-stdlib.deb $(python .circleci/get_url.py "vc4cl-stdlib-" "/tmp/dump")
--2018-02-23 09:19:59--  https://76-106166771-gh.circle-artifacts.com/0/root/project/vc4cl-stdlib-0.4-Linux.deb
Resolving 76-106166771-gh.circle-artifacts.com (76-106166771-gh.circle-artifacts.com)... 34.194.4.99, 34.198.74.173, 52.44.239.229, ...
Connecting to 76-106166771-gh.circle-artifacts.com (76-106166771-gh.circle-artifacts.com)|34.194.4.99|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-debian-package]
Saving to: '/tmp/vc4cl-stdlib.deb'

/tmp/vc4cl-stdlib.d     [  <=>               ] 103.90K   294KB/s    in 0.4s    

2018-02-23 09:20:01 (294 KB/s) - '/tmp/vc4cl-stdlib.deb' saved [106390]

====>> get vc4c package
  #!/bin/bash -eo pipefail
wget -O /tmp/vc4c.deb $(python .circleci/get_url.py "vc4c-" "/tmp/dump")
--2018-02-23 09:20:01--  https://76-106166771-gh.circle-artifacts.com/0/root/project/vc4c-0.4-Linux.deb
Resolving 76-106166771-gh.circle-artifacts.com (76-106166771-gh.circle-artifacts.com)... 52.72.204.202, 52.44.239.229, 34.198.74.173, ...
Connecting to 76-106166771-gh.circle-artifacts.com (76-106166771-gh.circle-artifacts.com)|52.72.204.202|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-debian-package]
Saving to: '/tmp/vc4c.deb'

/tmp/vc4c.deb           [      <=>           ]   7.84M  4.21MB/s    in 1.9s    

2018-02-23 09:20:04 (4.21 MB/s) - '/tmp/vc4c.deb' saved [8223320]

====>> unpack dependencies
  #!/bin/bash -eo pipefail
dpkg-deb -x /tmp/vc4cl-stdlib.deb / && dpkg-deb -x /tmp/vc4c.deb /
====>> configure
  #!/bin/bash -eo pipefail
cmake . -DCROSS_COMPILE=ON -DBUILD_TESTING=ON -DREGISTER_POKE_KERNELS=ON
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Cross compiling for Raspbian with compiler: /usr/bin/arm-linux-gnueabihf-g++
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - found
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find OpenCL (missing: OpenCL_LIBRARY) (found version "2.0")
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindOpenCL.cmake:127 (find_package_handle_standard_args)
  CMakeLists.txt:66 (find_package)

-- Configuring incomplete, errors occurred!
See also "/root/project/CMakeFiles/CMakeOutput.log".
Error: Exited with code 1
Step failed
====>> Uploading artifacts
Uploading /root/project/build/libVC4CL.so.0.4 to root/project/build/libVC4CL.so.0.4
  No artifact files found at /root/project/build/libVC4CL.so.0.4
====>> Uploading artifacts
Uploading /root/project/build/test/TestVC4CL to root/project/build/test/TestVC4CL
  No artifact files found at /root/project/build/test/TestVC4CL
====>> Uploading artifacts
Uploading /root/project/vc4cl-0.4-Linux.deb to root/project/vc4cl-0.4-Linux.deb
  No artifact files found at /root/project/vc4cl-0.4-Linux.deb
Task failed
```

I installed opencl-headers (see https://github.com/nomaddo/cross-rpi/blob/master/Dockerfile#L36).
I am not sure why failed. @doe300 Do you have any idea?

P.S.
Now, In my dockerfile, I give up building SPIRV-LLVM and download it from archive.
Auto-building of docker images works. You can push `nomaddo/cross-rpi` and try images.

doe300, 2018-02-23:
We need to add these packages (in armhf version): `ocl-icd-opencl-dev` (contains the missing `libOpenCL.so`) and `ocl-icd-dev` (contains the OpenCL ICD header)

doe300, 2018-02-23:
CircleCI suddenly fails again with python not available (see [here](https://circleci.com/gh/doe300/VC4CL/23)).
@nomaddo , is the update in docker not yet built?

nomaddo, 2018-02-23:
The new docker image seems still building.
After about minutes, it will be finished.

nomaddo, 2018-02-23:
@doe300 Can you add me to VC4CL's collaborator?
Then, I can try to rebuild VC4CL in circleci (I can do for VC4C, but not for VC4CL)

doe300, 2018-02-23:
Done.
FYI, I will off-line starting tomorrow for a few days.

nomaddo, 2018-02-23:
The build status become green. This issue can be closed !!!
And, we need to try cross-compiled binaries in real Raspberry Pi

closed 2018-02-23 by nomaddo