doe300/VC4CL #31: Running with DeepSpeech (TensorFlow OpenCL/ComputeCpp)

lissyx, 2018-05-23:
I'm currently trying to assert the status of expectation we can have on this setup for DeepSpeech, relying on TensorFlow with ComputeCpp. I have been able to cross-build the driver, and most of the `TestVC4C` do run (properly or not). That means, I can see `clang` doing its job and compile some cl stuff.

The GPU is also visible by `computecpp_info`.

Now, I'm trying to run our code on top of that. So far, it's not being very successfull, but in an unexpected way: as documented in https://github.com/codeplaysoftware/computecpp-sdk/issues/117#issuecomment-391356616, ComputeCpp does see the GPU, and makes use of it. But then, monitoring the system, it's sitting with the `deepspeech` process at 100%.

I don't see that much of `clang` running, but I did spot some process `llvm-spirv /tmp/vc4c-EgnXeW /dev/stdin` being ran. The file `/tmp/vc4c-EgnXeW` seems to be non-zero size. But no error when running, so I don't know if there's something going on.

The OpenCL kernels might be big (too big for the current limitations? I'm not sure how to check that), and/or the project might be still too young?

As a comparison, we are able to run with the same stack on the Intel Neo driver on my laptop (i7-8650U) using the GPU. The first run of the intel driver does compile the OpenCL code and can cache it on-disk, and this ~220 secs to compile.

Resulting `cl_cache` for Intel is:
```
$ LC_ALL=C ll cl_cache/
total 16M
drwxr-xr-x 1 alex alex  700 May 23 16:31 .
drwxr-xr-x 1 alex alex  222 May 23 16:28 ..
-rw-r--r-- 1 alex alex  73K May 23 16:31 34dcc67199396f96.cl_cache
-rw-r--r-- 1 alex alex 734K May 23 16:30 3b459da76e3bb7c4.cl_cache
-rw-r--r-- 1 alex alex 6.7M May 23 16:30 3d76036abef34a33.cl_cache
-rw-r--r-- 1 alex alex 206K May 23 16:31 66637feca75f4aba.cl_cache
-rw-r--r-- 1 alex alex 443K May 23 16:30 6ad895b3f2af8633.cl_cache
-rw-r--r-- 1 alex alex 728K May 23 16:31 6d252b514c3b944f.cl_cache
-rw-r--r-- 1 alex alex 165K May 23 16:30 8a505eb1718a2ecd.cl_cache
-rw-r--r-- 1 alex alex 425K May 23 16:30 98e4482263ced159.cl_cache
-rw-r--r-- 1 alex alex 523K May 23 16:30 9e55731fa2c7901e.cl_cache
-rw-r--r-- 1 alex alex 124K May 23 16:30 a493ba4fcff0b7d1.cl_cache
-rw-r--r-- 1 alex alex 122K May 23 16:29 b1a69f78813be187.cl_cache
-rw-r--r-- 1 alex alex 108K May 23 16:31 d2e0ea42ff6b24ac.cl_cache
-rw-r--r-- 1 alex alex 344K May 23 16:31 ec50bb7158952ee2.cl_cache
-rw-r--r-- 1 alex alex 5.2M May 23 16:31 f4146d630607b157.cl_cache
```

I've let run `deepspeech` with the `VC4` driver run for ~120m without any visible output or error: is it possible our code is too much compute intensive for now, and it's expected to take that much time ? Or could there be some silent error happening and breaking something ?

As much as I could read of the docs / wiki available, I could not find anything (e.g., env variable) that could be used to get a bit more of informations at runtime. I'm a bit relunctant to try a debug build, considering how slow things are already with a release build, but if that can provide useful feedback, I'd be glad to give it a try.

doe300, 2018-05-24:
> I don't see that much of clang running, but I did spot some process llvm-spirv /tmp/vc4c-EgnXeW /dev/stdin being ran. The file /tmp/vc4c-EgnXeW seems to be non-zero size. But no error when running, so I don't know if there's something going on.

VC4CL is using `clang` to convert OpenCL to LLVM-IR and in your configuration also `llvm-spirv` for LLVM-IR to SPIR-V conversion. The remaining compilation is done in VC4C itself, so no sub-processes will shown for this step.

> The OpenCL kernels might be big (too big for the current limitations? I'm not sure how to check that), and/or the project might be still too young?

> As a comparison, we are able to run with the same stack on the Intel Neo driver on my laptop (i7-8650U) using the GPU. The first run of the intel driver does compile the OpenCL code and can cache it on-disk, and this ~220 secs to compile.

220 secs on a fully powered machine is very long and will be much longer on a weak Raspbery Pi. If it will take that long, I don't know (see below).

> As much as I could read of the docs / wiki available, I could not find anything (e.g., env variable) that could be used to get a bit more of informations at runtime.

No, there currently is no such option. If you have the OpenCL source-code (or any intermediate version) as a file (e.g. the `/tmp/vc4c-xxxxxx` file), you can run the standalone `VC4C` compiler and generate a ton of debug information. But built in into the VC4CL runtime, it will stay silent (except for errors).

Using VC4CL there are 2 things that could take very long:

1. Trying to compile a very large kernel with VC4C
2. Executing an erroneous kernel on the GPU. This hangs the GPU and will result in a time-out. If the error is not caught, it will time out every single kernel work-group, which can be a long time.

To further analyse what goes wrong, there are a few options:
1. can be analyzed with trying to compile the kernels/intermediate code with standalone `VC4C`.
2. can be analyzed with running the additional tool provided in the VC4CL project `v3d_profile`, which will show the GPU usage. If it is high then either the GPU is really doing work, just enough that it takes very long or the GPU hangs somewhere, which is more likely.

If you can point me to the kernel files or send me the intermediate compilation results (`/tmp/vc4c-xxxxxx` files), I can also take a look at the problem.

lissyx, 2018-05-24:
Thanks for the quick reply @doe300. I'm not sure if I can get my hands onto the OpenCL source code. Is there any way to dump it somehow? This way I could share them for sure with you. I don't seem to be able to get any new file `/tmp/vc4c-xxx`.

I'm also giving a try to very small model (protocobuffer file is 54k), but so far it got blocked at the same level. Latest TensorFlow's logging trace is `Instantiating kernel for node: bidirectional_rnn/fw/basic_lstm_cell/bias = Const[dtype=DT_FLOAT, value=Tensor<type: float shape: [1976] values: 0.000547592354 -0.0        0127190584 -0.00106085732...>, _device="/job:localhost/replica:0/task:0/device:SYCL:0"]()`, if it can be any hint.

I'm going to have a look at `v3d_profile`, see if it helps getting an insight :-)

lissyx, 2018-05-24:
Rebuilding `vc4cl` with `-DBUILD_DEBUG=ON`, this is where I'm getting.

```
$ sudo TF_CPP_MIN_VLOG_LEVEL=1 ./deepspeech ~/tmp/deepspeech/models/output_graph-n_hidden_1.pb ~/tmp/deepspeech/models/alphabet.txt ~/tmp/deepspeech/audio/ -t 2>&1 | grep VC4
[VC4CL] get extension function address: clIcdGetPlatformIDsKHR
[VC4CL] get extension function address: clGetPlatformInfo
[VC4CL] base=0x3fc00000, mem=0x76fc4000
[VC4CL] V3D base: 0x76fc4000
[VC4CL] Tracking live-time of object: cl_context
[VC4CL] Starting queue handler thread...
[VC4CL] Tracking live-time of object: cl_command_queue
[VC4CL] Mailbox file descriptor opened: 4
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x00000001
[VC4CL] 0018: 0x00000004
[VC4CL] 001c: 0x01528be8
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00030012
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x00000000
[VC4CL] 0018: 0x00000004
[VC4CL] 001c: 0x01528be8
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000000
[VC4CL] 0014: 0x00000081
[VC4CL] 0018: 0x7fffffff
[VC4CL] 001c: 0x01528be8
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000008
[VC4CL] 0014: 0x3b400000
[VC4CL] 0018: 0x04c00000
[VC4CL] 001c: 0x01528be8
[VC4CL] Mailbox request: succeeded
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000000
[VC4CL] 0014: 0x6dbb6d33
[VC4CL] 0018: 0x04c00000
[VC4CL] 001c: 0x7e806e5c
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000008
[VC4CL] 0014: 0x3b400000
[VC4CL] 0018: 0x04c00000
[VC4CL] 001c: 0x7e806e5c
[VC4CL] Mailbox request: succeeded
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000000
[VC4CL] 0014: 0x6dbb6d33
[VC4CL] 0018: 0x00000001
[VC4CL] 001c: 0x00000000
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x00010006
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000008
[VC4CL] 0014: 0x3b400000
[VC4CL] 0018: 0x04c00000
[VC4CL] 001c: 0x00000000
[VC4CL] Mailbox request: succeeded
[VC4CL] Tracking live-time of object: cl_mem
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000024
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x0003000c
[VC4CL] 000c: 0x0000000c
[VC4CL] 0010: 0x0000000c
[VC4CL] 0014: 0x00000030
[VC4CL] 0018: 0x00001000
[VC4CL] 001c: 0x0000000c
[VC4CL] 0020: 0x01528be8
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000024
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x0003000c
[VC4CL] 000c: 0x0000000c
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0x0000000b
[VC4CL] 0018: 0x00001000
[VC4CL] 001c: 0x0000000c
[VC4CL] 0020: 0x01528be8
[VC4CL] Mailbox buffer before:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x00000000
[VC4CL] 0008: 0x0003000d
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x00000004
[VC4CL] 0014: 0x0000000b
[VC4CL] 0018: 0x01528be8
[VC4CL] 001c: 0x0000000b
[VC4CL] Mailbox buffer after:
[VC4CL] 0000: 0x00000020
[VC4CL] 0004: 0x80000000
[VC4CL] 0008: 0x0003000d
[VC4CL] 000c: 0x00000008
[VC4CL] 0010: 0x80000004
[VC4CL] 0014: 0xbeae9000
[VC4CL] 0018: 0x01528be8
[VC4CL] 001c: 0x0000000b
[VC4CL] base=0x3eae9000, mem=0x76fc3000
[VC4CL] Allocated 48 bytes of buffer: handle 11, device address 3199111168, host address 0x76fc3000
[VC4CL] Tracking live-time of object: cl_event
```

doe300, 2018-05-24:
>  I don't seem to be able to get any new file `/tmp/vc4c-xxx`.

This could be, because the compilation results are cached. At least that is what your first post suggests.

> Checking with htop, I see some `VC4CL Queue Han` thread listed with some (light) activity. There are two other threads, one being the main process, taking 100% of CPU. Attaching strace on both reveals:
> 
> - main process blocked on a futex()
> - other thread constantly calling sched_yield()

The `VC4CL Queue Handler` is the thread actually communicating with the GPU. Main uses a condition variable to wait for it, but should do this without utilizing 100% CPU.

Is this log the whole log? If so, then it seems to hang before it actually executes the kernel.
Did you reboot your Raspberry Pi before testing this? Because when the GPU is screwed up once, it sometimes does not recover (except when powered off).

lissyx, 2018-05-24:
That's the full log filtered on `VC4`. I already tried rebooting several times, for the same reason you mention, because I suspected it might be the case.

doe300, 2018-05-24:
> Just to make it clear, cl_cache in my first post refers to Intel's Neo OpenCL driver's caching feature

Ah okay, I misunderstood that.

If this is all the VC4CL log, then I currently do not know where the problem is. Both likely cases, I mentioned earlier are now excluded, and it looks like more a problem in the VC4CL runtime.
For further analysis, the exact location as to where the program hangs and which OpenCL API-calls were done before will have to be analyzed.

Are there instructions I can use to build the program and debug it/test it out?

lissyx, 2018-05-24:
@doe300 I can share you binaries and data, building deepspeech with OpenCL is quite painful. I wanted to give a try to your CircleCI's binaries, but somehow there's no artifact available, at least for master branch :-(.

lissyx, 2018-05-24:
@doe300 You should have that in your mailbox (one used for your git commits) :-)

lissyx, 2018-05-24:
I've added some more debug into queue handler:
```
[VC4CL] base=0x3eaed000, mem=0x76f08000
[VC4CL] Allocated 48 bytes of buffer: handle 7, device address 3199127552, host address 0x76f08000
[VC4CL] Tracking live-time of object: cl_event
[VC4CL] PUSH event: 0x6a3fe73c
[VC4CL] POP event: 
[VC4CL] POP event: 0x6abfecf8
[VC4CL] Event updated 
[VC4CL] HAS event->action 
[VC4CL] event status: 0
[VC4CL] POP event: 
[VC4CL] POP empty event: 
[VC4CL] POP returned empty event: 
```

And then it loops somehow forever on empty queue.

doe300, 2018-05-24:
I will take a look the at the execution in the next few days. The output and your assessment of it sounds like a discrepancy between the program and the VC4CL library (or within the VC4CL library) regarding the handling of events...

lissyx, 2018-05-24:
@doe300 Thanks! It's interesting, because I do remember also having to deal with some kind of weirdness related to events and/or handling of some error cases, in the past, when I was testing with the Intel Neo driver: somehow, TensorFlow or ComputeCpp layer (I cannot remember which one) was handling exception cases in an unexpected way, and this was making the tracing / profiling of OpenCL in this stack configuration impossible.

If you have any hint on where I should poke around in vc4cl to try and find some hint about what is going on, do not hesitate :-).

lissyx, 2018-05-24:
Just found out I only ran `TestVC4C`. I'm running `TestVC4CL` as well :-)

lissyx, 2018-05-24:
Looks like `TestVC4CL` allows me to reproduce some of the issues, like `./TestVC4CL --executions` that does seem to block doing nothing.

lissyx, 2018-05-24:
Hacking deeper inside `vc4c::runProcess()`, I could somehow improve the execution of `./TestVC4CL --executions` by changing the stop condition on the while: seems like it's not able to properly detect child process exit and/or stdout/stderr being done. I'll need to check that thoroughly.

lissyx, 2018-05-24:
So, current debugging gets me trapped inside the pthread mutex on the `libComputeCpp.so` side. I'm not sure there is any thing actionnable on the VC4CL side for the moment :)

lissyx, 2018-05-24:
@doe300 So, from further investigation, it seems TensorFlow using properly ComputeCpp library to push stuff to the GPU and then it sits in `wait_and_throw()` waiting for all elements in the queue to be dealt with. More details here: https://github.com/codeplaysoftware/computecpp-sdk/issues/117#issuecomment-391791126

I will try and continue to find what's going on inside VC4CL and why does this call to `wait_and_throw()` waits forever. if you have any suggestions on where I should concentrate, I'm all ears :)

lissyx, 2018-05-25:
Not sure yet why, but disabling `eventAvailable.wait_for(lock, WAIT_DURATION);` in `src/queue_handler.cpp` gets me further ... :
```
pi@rpi3-opencl-20180518:~/deepspeech $ sudo ./deepspeech ~/tmp/deepspeech/models/output_graph-n_hidden_1.pb ~/tmp/deepspeech/models/alphabet.txt ~/tmp/deepspeech/audio/ -t 2>&1 | grep -i allocated
[VC4CL] Allocated 48 bytes of buffer: handle 67, device address 3198869504, host address 0x76f1e000
pi@rpi3-opencl-20180518:~/deepspeech sudo ./deepspeech ~/tmp/deepspeech/models/output_graph-n_hidden_1.pb ~/tmp/deepspeech/models/alphabet.txt ~/tmp/deepspeech/audio/ -t 2>&1 | grep -i allocated
[VC4CL] Allocated 48 bytes of buffer: handle 68, device address 3198865408, host address 0x76f94000
[VC4CL] Allocated 48 bytes of buffer: handle 69, device address 3198861312, host address 0x76f93000
[VC4CL] Allocated 16 bytes of buffer: handle 70, device address 3198857216, host address 0x76f92000
[VC4CL] Allocated 48 bytes of buffer: handle 71, device address 3198853120, host address 0x76f91000
[VC4CL] Allocated 48 bytes of buffer: handle 72, device address 3198849024, host address 0x76f90000
[VC4CL] Allocated 16 bytes of buffer: handle 73, device address 3198844928, host address 0x76f8f000
[VC4CL] Allocated 1984 bytes of buffer: handle 74, device address 3198836736, host address 0x76f8e000
[VC4CL] Allocated 16 bytes of buffer: handle 75, device address 3198832640, host address 0x76f8d000
[VC4CL] Allocated 16 bytes of buffer: handle 76, device address 3198828544, host address 0x76f8c000
pi@rpi3-opencl-20180518:~/deepspeech $ 
```

doe300, 2018-05-25:
In the second version, since it allocated a few buffers, it looks like it is may now be running a kernel.

> [VC4CL] Allocated 1984 bytes of buffer: handle 74, device address 3198836736, host address 0x76f8e000

This buffer size looks about the right size for kernel code

lissyx, 2018-05-25:
Yes, this was my thought as well, but as much as I can tell, nothing shows up on `v3d_profile`. It is somehow again stuck.

doe300, 2018-05-25:
Thanks for looking that deep into the issue, especially the pointer to `eventAvailable.wait_for(lock, WAIT_DURATION)` helps a lot. I will investigate this issue in depth tomorrow.

lissyx, 2018-05-25:
@doe300 Removing the `lock` there, it seems I can (randomly) get further, sometimes. I got an instance of execution where all the allocations would finish and be deallocated :)
```
$ sudo ./deepspeech ~/tmp/deepspeech/models/output_graph-n_hidden_1.pb ~/tmp/deepspeech/models/alphabet.txt ~/tmp/deepspeech/audio/ -t 2>&1 | grep -i allocated
[VC4CL] Allocated 48 bytes of buffer: handle 124, device address 3198623744, host address 0x76fa3000
[VC4CL] Allocated 48 bytes of buffer: handle 125, device address 3198619648, host address 0x76fa2000
[VC4CL] Allocated 16 bytes of buffer: handle 126, device address 3198615552, host address 0x76fa1000
[VC4CL] Allocated 48 bytes of buffer: handle 127, device address 3198611456, host address 0x76fa0000
[VC4CL] Allocated 48 bytes of buffer: handle 128, device address 3198607360, host address 0x76f9f000
[VC4CL] Allocated 16 bytes of buffer: handle 129, device address 3198603264, host address 0x76f9e000
[VC4CL] Allocated 1984 bytes of buffer: handle 130, device address 3198595072, host address 0x76f9d000
[VC4CL] Allocated 16 bytes of buffer: handle 131, device address 3198590976, host address 0x76f9c000
[VC4CL] Allocated 16 bytes of buffer: handle 132, device address 3198586880, host address 0x76f9b000
[VC4CL] Allocated 16 bytes of buffer: handle 133, device address 3198582784, host address 0x76f9a000
[VC4CL] Allocated 16 bytes of buffer: handle 134, device address 3198578688, host address 0x76f99000
[VC4CL] Allocated 16 bytes of buffer: handle 135, device address 3198574592, host address 0x6cbce000
[VC4CL] Allocated 16 bytes of buffer: handle 136, device address 3198570496, host address 0x6cbcd000
[VC4CL] Allocated 16 bytes of buffer: handle 137, device address 3198566400, host address 0x6cbcc000
[VC4CL] Allocated 16 bytes of buffer: handle 138, device address 3198562304, host address 0x6cbcb000
[VC4CL] Allocated 128 bytes of buffer: handle 139, device address 3198558208, host address 0x6cbca000
[VC4CL] Allocated 128 bytes of buffer: handle 140, device address 3198554112, host address 0x6cbc9000
[VC4CL] Allocated 16 bytes of buffer: handle 141, device address 3198550016, host address 0x6cbc8000
[VC4CL] Deallocated 48 bytes of buffer: handle 124, device address 3198623744, host address 0x76fa3000
[VC4CL] Deallocated 48 bytes of buffer: handle 125, device address 3198619648, host address 0x76fa2000
[VC4CL] Deallocated 16 bytes of buffer: handle 126, device address 3198615552, host address 0x76fa1000
[VC4CL] Deallocated 48 bytes of buffer: handle 127, device address 3198611456, host address 0x76fa0000
[VC4CL] Deallocated 48 bytes of buffer: handle 128, device address 3198607360, host address 0x76f9f000
[VC4CL] Deallocated 16 bytes of buffer: handle 129, device address 3198603264, host address 0x76f9e000
[VC4CL] Deallocated 1984 bytes of buffer: handle 130, device address 3198595072, host address 0x76f9d000
[VC4CL] Deallocated 16 bytes of buffer: handle 131, device address 3198590976, host address 0x76f9c000
[VC4CL] Deallocated 16 bytes of buffer: handle 132, device address 3198586880, host address 0x76f9b000
[VC4CL] Deallocated 16 bytes of buffer: handle 133, device address 3198582784, host address 0x76f9a000
[VC4CL] Deallocated 16 bytes of buffer: handle 134, device address 3198578688, host address 0x76f99000
[VC4CL] Deallocated 16 bytes of buffer: handle 135, device address 3198574592, host address 0x6cbce000
[VC4CL] Deallocated 16 bytes of buffer: handle 136, device address 3198570496, host address 0x6cbcd000
[VC4CL] Deallocated 16 bytes of buffer: handle 137, device address 3198566400, host address 0x6cbcc000
[VC4CL] Deallocated 16 bytes of buffer: handle 138, device address 3198562304, host address 0x6cbcb000
[VC4CL] Deallocated 128 bytes of buffer: handle 139, device address 3198558208, host address 0x6cbca000
[VC4CL] Deallocated 128 bytes of buffer: handle 140, device address 3198554112, host address 0x6cbc9000
[VC4CL] Deallocated 16 bytes of buffer: handle 141, device address 3198550016, host address 0x6cbc8000
[VC4CL] Allocated 48 bytes of buffer: handle 141, device address 3198623744, host address 0x76fa3000
```

lissyx, 2018-05-25:
@doe300 Complement to my previous answer: somehow, after killing some locks (MUCH BAD) and mutexes (BAD), I've got one run to complete. It failed, though, but that (seems) to be not because of vc4cl. Looks like there are some deadlock issues to address though :(

lissyx, 2018-05-25:
Out of the locking issues, I have been able to catch some tensorflow-level errors, and now I'm hitting a vc4c compilation error, which is kind of good :) :
```
[E] Fri May 25 15:41:17 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xb1 [0x6dc37b96]
[E] Fri May 25 15:41:17 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::spirv2qasm::SPIRVParser::parse(vc4c::Module&)+0x3c1 [0x6dddbf4c]
[E] Fri May 25 15:41:17 2018:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::convert()+0xbb [0x6dc38cf8]
[E] Fri May 25 15:41:17 2018:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x193 [0x6dc3939c]
[E] Fri May 25 15:41:17 2018:  (8) /home/pi/deepspeech/libComputeCpp.so : cl::sycl::detail::program::build_current_program(std::string, bool)+0x19f [0x70b8f54c]
[E] Fri May 25 15:41:17 2018:  (9) /home/pi/deepspeech/libComputeCpp.so : cl::sycl::detail::program::build(unsigned char const*, unsigned int, std::string, bool)+0x47 [0x70b8f7c8]
[E] Fri May 25 15:41:17 2018:  (10) /home/pi/deepspeech/libComputeCpp.so : cl::sycl::detail::context::create_program_for_binary(std::shared_ptr<cl::sycl::detail::context> const&, unsigned char const*, int, bool)+0x11b [0x70b76310]
[E] Fri May 25 15:41:17 2018:  (11) /home/pi/deepspeech/libComputeCpp.so : cl::sycl::program::create_program_for_kernel_impl(std::string, unsigned char const*, int, char const* const*, std::shared_ptr<cl::sycl::detail::context>, bool)+0x87 [0x70b8a4ec]
[E] Fri May 25 15:41:17 2018:  (12) /home/pi/deepspeech/libdeepspeech.so : +0x1357564 [0x71f3c564]
[E] Fri May 25 15:41:17 2018:  (13) /home/pi/deepspeech/libdeepspeech.so : +0x13569e8 [0x71f3b9e8]
[E] Fri May 25 15:41:17 2018:  (14) /home/pi/deepspeech/libdeepspeech.so : +0x1354830 [0x71f39830]
[E] Fri May 25 15:41:17 2018:  (15) /home/pi/deepspeech/libdeepspeech.so : +0x135453c [0x71f3953c]
[E] Fri May 25 15:41:17 2018:  (16) /home/pi/deepspeech/libdeepspeech.so : +0x1354184 [0x71f39184]
[E] Fri May 25 15:41:17 2018:  (17) /home/pi/deepspeech/libdeepspeech.so : +0x1353ee4 [0x71f38ee4]
[E] Fri May 25 15:41:17 2018:  (18) /home/pi/deepspeech/libdeepspeech.so : +0x1353d44 [0x71f38d44]
[E] Fri May 25 15:41:17 2018:  (19) /home/pi/deepspeech/libdeepspeech.so : +0x118c7c4 [0x71d717c4]
[E] Fri May 25 15:41:17 2018:  (20) /home/pi/deepspeech/libdeepspeech.so : +0x3fc8d34 [0x74badd34]
[E] Fri May 25 15:41:17 2018:  (21) /home/pi/deepspeech/libdeepspeech.so : +0x3fc9b6c [0x74baeb6c]
[E] Fri May 25 15:41:17 2018:  (22) /home/pi/deepspeech/libdeepspeech.so : +0x4112a88 [0x74cf7a88]
[E] Fri May 25 15:41:17 2018:  (23) /home/pi/deepspeech/libdeepspeech.so : +0x4112034 [0x74cf7034]
[E] Fri May 25 15:41:17 2018:  (24) /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 : +0x9cd44 [0x704bed44]
[E] Fri May 25 15:41:17 2018: Compiler threw exception: Parser: Unsupported operation: 

2018-05-25 15:41:17.393737: I ./tensorflow/core/framework/log_memory.cc:35] __LOG_MEMORY__ MemoryLogTensorDeallocation { allocator_name: "device:SYCL" }
terminate called after throwing an instance of 'cl::sycl::detail::exception_implementation<(cl::sycl::detail::exception_types)7, cl::sycl::detail::exception_implementation<(cl::sycl::detail::exception_types)6, cl::sycl::exception> >'
```

lissyx, 2018-05-25:
Right, I need to find a way to dump the source code being passed `llvm-spirv`, but this will have to wait next monday.

DuncanMcBain, 2018-05-25:
The easiest thing at this stage is likely to grab it from the VC4CL side - if you look at clCreateProgramWithBinaries, there's a const char ** argument, which will be the kernels. Since you already have debugging set up, this will be easier than the alternative (which would be trying to find the correct files that are crashing, which will also all only exist in /tmp, somewhere!)

lissyx, 2018-05-25:
Thanks, I'll investigate this path. I was also on the verge of swapping `llvm-spirv`with a shell script to intercept the stdin stream 😎. But next monday 😁

doe300, 2018-05-26:
So to dump the kernel, there are a few possibilities in VC4C with only slight modifications needed:

1. If you remove the deletion of the file in the descructor in VC4C in `src/precompilation/TemporaryFile.cpp`, you will get the temporary files for every intermediate compilation steps.
2. When VC4CL is built in debug mode, it dumps The whole memory buffer for the kernel execution itself into `/tmp/vc4cl-dump.bin` just before running the first work-group. So if the program gets to this point, the output could be analysed for code which hangs the GPU.

doe300, 2018-05-26:
So I found the reason for the blocking:

- The program registers an event callback on finishing the memory copy and waits for the callback to be triggered before continuing with anything
- VC4CL did only trigger the callback if it was registered after the status for which it was registered was reached.

I looked into the OpenCL 1.2 specification which states that if the status has already been reached, the callback still needs to be triggered. I adapted the code according to that (using the same behaviour as intel/beignet and pocl)

Now the test command you sent me fails with:

    Error running session: Not found: FetchOutputs node logits_output_node: not found
    terminate called after throwing an instance of 'cl::sycl::exception'

Looks like the error is throw here: https://github.com/tensorflow/tensorflow/blob/master/tensorflow/core/graph/subgraph.cc#L134, which I don't know anything about.

lissyx, 2018-05-26:
Thanks @doe300, this error is a fallout of a missing tensorflow op in the binary I sent you. I've rebuilt libdeepspeech.so including Snapshot op, I'll send it to you 😀.

doe300, 2018-05-26:
Now, independent of the VC4C front-end used, the compilation fails with a compilation error due to the kernel using double, which is not supported (it also uses 64-bit integer which are also not supported, except in some edge cases).
This is due to the hardware being 32-bit and implementing software-base support for 64-bit types is not that easy.

mirh, 2018-05-26:
Computecpp/tensorflow should only **optionally** support fp64 and fp16. 
So any use of them is a bug of theirs afaik (or forgetfulness in OP's configuring?)

lissyx, 2018-05-26:
Thanks, that's not expected, I've built libdeepspeech.so disabling those. Maybe I messed up config somewhere, I'll have to verify that.

lissyx, 2018-05-28:
So, I've updated vc4c and vc4cl repo, and I can confirm all locking issues are now fixed :)

lissyx, 2018-05-28:
Rebuilt `VC4C` and `VC4CL` with `-D_GLIBCXX_USE_CXX11_ABI=0`, and I'm still hitting this around use of `Optional`. Is it possible i'm missing something ?:
```
[VC4CL] base=0x3e35f000, mem=0x6cd0c000
[VC4CL] Allocated 16 bytes of buffer: handle 592, device address 3191205888, host address 0x6cd0c000
[VC4CL] API call: cl_int clEnqueueWriteBuffer(cl_command_queue 0x28f5dc, cl_mem 0x65a01734, cl_bool 0, size_t 0, size_t 4, void* 0x208f60, cl_uint 0, const cl_event* 0, cl_event* 0x6abfe918)
[VC4CL] Tracking live-time of object: cl_event
[VC4CL] API call: cl_int clRetainEvent(cl_event 0x65a017e4)
[VC4CL] API call: cl_int clSetEventCallback(cl_event 0x65a017e4, cl_int 0, void(CL_CALLBACK*)(cl_event event, cl_int event_command_exec_status, void* user_data) 0x6abfec24, void* 0)
[VC4CL] API call: cl_int clFlush(cl_command_queue 0x28f5dc)
[VC4CL] API call: cl_int clGetEventInfo(cl_event 0x65a017e4, cl_event_info 4563, size_t 4, void* 0x6abfecbc, size_t* 0)
[VC4CL] API call: cl_int clGetEventInfo(cl_event 0x65a017e4, cl_event_info 4563, size_t 4, void* 0x6abfec3c, size_t* 0)
[VC4CL] API call: cl_int clReleaseEvent(cl_event 0x65a017e4)
[VC4CL] API call: cl_int clReleaseEvent(cl_event 0x65a017e4)
[VC4CL] Releasing live-time of object: cl_event
[VC4CL] API call: cl_program clCreateProgramWithBinary(cl_context 0x28eb2c, cl_uint 1, const cl_device_id* 0x65907f30, const size_t* 0x65907f60, const unsigned char** 0x65907f50, cl_int* 0x65907f40, cl_int* 0x673fe1b8)
[VC4CL] Tracking live-time of object: cl_program
[VC4CL] API call: cl_int clBuildProgram(cl_program 0x659143fc, cl_uint 1, const cl_device_id* 0x65907f30, const char* "-x spir -spir-std=1.2 ", void(CL_CALLBACK*)(cl_program program, void* user_data) 0x673fe198, void* 0)

Thread 18 "deepspeech" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x673ff3e0 (LWP 1221)]
0x705b7754 in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
(gdb) bt
#0  0x705b7754 in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#1  0x6e2d0c7e in std::experimental::fundamentals_v1::_Optional_base<std::string, true>::_Optional_base<std::string const&> (this=0x673fddbc) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include/c++/7.2.1/experimental/optional:225
#2  0x6e2ce43c in std::experimental::fundamentals_v1::optional<std::string>::optional<std::string const&, true> (this=0x673fddbc, __t=<error reading variable: Cannot access memory at address 0x0>)
    at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/arm-linux-gnueabihf/include/c++/7.2.1/experimental/optional:530
#3  0x6e2cb830 in vc4c::Optional<std::string>::Optional (this=0x673fddbc, value=<error reading variable: Cannot access memory at address 0x0>) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4C/include/Optional.h:38
#4  0x6dea8eb4 in vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)4>::PrecompilationResult (this=0x673fddbc, file=<error reading variable: Cannot access memory at address 0x0>) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4C/src/precompilation/FrontendCompiler.h:33
#5  0x6dea76b4 in vc4c::Precompiler::linkSourceCode (inputs=std::unordered_map with 1 elements = {...}, output=..., includeStandardLibrary=true) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4C/src/precompilation/Precompiler.cpp:222
#6  0x6e2c53c0 in link_programs (program=0x659143f0, otherPrograms=std::vector of length 0, capacity 0, includeStandardLibrary=true) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4CL/src/Program.cpp:177
#7  0x6e2c5c54 in vc4cl::Program::link (this=0x659143f0, options="-x spir -spir-std=1.2 ", callback=0x0, userData=0x0, programs=std::vector of length 0, capacity 0) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4CL/src/Program.cpp:296
#8  0x6e2c83aa in VC4CL_clBuildProgram (program=0x659143fc, num_devices=1, device_list=0x65907f30, options=0x6590117c "-x spir -spir-std=1.2 ", pfn_notify=0x0, user_data=0x0) at /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/VC4CL/src/Program.cpp:914
#9  0x70c6b54c in cl::sycl::detail::program::build_current_program(std::string, bool) () from /home/pi/deepspeech/libComputeCpp.so
#10 0x70c6b7c8 in cl::sycl::detail::program::build(unsigned char const*, unsigned int, std::string, bool) () from /home/pi/deepspeech/libComputeCpp.so
#11 0x70c52310 in cl::sycl::detail::context::create_program_for_binary(std::shared_ptr<cl::sycl::detail::context> const&, unsigned char const*, int, bool) () from /home/pi/deepspeech/libComputeCpp.so
#12 0x70c664ec in cl::sycl::program::create_program_for_kernel_impl(std::string, unsigned char const*, int, char const* const*, std::shared_ptr<cl::sycl::detail::context>, bool) () from /home/pi/deepspeech/libComputeCpp.so
#13 0x71ff3720 in cl::sycl::program cl::sycl::program::create_program_for_kernel<Eigen::TensorSycl::ExecExprFunctorKernel<Eigen::TensorAssignOp<Eigen::TensorMap<Eigen::Tensor<long long, 1, 1, int>, 16, Eigen::MakePointer>, Eigen::TensorConversionOp<long long, Eigen::TensorMap<Eigen::Tensor<int const, 1, 1, int>, 16, Eigen::MakePointer> const> const> const, utility::tuple::Tuple<utility::tuple::Tuple<Eigen::DSizes<int, 1> >, utility::tuple::Tuple<utility::tuple::Tuple<Eigen::DSiz [... cut ...]
#23 0x74d2a168 in Eigen::NonBlockingThreadPoolTempl<tensorflow::thread::EigenEnvironment>::WorkerLoop(int) () from /home/pi/deepspeech/libdeepspeech.so
#24 0x74d29744 in tensorflow::thread::EigenEnvironment::CreateThread(std::function<void ()>)::{lambda()#1}::operator()() const () from /home/pi/deepspeech/libdeepspeech.so
#25 0x7059c9dc in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#26 0x7041dfc4 in start_thread (arg=0x673ff3e0) at pthread_create.c:335
#27 0x76f3ebc8 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:76 from /lib/arm-linux-gnueabihf/libc.so.6
```

lissyx, 2018-05-28:
Okay, maybe the segfault was unrelated to C++11 String ABI. With those changes, I'm getting further:
```
diff --git a/src/precompilation/Precompiler.cpp b/src/precompilation/Precompiler.cpp
index 85c3f95..ed2425b 100644
--- a/src/precompilation/Precompiler.cpp
+++ b/src/precompilation/Precompiler.cpp
@@ -218,7 +218,7 @@ SourceType Precompiler::linkSourceCode(const std::unordered_map<std::istream*, O
 
         if(includeStandardLibrary)
         {
-            tempFiles.emplace_back();
+            tempFiles.emplace_back(new TemporaryFile());
             SPIRVResult stdLib(tempFiles.back()->fileName);
             compileLLVMToSPIRV(LLVMIRSource(VC4CL_STDLIB_MODULE ""), "", stdLib);
             sources.emplace_back(stdLib);
diff --git a/src/spirv/SPIRVHelper.cpp b/src/spirv/SPIRVHelper.cpp
index 06bd629..622b3a3 100644
--- a/src/spirv/SPIRVHelper.cpp
+++ b/src/spirv/SPIRVHelper.cpp
@@ -754,7 +754,7 @@ std::vector<uint32_t> spirv2qasm::readStreamOfWords(std::istream* in)
 
 void spirv2qasm::linkSPIRVModules(const std::vector<std::istream*>& inputModules, std::ostream& output)
 {
-#ifndef SPIRV_LINKER_HEADER
+#ifndef SPIRV_FRONTEND
     throw CompilationError(CompilationStep::LINKER, "SPIRV-Tools linker is not available!");
 #else
     std::vector<std::vector<uint32_t>> binaries;
```

Now, I'm hitting some new error, could this be what you referred to @doe300 about using double?
```
[VC4CL] Compiling source with: -x spir -spir-std=1.2 
[VC4CL] Compilation error: Parser: Unsupported operation: 
[VC4CL] Compilation complete with status: -11
[VC4CL] Compilation log: [E] Mon May 28 10:35:01 2018: Met unsupported instruction-decoration 11
```

lissyx, 2018-05-28:
I'm wondering if it's not exploding from `case spv::Decoration::FPRoundingMode` in `src/spirv/SPIRVParser.cpp`

doe300, 2018-05-28:
> Okay, maybe the segfault was unrelated to C++11 String ABI

Just as a note, idk if it makes a difference: VC4C and VC4CL is built with C++14

> Now, I'm hitting some new error, could this be what you referred to @doe300 about using double?

No its not, this is an error I thought I fixed. Looking [here](https://github.com/KhronosGroup/SPIRV-Headers/blob/master/include/spirv/unified1/spirv.hpp11#L356) you can see that the decoration 11 is `BuiltIn` which should never be set for any value, unless (and that seems to be the problem) the VC4CL standard-library implementation is not linked in beforehand.

I remember now that I tested the latest fixes only for linking with the `llvm-link` command and not for linking with the `SPIRV-Tools` linker (which you use). Given the line you changed, the code could not have worked before and it seems it still does not work. I will have to investigate this further.

lissyx, 2018-05-28:
Thanks @doe300 ! I'm also getting some run instances where it fails with `[VC4CL] Compilation error: Parser: Unsupported operation: OpControlBarrier`

doe300, 2018-05-28:
> [VC4CL] Compilation error: Parser: Unsupported operation: OpControlBarrier

Same issue. The `barrier()` OpenCL C standard-library function is implemented in the VC4CL standard-library implementation with VC4 custom behavior. If this implementation is not available/not linked in, the SPIR-V compiler converts the call to `barrier()` to the `OpControlBarrier` operation, which is not supported by VC4C SPIR-V front-end.

doe300, 2018-05-28:
You could try to re-build VC4C to use the LLVM IR front-end (by disabling `SPIRV_FRONTEND`), which should link in the std-lib implementation correctly.

lissyx, 2018-05-28:
> You could try to re-build VC4C to use the LLVM IR front-end (by disabling SPIRV_FRONTEND), which should link in the std-lib implementation correctly.

Sure, but does it means I've made some mistake at setup or build time?

doe300, 2018-05-28:
No, the problem that the linking of the VC4CL std-lib implementation does not work is probably an issue within VC4CL or maybe an interoperability-issue between the way VC4CL works and the SPIR-V compiler.

> -#ifndef SPIRV_LINKER_HEADER
+#ifndef SPIRV_FRONTEND

Did you change this because it threw an error with the other macro? If so, then it might be a configuration/system setup issue.

lissyx, 2018-05-28:
Yes, I changed it because it would be hit. Looking at the git log I saw all occurrences of SPIRV_LINKER_HEADER were removed in favor of SPIRV_FRONTEND so I took my chance. I'll check again my configuration and build step then. And try the other front-end in case it's still not good 😀

lissyx, 2018-05-28:
Trying to verify that link stage, I'm getting that:
```
compileLLVMToSPIRV(): /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc => /tmp/vc4c-sNG5JQ
void vc4c::precompilation::linkSPIRVModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)4> >&&, const string&, vc4c::precompilation::SPIRVResult&):269 adding one source:
void vc4c::precompilation::linkSPIRVModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)4> >&&, const string&, vc4c::precompilation::SPIRVResult&):272 => source.file:/tmp/vc4c-yJLazn
void vc4c::precompilation::linkSPIRVModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)4> >&&, const string&, vc4c::precompilation::SPIRVResult&):269 adding one source:
void vc4c::precompilation::linkSPIRVModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)4> >&&, const string&, vc4c::precompilation::SPIRVResult&):272 => source.file:/tmp/vc4c-sNG5JQ
```

I'm a bit doubtful on the "compiteLLVMToSPIRV()" for the stdlib there, because of the sizes:
```
$ sudo ls -hal /usr/local/include/vc4cl-stdlib/VC4CLStdLib.* /tmp/vc4c-sNG5JQ
-rw------- 1 root root  244 May 28 13:11 /tmp/vc4c-sNG5JQ
-rw-r--r-- 1 root root 1.4K May 28 12:46 /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc
-rw-r--r-- 1 root root  670 May 15 11:31 /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
-rw-r--r-- 1 root root  13M May 28 12:42 /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch
```

lissyx, 2018-05-28:
And this is the postinst script that produces the `.bc`:
```
$ cat /var/lib/dpkg/info/vc4c.postinst 
#!/bin/sh
set -e
cd /usr/local/include/vc4cl-stdlib
/opt/llvm-spirv/bin//clang -cc1 -triple spir-unknown-unknown -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-all -Wno-gcc-compat -x cl -emit-pch -o /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
/opt/llvm-spirv/bin//clang -cc1 -triple spir-unknown-unknown -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-all -Wno-gcc-compat -x cl -emit-llvm-bc -o /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
echo "VC4CL standard library precompiled into /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch and /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc"
```

lissyx, 2018-05-28:
@doe300 So, it would looks like the `.pch` file is being properly generated. I'm a bit unsure about the `VC4CLStdLib.bc` one, it seems very small. And the output of `/opt/llvm-spirv/bin//llvm-spirv -o /tmp/vc4c-stdlib /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc ` is even more suspect (244 bytes).

lissyx, 2018-05-28:
@doe300 Are you referring to the `.bc` file (1.4K) or to the `/tmp/vc4c-*` one (244 bytes) ?

doe300, 2018-05-28:
Both. 1.4K is still very small for all OpenCL C standard-library functions.

lissyx, 2018-05-28:
> Both. 1.4K is still very small for all OpenCL C standard-library functions.

What would be a good ballpark value I should expect ?

lissyx, 2018-05-28:
Trying to disable all LLVM optimizations:
```
sudo /opt/llvm-spirv/bin//clang -cc1 -v -triple spir-unknown-unknown -O0 -ffp-contract=off -femit-all-decls -disable-llvm-optzns -cl-opt-disable -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-all -Wno-gcc-compat -x cl -emit-llvm-bc -o /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
```

This produces a `.bc` file which is 1.8MB, that looks better.

But, then, trying to use `llvm-spirv`, it fails referring to OpenCL 2.0:
```
/opt/llvm-spirv/bin//llvm-spirv -o /tmp/vc4c-stdlib /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc
llvm-spirv: /home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/LLVM-SPIRV/llvm/include/llvm/Support/Casting.h:237: typename llvm::cast_retty<X, Y*>::ret_type llvm::cast(Y*) [with X = llvm::ConstantInt; Y = llvm::Value; typename llvm::cast_retty<X, Y*>::ret_type = llvm::ConstantInt*]: Assertion `isa<X>(Val) && "cast<Ty>() argument of incompatible type!"' failed.
Stack dump:
0.	Program arguments: /opt/llvm-spirv/bin//llvm-spirv -o /tmp/vc4c-stdlib /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc 
1.	Running pass 'Transform OCL 2.0 to SPIR-V' on module '/usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc'.
Aborted
```

doe300, 2018-05-28:
> What would be a good ballpark value I should expect ?

So on my development machine with clang 5.0.2 I got 2.1MB with the default command used by VC4C

lissyx, 2018-05-28:
@doe300 I've used the documented LLVM: https://github.com/KhronosGroup/SPIR/tree/spirv-1.0 is that wrong ? The wiki does not document more: https://github.com/doe300/VC4CL/wiki/CrossCompile

doe300, 2018-05-28:
No its not wrong, I can reproduce the issue, I also get a 1.4KB file with my SPIRV-LLVM.

> But then the llvm-spirv compiler I cross-built as documented in LLVM's doc fails:

You compiled the VC4CL standard-library implementation with clang-5.0 and then you use it with SPIRV-LLVM? AFAIK, this cannot work, since the module is an internal LLVM representation which changes with the LLVM versions.

lissyx, 2018-05-28:
Yes I did. I was kind-of expecting that. Retrying with clang-3.9 now

lissyx, 2018-05-28:
So, clang-3.9 + documented LLVM-SPIRV results in:
```
$ /opt/llvm-spirv/bin//llvm-spirv -o /tmp/vc4c-stdlib /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc
error: Unknown attribute kind (48)
```

Seems like the `LLVM-SPIRV` is clang-3.6, likely bytecode is incompatible as well? :'(

doe300, 2018-05-28:
SPIRV-LLVM is based on clang 3.6.1. And the modules are probably even binary incompatible with clang 3.6.

Is using the LLVM IR front-end an option or do you have to use the SPIR-V front-end (e.g. because the kernels are in SPIR-V code)?

lissyx, 2018-05-28:
@doe300 Well, so far I've configured everything with the SPIR-V frontend because that was the one recommended. It seems some work to switch and cross-compile with LLVM IR front-end, but if that's the only solution :-(

doe300, 2018-05-28:
Oh, the recommendation needs to be updated;)
Some time ago I managed to speed up compilation using default clang and the LLVM library front-end resulting is similar speed as the SPIR-V way with actually less set up (no need to compile SPIRV-LLVM).

lissyx, 2018-05-28:
Right, any special trick? I'm doing that:
```
LC_ALL=C cmake -DBUILD_DEBUG=ON -DMULTI_THREADED=ON -DVERIFY_OUTPUT=ON -DLLVMLIB_FRONTEND=OFF -DLLVMIR_FRONTEND=OON -DSPIRV_FRONTEND=OFF -DSPIRV_COMPILER_ROOT=/dev/null/ -DVC4CL_STDLIB_HEADER_SOURCE=/usr/local/include/vc4cl-stdlib/VC4CLStdLib.h -DCROSS_COMPILE=ON -DSYSROOT_CROSS=/home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/raspbian-stretch/ -DCROSS_COMPILER_PATH=/home/alex/codaz/Mozilla/DeepSpeech/RPI3-GPU/gcc-linaro-7.2.1-2017.11-x86_64_arm-linux-gnueabihf/bin/ -DCROSS_COMPILER_PREFIX=arm-linux-gnueabihf- -DBUILD_DEB_PACKAGE=ON ../../VC4C/ 
```

doe300, 2018-05-28:
You will need to set `-DLLVMLIB_FRONTEND=ON`, `-DLLVMIR_FRONTEND` can be disabled, since it is no longer maintained and painfully slow (imagine textual parsing of LLVM IR code).
The rest should be good.
And you will need to have `llvm-3.9-dev` (or any llvm-xx-dev as long it is the same version) installed on both machines.

[44 more replies]