doe300/VC4CL #33: TestVC4C & TestVC4CL Error

JelixLi, 2018-06-07:
Hello!  I have followed the README file to build the VC4CLStdLib, & VC4C &  VC4CL. And I have got something really confusing. 
I didn't make any change of the CMakeList.txt (in the top dir) but just turned off the Multi-Thread option in the VC4C.
And then I run 
`cmake . `
`make`
`make install`

And then 
`sudo ./build/test/TestVC4C `

I got the output below 
<pre><code>
./example/fft2_2.cl
[E] Thu Jun  7 11:01:32 2018: Errors in precompilation:
[E] Thu Jun  7 11:01:32 2018: Expected no forward declarations!
!3 = <temporary!> !{}
incorrect number of operands in llvm.ident metadata
!0 = !{i32 0, i32 1}
/usr/bin/llvm-link: /tmp/vc4c-KQBI2v: error: input module is broken!

[E] Thu Jun  7 11:01:32 2018:  (1) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x76b776a8]
[E] Thu Jun  7 11:01:32 2018:  (2) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : +0x6a27d0 [0x76e017d0]
[E] Thu Jun  7 11:01:32 2018:  (3) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::precompilation::linkLLVMModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>, std::allocator<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3> > >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x364 [0x76e02770]
[E] Thu Jun  7 11:01:32 2018:  (4) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::precompilation::linkInStdlibModule(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x78 [0x76e01f9c]
[E] Thu Jun  7 11:01:32 2018:  (5) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), void (*)(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x60 [0x76e04ef0]
[E] Thu Jun  7 11:01:32 2018:  (6) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76e03fb0]
[E] Thu Jun  7 11:01:32 2018:  (7) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0xd0 [0x76e02e9c]
[E] Thu Jun  7 11:01:32 2018:  (8) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x5c [0x76e05130]
[E] Thu Jun  7 11:01:32 2018:  (9) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76e03d00]
[E] Thu Jun  7 11:01:32 2018:  (10) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x4ac [0x76e09af4]
[E] Thu Jun  7 11:01:32 2018:  (11) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x158 [0x76e07a80]
[E] Thu Jun  7 11:01:32 2018:  (12) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0xc4 [0x76b79898]
[... cut ...]
[E] Thu Jun  7 11:01:56 2018:  (11) /home/pi/TestOpenCL/VC4C/build/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x158 [0x76e07a80]
[E] Thu Jun  7 11:01:56 2018:

JelixLi, 2018-06-07:
Does anyone have suggestions? Thanks

doe300, 2018-06-07:
The fact that you get errors at all is expected. `TestVC4C` runs about 200+ test-cases, of which some if them fail.

The actual error (`error: input module is broken!`) is strange and stems from the LLVM linker.

What versions of CLang/LLVM are you using?

JelixLi, 2018-06-07:
the version 3.9

just like this 
<pre><code>-- CLang compiler found: /usr/bin/clang-3.9
-- Using enhanced OpenCL features of Clang 3.9+!
-- LLVM-link found: /usr/bin/llvm-link
-- Compiling LLVM library front-end with LLVM in version 3.9.1 located in '/usr/lib/llvm-3.9/lib'
-- found clang-format: /usr/bin/clang-format
-- build deb package...
-- Debian package expects VC4CL standard library PCH to be located in: /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/TestOpenCL/VC4C</pre></code>

JelixLi, 2018-06-07:
`./build/test/TestVC4CL: symbol lookup error: ./build/test/TestVC4CL: undefined symbol: clGetPlatformIDs`
And what about this ? 
(Sorry , I really can't understand )

doe300, 2018-06-07:
By default, VC4CL is built to support the Khronos ICD loader.
So to make it work, you will either need to set up an ICD loader file (see [here](https://github.com/doe300/VC4CL#khronos-icd-loader)) or build without ICD loader support (by disabling `BUILD_ICD`)

JelixLi, 2018-06-07:
I have disabled `BUILD_ICD`
and I got messages below, does it ok?
<pre><code>Test 'TestSystem::testGetSystemInfo()' failed!
	Suite: TestSystem
	File: TestSystem.cpp
	Line: 28
	Failure: Got 11, expected 2
Test 'TestSystem::testGetSystemInfo()' failed!
	Suite: TestSystem
	File: TestSystem.cpp
	Line: 30
	Failure: Got 14, expected 4
Test 'TestSystem::testGetSystemInfo()' failed!
	Suite: TestSystem
	File: TestSystem.cpp
	Line: 32
	Failure: Got 14, expected 3
Suite 'TestSystem' finished, 0/1 successful (0%) in 422 microseconds (0.422 ms).
Test 'TestProgram::testLinkProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 118
	Failure: Got -16, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 165
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 166
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 167
	Failure: Got 1831149630, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 170
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 171
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 172
	Failure: Got 0x6d25203e, expected 0x1b95e44
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 175
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 176
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 177
	Failure: Got 1831149630, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 180
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 181
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 182
	Failure: Got 0x6d25203e, expected 0x1b9878c
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 190
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 191
	Failure: Assertion 'info_size > 0u && info_size < 2048u' failed
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 198
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 199
	Failure: Got 1, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 203
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 206
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 207
	Failure: Got 1, expected 4
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 208
	Failure: Got 1831149568, expected 1
Test 'TestProgram::testGetProgramInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 211
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 142
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 143
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 144
	Failure: Got 1831149568, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 147
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 148
	Failure: Assertion 'info_size > 0u' failed
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 151
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 152
	Failure: Assertion 'info_size > 0u' failed
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 155
	Failure: Got -44, expected 0
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 156
	Failure: Got 0, expected 4
Test 'TestProgram::testGetProgramBuildInfo()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 157
	Failure: Got 1831149568, expected 4
Segmentation fault
</pre></code>

In fact, I am a totally freshman to the programming  of opencl, And it is ok that I just need to include the opencl headers  if I want to write a  gpu gemm (maybe) (with the linking of the VC4CL Lib ), right?

JelixLi, 2018-06-07:
Sorry for the bothering.

doe300, 2018-06-07:
Yes, to write OpenCL kernels, you need only the OpenCL C headers.

These errors should not be there, but I currently don't know why they appear.

What is the output of `VC4C --version`? And does `clinfo` now work?

JelixLi, 2018-06-07:
the output of `VC4C --version` 
<pre><code>
Running VC4C in version: 0.4
Build configuration:
debug mode; clang 3.9+ OpenCL features; clang in /usr/bin/clang-3.9; LLVM library front-end with libLLVM 3.9; VC4CL standard-library in /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch; vc4asm verification
</pre></code>
And the `clinfo` still doesn't  work ( this clinfo is from the git)

JelixLi, 2018-06-08:
And VC4C can't  build for Release.
<pre><code>
<command-line>:0:0: warning: "_GNU_SOURCE" redefined
<command-line>:0:0: note: this is the location of the previous definition
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp: In member function ‘void vc4c::ControlFlowGraph::dumpGraph(const string&) const’:
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:5: error: ‘DebugGraph’ was not declared in this scope
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
     ^~~~~~~~~~
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:26: error: expected primary-expression before ‘*’ token
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                          ^
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:27: error: expected primary-expression before ‘,’ token
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                           ^
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:40: error: expected primary-expression before ‘,’ token
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                                        ^
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:60: error: ‘::dumpGraph’ has not been declared
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                                                            ^~
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:88: error: expected primary-expression before ‘>’ token
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                                                                                        ^
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:90: warning: value computed is not used [-Wunused-value]
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                                                                                          ^~~~~
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:90: warning: left operand of comma operator has no effect [-Wunused-value]
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:237:97: warning: right operand of comma operator has no effect [-Wunused-value]
     DebugGraph<BasicBlock*, CFGRelation, CFGEdge::Directed>::dumpGraph<ControlFlowGraph>(*this, path, nameFunc,
                                                                                                 ^~~~
/home/pi/TestOpenCL/VC4C/src/analysis/ControlFlowGraph.cpp:241:9: warning: right operand of comma operator has no effect [-Wunused-value]
         },
         ^

</code></pre>

JelixLi, 2018-06-08:
I added  <code>#define CL_TARGET_OPENCL_VERSION 120</code> in the <code><opencl.h></code> to  to enforce usage of no more than the OpenCL 1.2 APIs
And it seems the Test works
output like this
<pre><code>Suite 'TestExecutions' finished, 13/15 successful (86.66%) in 22441148 microseconds (22441.1 ms).</code></pre>

JelixLi, 2018-06-08:
But the clinfo still doesn't work

doe300, 2018-06-08:
> I added `#define CL_TARGET_OPENCL_VERSION 120` in the `<opencl.h>`

So before it failed and with this define it works?

> But the clinfo still doesn't work

Still the ` Number of platforms 0` error? Did you pull HEAD or any release version?

JelixLi, 2018-06-08:
Yes ,it seems to work with that define. And the error <code>Number of platforms 0</code> still exists. I simply cloned the clinfo from [here](https://github.com/Oblomov/clinfo)
And I write a simply program  with just one call function but I got this error
<code>hello_cl.cpp:(.text+0x2c): undefined reference to `clGetPlatformIDs'</code>
I have installed the VC4CL

closed 2020-01-08 by doe300