doe300/VC4CL #34: Building Debian package on RPi: manual copying required

alexsee75, 2018-06-15:
When building a debian package on RPi using...

`mkdir build && cd build`

`cmake .. -DBUILD_DEBUG=ON -DBUILD_DEB_PACKAGE=ON -DBUILD_TESTING=ON -DCROSS_COMPILE=ON -DBUILD_ICD=ON -DREGISTER_POKE_KERNELS=ON -DIMAGE_SUPPORT=ON`

`make`

`cpack -G DEB`

an error is output concerning a missing VC4CLIdl.

```
Pack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: VC4CL
CPack: - Install project: VC4CL
CMake Error at /usr/local/src/VC4CL/build/cmake_install.cmake:44 (file):
  file INSTALL cannot find "/usr/local/src/VC4CL/VC4CL.icd".

CPack Error: Error when generating package: vc4cl
```

It can be fixed by copying VC4CL.icd to the parent directory

`cp VC4CL.icd ..`

At some point this should be fixed in the build files... it's a minor issue, I've posted it here for reference.

doe300, 2018-06-15:
Do you run `cp VC4CL.icd ..` from the build-directory?
I usually build directly from the project root (the build files are automatically placed in the `./build` subdirectory)

doe300, 2018-08-25:
See #39:

It is intended to build from the project root:

    cmake .
    make
    make install

The build files are automatically built into the `build/` directory and all relative paths resolve correctly.

closed 2018-08-25 by doe300