doe300/VC4CL #35: Some test cases failing for TestVC4CL (minor)

alexsee75, 2018-06-15:
Good news! I've finally managed to compile and install everything on my RPi and the tests look good...

```
Suite 'TestImage' finished, 8/10 successful (80%) in 276470 microseconds (276.47 ms).
Suite 'TestExecutions' finished, 13/15 successful (86.66%) in 100462261 microseconds (100462 ms).
```

Is this an ok result or should it be 100% for both? Are these all test suite or is something missing?

Cmake was called like this:
`cmake .. -DBUILD_DEBUG=ON -DBUILD_DEB_PACKAGE=ON -DBUILD_TESTING=ON -DCROSS_COMPILE=ON -DBUILD_ICD=ON -DREGISTER_POKE_KERNELS=ON -DIMAGE_SUPPORT=ON`

I am still getting some strange errors but since I build with `-DIMAGE_SUPPORT=ON` perhaps that is the reason. Attached you find the 4285 line log with -DDEBUG=ON, perhaps it is helpful (or not)

[TestVC4CL.log](https://github.com/doe300/VC4CL/files/2105582/TestVC4CL.log)

May I consider VC4CL properly installed and ready for doing something useful with it? (such as porting a tensor flow lite model to OpenCL via code generator?)

doe300, 2018-06-15:
> Is this an ok result or should it be 100% for both?

No, currently a few test cases fail here and there, so this is okay. Image support is very basic on host-side and non existent GPU-side, so it is expected not to work.

> May I consider VC4CL properly installed and ready for doing something useful with it?

VC4CL should install properly, but it is not ready for any productive use. More complex kernels will most likely fail to compile and easier kernels might fail due to wrong code being generated.

alexsee75, 2018-06-15:
Can you just take a quick look at clinfo output if everything looks fine?
[clinfo.log](https://github.com/doe300/VC4CL/files/2105619/clinfo.log)

alexsee75, 2018-06-15:
> VC4CL should install properly, but it is not ready for any productive use. More complex kernels will most likely fail to compile and easier kernels might fail due to wrong code being generated.
I don't mind... 20 years ago I was working with the first generator of GNU compilers that had template support, building a data mining suite.... some code parts needed to be reformulated a dozen times until they would compile without the dreaded _internal compiler error_..... so this actually brings me back to my youth.

And yours is probably the only way to run TFLite Models on RPi Zero as fast as on RPi3B+ ... without GPU support they are about 15x slower (mostly because of missing NEON instructions on ARMv6)

BTW - Excellent work! I also skimmed your master thesis - good work, hopefully you received an A+ (resp. 1+)

Greetings from Austria,
  Alex

doe300, 2018-06-15:
> Can you just take a quick look at clinfo output if everything looks fine?

That's okay. The debug output is a bit annoying in this case, but the rest looks good.

closed 2018-06-15 by alexsee75