doe300/VC4CL #36: clpeak have a lot of error:(

lygstate, 2018-06-19:
```
[W] Tue Jun 19 18:10:46 2018: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Tue Jun 19 18:10:46 2018: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Tue Jun 19 18:10:46 2018: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[E] Tue Jun 19 18:11:23 2018: Failed to in-line or intrinsify function-call: <16 x f16> %call31 = <16 x f16> madDhS_S_(<16 x f16> %splat.splat, <16 x f16> %vecinit29, <16 x f16> %splat.splat)
[W] Tue Jun 19 18:11:23 2018: Candidates:
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v1_local_offset(f32* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: i32 _Z12get_group_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z14get_local_sizej(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z12get_local_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z13get_global_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v2_local_offset(<2 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v4_local_offset(<4 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v8_local_offset(<8 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v16_local_offset(<16 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v1_global_offset(f32* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: i32 _Z15get_global_sizej(i32 %dim
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v2_global_offset(<2 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v4_global_offset(<4 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v8_global_offset(<8 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v16_global_offset(<16 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v1(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: f32 _Z3madfff(f32 %a, f32 %b, f32 %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v2(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <2 x f32> _Z3madfS_S_(<2 x f32> %a, <2 x f32> %b, <2 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v4(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <4 x f32> _Z3madfS_S_(<4 x f32> %a, <4 x f32> %b, <4 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v8(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <8 x f32> _Z3madfS_S_(<8 x f32> %a, <8 x f32> %b, <8 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v16(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <16 x f32> _Z3madfS_S_(<16 x f32> %a, <16 x f32> %b, <16 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_hp_v1(f16* %ptr, f32 %_B
[W] Tue Jun 19 18:11:23 2018: void compute_hp_v2(f16* %ptr, f32 %_B
[W] Tue Jun 19 18:11:23 2018: void compute_hp_v4(f16* %ptr, f32 %_B
[W] Tue Jun 19 18:11:23 2018: void compute_hp_v8(f16* %ptr, f32 %_B
[W] Tue Jun 19 18:11:23 2018: void compute_hp_v16(f16* %ptr, f32 %_B
[W] Tue Jun 19 18:11:23 2018: void compute_dp_v1(f64* %ptr, f64 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_dp_v2(f64* %ptr, f64 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_dp_v4(f64* %ptr, f64 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_dp_v8(f64* %ptr, f64 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_dp_v16(f64* %ptr, f64 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_integer_v1(i32* %ptr, i32 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_integer_v2(i32* %ptr, i32 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_integer_v4(i32* %ptr, i32 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_integer_v8(i32* %ptr, i32 %_A
[W] Tue Jun 19 18:11:23 2018: void compute_integer_v16(i32* %ptr, i32 %_A
[E] Tue Jun 19 18:11:23 2018: Failed to in-line or intrinsify function-call: <8 x f16> %call15 = <8 x f16> madDhS_S_(<8 x f16> %splat.splat, <8 x f16> %vecinit13, <8 x f16> %splat.splat)
[W] Tue Jun 19 18:11:23 2018: Candidates:
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v1_local_offset(f32* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: i32 _Z12get_group_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z14get_local_sizej(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z12get_local_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: i32 _Z13get_global_idj(i32 %dim
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v2_local_offset(<2 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v4_local_offset(<4 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v8_local_offset(<8 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v16_local_offset(<16 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v1_global_offset(f32* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: i32 _Z15get_global_sizej(i32 %dim
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v2_global_offset(<2 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v4_global_offset(<4 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v8_global_offset(<8 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void global_bandwidth_v16_global_offset(<16 x f32>* %A, f32* %B
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v1(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: f32 _Z3madfff(f32 %a, f32 %b, f32 %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v2(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <2 x f32> _Z3madfS_S_(<2 x f32> %a, <2 x f32> %b, <2 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v4(f32* %ptr, f32 %_A
[W] Tue Jun 19 18:11:23 2018: <4 x f32> _Z3madfS_S_(<4 x f32> %a, <4 x f32> %b, <4 x f32> %c
[W] Tue Jun 19 18:11:23 2018: void compute_sp_v8(f32* %ptr, f32 %_A
[... 327 lines cut ...]
[E] Tue Jun 19 18:11:24 2018: While running worker task: Normalization
[E] Tue Jun 19 18:11:24 2018: Background worker threw error: Normalizer: Not normalized instruction found: f16 %call2 = f16 madDhDhDh(f16 %conv1, f16 %conv, f16 %conv1)
[E] Tue Jun 19 18:11:24 2018: While running worker task: Normalization
[E] Tue Jun 19 18:11:24 2018:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x7682e5d0]
[E] Tue Jun 19 18:11:24 2018:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x677344 [0x76a57344]
[E] Tue Jun 19 18:11:24 2018:  (3) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::Module&, vc4c::Method&, vc4c::InstructionWalker, vc4c::Configuration const&), void (*)(vc4c::Module&, vc4c::Method&, vc4c::InstructionWalker, vc4c::Configuration const&)>::_M_invoke(std::_Any_data const&, vc4c::Module&, vc4c::Method&, vc4c::InstructionWalker&&, vc4c::Configuration const&)+0x70 [0x76a5c390]
[E] Tue Jun 19 18:11:24 2018:  (4) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::Module&, vc4c::Method&, vc4c::InstructionWalker, vc4c::Configuration const&)>::operator()(vc4c::Module&, vc4c::Method&, vc4c::InstructionWalker, vc4c::Configuration const&) const+0x94 [0x76a5bd00]
[E] Tue Jun 19 18:11:24 2018:  (5) /usr/local/lib/libVC4CC.so.1.2 : +0x677558 [0x76a57558]
[E] Tue Jun 19 18:11:24 2018:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::normalization::Normalizer::normalizeMethod(vc4c::Module&, vc4c::Method&) const+0x38c [0x76a5823c]
[E] Tue Jun 19 18:11:24 2018:  (7) /usr/local/lib/libVC4CC.so.1.2 : +0x6775f8 [0x76a575f8]
[E] Tue Jun 19 18:11:24 2018:  (8) /usr/local/lib/libVC4CC.so.1.2 : +0x678f74 [0x76a58f74]
[E] Tue Jun 19 18:11:24 2018:  (9) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::Method* const&)>::operator()(vc4c::Method* const&) const+0x54 [0x76834dec]
[E] Tue Jun 19 18:11:24 2018:  (10) /usr/local/lib/libVC4CC.so.1.2 : void vc4c::BackgroundWorker::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)::{lambda()#1}::operator()() const+0xc8 [0x768334ec]
[E] Tue Jun 19 18:11:24 2018:  (11) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (), void vc4c::BackgroundWorker::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, std::__cx [... cut ...]
[E] Tue Jun 19 18:11:24 2018: Background worker threw error: Normalizer: Not normalized instruction found: <16 x f16> %call31 = <16 x f16> madDhS_S_(<16 x f16> %splat.splat, <16 x f16> %vecinit29, <16 x f16> %splat.splat)
[E] Tue Jun 19 18:11:24 2018: While running worker task: Normalization
[E] Tue Jun 19 18:11:34 2018: Compiler threw exception: Normalizer: Not normalized instruction found: <8 x f16> %call15 = <8 x f16> madDhS_S_(<8 x f16> %splat.splat, <8 x f16> %vecinit13, <8 x f16> %splat.splat)

```

doe300, 2018-06-19:
With what configuration did you compile VC4C?

lygstate, 2018-06-19:
@doe300 I didn't compile VC4C, I download it through https://github.com/doe300/VC4CL/wiki/How-to-get

closed 2018-06-26 by doe300