doe300/VC4CL #37: Kernel Float  Multiplication Bug

JelixLi, 2018-06-20:
when I write code like
<code>
int i;
int j;
int k=i * j;
</code>
I can't get correct results
but if I write
<code>
int i;
int k=i * 10;
</code>
The result is correct, what's  wrong with the multiplication between two variables.

JelixLi, 2018-06-20:
And it seems that only "int"  has this problem

doe300, 2018-06-20:
The VideoCore IV hardware does not support full-width 32-bit integer multiplication, so we have to emulate it with multiple 24.bit multiplications, additions and shifts, which may be wrong in some cases. The second example (multiplication with a constant) may be optimized to generate different code.
In contrast, floating-point multiplication is supported on the VC4 and therefore should yield the correct result.

JelixLi, 2018-06-20:
All right. Thanks for your feedback !

closed 2018-06-20 by JelixLi