doe300/VC4CL #39: Compile

alexzk1, 2018-07-05:
Any fast ideas whats going on?

1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 136.308/136.308/136.308/0.000 ms
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/libOpenCL.so (found version "2.2") 
-- Found headers for OpenCL up to version 2.2 located in /usr/include
-- VC4C library found: /usr/lib/libVC4CC.so
-- VC4C compiler header found at: /usr/include/vc4cc/VC4C.h
-- Building with ICD support
-- Checking for one of the modules 'ocl-icd>=1.3'
-- Found Khronos ICD Loader in version 2.2.12 in /usr/lib
-- Enabling register-poking to run kernels
-- found clang-format: /usr/bin/clang-format
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alex/packages/vc4cl_arch/src/build
Scanning dependencies of target generate_icd
[  3%] Generating VC4CL.icd
[  3%] Built target generate_icd
Scanning dependencies of target VC4CL
[  6%] Building CXX object build/CMakeFiles/VC4CL.dir/CommandQueue.cpp.o
[ 10%] Building CXX object build/CMakeFiles/VC4CL.dir/Buffer.cpp.o
c++: fatal error: no input files
compilation terminated.
distcc[1055] ERROR: compile (null) on localhost failed
c++: fatal error: no input files
compilation terminated.
distcc[1056] ERROR: compile (null) on localhost failed
/bin/sh: -g3: command not found
make[2]: *** [build/CMakeFiles/VC4CL.dir/build.make:63: build/CMakeFiles/VC4CL.dir/Buffer.cpp.o] Error 127
make[2]: *** Waiting for unfinished jobs....
/bin/sh: -g3: command not found
make[2]: *** [build/CMakeFiles/VC4CL.dir/build.make:76: build/CMakeFiles/VC4CL.dir/CommandQueue.cpp.o] Error 127
make[1]: *** [CMakeFiles/Makefile2:183: build/CMakeFiles/VC4CL.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...

alexzk1, 2018-07-05:
I see in your code 1 mention only
if(BUILD_DEBUG)
	LIST(APPEND CMAKE_CXX_FLAGS " -g3 -Wall -Wextra -Wold-style-cast -Wno-unused-parameter -Wno-ignored-attributes -fno-strict-aliasing -pthread")
	SET(CMAKE_BUILD_TYPE 		Debug)
	add_definitions(-DDEBUG_MODE=1)
else()
	LIST(APPEND CMAKE_CXX_FLAGS " -g3 -Wall -Wextra -Wold-style-cast -Wno-unused-parameter -Wno-ignored-attributes -fno-strict-aliasing -pthread")
	SET(CMAKE_BUILD_TYPE 		Release)
endif()

Don't you mean -O3 there ? -g leaves debug info as far as I recall.

alexzk1, 2018-07-05:
Also install is bugged, does not respect build folder.

 file INSTALL cannot find
  "/home/alex/packages/vc4cl_arch/src/VC4CL/VC4CL.icd".

while it is in src/build

doe300, 2018-07-05:
Why would you build into `src/build`?

Usually you execute (from the project-root):

    cmake .
    make
    make install

And it automatically creates a `build` folder to store the build info in

alexzk1, 2018-07-05:
well, different "usually". I do parallel build folder and execute as cmake ../src
So always can delete build to make fresh compile. Just got used so.

doe300, 2018-07-05:
Okay, then more like "intended" than "usually" ;)
#34 has a very similar problem.

So it looks for the file in `<prefix>/VC4CL/VC4CL.icd` but is created into `<prefix>/build/VC4CL.icd`. Am I correct?

alexzk1, 2018-07-05:
Last question if I can ask....built and installed now. Will it start working with opencv immediatly (I use UMat, so on nvidia its ok) or need some changes to code ?

alexzk1, 2018-07-05:
As I said it looks for files in source directories, but it was build elsewhere in tree and had that file there (where Makefile is).

doe300, 2018-07-05:
In the current state, it will probably not work. There are still too many things in VC4C that do not work.
It might work, if the OpenCV does not use images and can handle limited ressources..

alexzk1, 2018-07-05:
Yeh
[W] Thu Jul  5 13:13:38 2018: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Thu Jul  5 13:13:38 2018: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[E] Thu Jul  5 13:13:40 2018: Failed to in-line or intrinsify function-call: void Lab2BGR_f(f32* %arrayidx, f32* %arrayidx20, f32* %coeffs, f32 %lThresh, f32 %fThresh)
[W] Thu Jul  5 13:13:40 2018: Candidates:
[W] Thu Jul  5 13:13:40 2018: void RGB2Gray(i8* %srcptr, i32 %src_step, i32 %src_offset, i8* %dstptr, i32 %dst_step, i32 %dst_offset, i32 %rows, i32 %cols
[W] Thu Jul  5 13:13:40 2018: i32 _Z13get_global_idj(i32 %dim
[W] Thu Jul  5 13:13:40 2018: i32 _Z5mad24iii(i32 %a, i32 %b, i32 %c

So I should not use it for images, right? But things like Sobel?

doe300, 2018-07-05:
A quick check reveals, that the "missing" function is only defined and used [here](https://github.com/opencv/opencv/blob/64916d3d83595dbe075e4494cafe0d2c4b5744a6/modules/imgproc/src/opencl/color_lab.cl#L400). I will have to investigate, why it is not found despite being defined in the same file

closed 2019-10-04 by doe300