doe300/VC4CL #44: "clinfo" can't open /dev/mem but "sudo clinfo" does not work

geoboom, 2018-09-01:
I have successfully compiled and installed VC4C, VC4CL and confirmed that both are working; when I do clinfo I get OpenCL information on my GPU and
```
[VC4CL] can't open /dev/mem
[VC4CL] This program should be run as root. Try prefixing command with: sudo
terminate called after throwing an instance of 'std::system_error'
what(): Failed to open /dev/mem: Operation not permitted
aborted
```
which is expected as running any VC4CL implementation requires root access. However, when I do "sudo clinfo", I get ``Number of platforms 0`` as output. 

Prior to the above, I created file /etc/OpenCL/vendors/VC4CL.icd which contains one line: ``/usr/local/lib/libVC4CL.so`` (I also tried libVC4CL.so as the one line but to no avail) as per instructions.

Please help me, thanks.

geoboom, 2018-09-01:
I don't think this issue is only pertinent to clinfo. When I try to run other programs that utilize OpenCL I get permission errors when not running as root (expected) but when I run as root, OpenCL is not detected.

doe300, 2018-09-01:
Maybe you also need to define the LD_LIBRARY_PATH, like in https://github.com/doe300/VC4C/issues/114 ? You may also take a lopk at https://github.com/doe300/VC4CL/issues/16#issuecomment-355947972

geoboom, 2018-09-01:
Indeed, I added ``LD_LIBRARY_PATH=/usr/local/lib`` to /etc/environment and now it works for root user. Thanks!

closed 2018-09-01 by geoboom