doe300/VC4CL #45: Errors with VC4C. How to use CPU also? Is LD_LIBRARY_PATH=/usr/local/lib/ required before programs?

gntook, 2018-10-09:
Hi.

Thank you for creating VC4VL. I am very eager to implement it on several machines.
On a RPi, I have compiled and installed VC4C and VC4CL including tools, but not test.
I do not yet have VC4VL functional. Details of three problems follow.

(1) In VC4C ./build/test/TestVC4C gave an enormity of errors! I mean page after page. I issued this command, as root, after cmake, make, and make install of both VC4CLStdLib and VC4C (but not VC4CL). After about 45min, and two and a half pages of
"64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!"
the program stopped. Another shell showed cpu usage at 100%, so after 15min, I killed the program.
Is this behavior normal?

(2) When I issue "cmake" in the directory named "test" I see,
cmake ../test
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:10 (ExternalProject_Get_Property):
  Unknown CMake command "ExternalProject_Get_Property".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.7)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/VC4C/VC4CL/test/CMakeFiles/CMakeOutput.log".

(3) clinfo shows Device 0 when mesa-opencl-icd is installed, which I presume means it is a superfluous. Please, correct me if I am wrong. I removed mesa-opencl-icd and Device 0 disappeared.
I want to do a comparison of efficacy between the GPU's and the CPU's use of OpenCL. How may I implement OpenCL support for the RPi's CPU?
The full output of clinfo is below. Note, I had to use the prefix for clinfo to work.
# LD_LIBRARY_PATH=/usr/local/lib/ clinfo
Number of platforms                               1
  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
  Platform Vendor                                 doe300
  Platform Version                                OpenCL 1.2 VC4CL 0.4
  Platform Profile                                EMBEDDED_PROFILE
  Platform Extensions                             cl_khr_il_program cl_khr_spir cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
  Platform Extensions function suffix             VC4CL

  Platform Name                                   OpenCL for the Raspberry Pi VideoCore IV GPU
Number of devices                                 1
  Device Name                                     VideoCore IV GPU
  Device Vendor                                   Broadcom
  Device Vendor ID                                0xa5c
  Device Version                                  OpenCL 1.2 VC4CL 0.4
  Driver Version                                  0.4
  Device OpenCL C Version                         OpenCL C 1.2 
  Device Type                                     GPU
  Device Profile                                  EMBEDDED_PROFILE
  Max compute units                               1
  Max clock frequency                             300MHz
  Device Partition                                (core)
    Max number of sub-devices                     0
    Supported partition types                     None
[... 57 lines cut ...]
  clCreateContext(NULL, ...) [default]            Success [VC4CL]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ACCELERATOR)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1

	How may I bring up the CPU as a second (and/or alternative) device?

	My summary questions are as follows.
(1) How may I run clinfo (and other programs) without LD_LIBRARY_PATH=/usr/local/lib/ preceding a command short of making symlinks to /usr/lib/?
(2) How may I bring up the CPU as a second (and/or alternative) device?
(3) The device in clinfo's output is not usable owing to errors such as vc4c::CompilationError::CompilationError,vc4c::precompilation::compileOpenCLWithDefaultHeader, etc. I need to know how to diagnose the underlying cause. Is ./build/test/TestVC4C supposed to give multiple pages of errors and then freeze? If so, where else can I look?

I would be very grateful for any assistance with these problems.

doe300, 2018-10-09:
For (1):
Yes, it does give a lot of errors (and runs for a long while), because it does not only contain regression tests, but it basically tests for every OpenCL C source code example I could find, where a lot of them do not compile yet. It should not hang, but it does take a long time.

For (2):
If you build the top-level `CMakeLists.txt`, the test directory will automatically be built, if you enable the `-DBUILD_TESTING=ON` cmake flag.

For (3):
>  clinfo shows Device 0 when mesa-opencl-icd is installed, which I presume means it is a superfluous. Please, correct me if I am wrong. I removed mesa-opencl-icd and Device 0 disappeared.

So you have a device 0 when mesa is installed and it disappears when mesa is uninstalled?

> How may I implement OpenCL support for the RPi's CPU?

Looks like the OpenCL support for the CPU (via mesa) is already there (if you install mesa again).

> (1) How may I run clinfo (and other programs) without LD_LIBRARY_PATH=/usr/local/lib/ preceding a command short of making symlinks to /usr/lib/?

This is a common problem for some Raspbian installations. You will need to add `/usr/local/lib` to your library path, see [here](https://stackoverflow.com/questions/4743233/is-usr-local-lib-searched-for-shared-libraries). 

> (2) How may I bring up the CPU as a second (and/or alternative) device?

See above, mesa is the implementation running on the CPU.

> (3) The device in clinfo's output is not usable owing to errors such as vc4c::CompilationError::CompilationError,vc4c::precompilation::compileOpenCLWithDefaultHeader, etc

Where do you get these error messages? What are you trying to compile while you get them?

gntook, 2018-10-09:
When running TestVC4CL I have results similar to the first output here, "TestVC4C & TestVC4CL Error #33" - https://github.com/doe300/VC4CL/issues/33
./build/test/TestVC4CL 
Test 'TestExecutions::testSFU()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 205
	Failure: Got -50, expected 0
Test 'TestExecutions::testSFU()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 212
	Failure: Got -52, expected 0
Test 'TestExecutions::testSFU()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 213
	Failure: Assertion 'event != nullptr' failed
Test 'TestExecutions::testSFU()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 215
	Failure: Got -58, expected 0
Test 'TestExecutions::testSFU()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 232
	Failure: Got -58, expected 0
Test 'TestExecutions::testWorkItems()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 325
	Failure: Got -50, expected 0
Test 'TestExecutions::testWorkItems()' failed!
	Suite: TestExecutions
	File: TestExecutions.cpp
	Line: 330
	Failure: Got -52, expected 0
..."Line" increases by 1 each time up to "Line: 455" ending with,
Suite 'TestExecutions' finished, 12/15 successful (80%) in 23737809 microseconds (23737.8 ms).

How do I run the latter tests on the link? Or am I misinterpreting what I see?

Also, could you tell me what I can/should do to compile CLang (instead of using the Raspian package) such that it gives debug information? I would prefer to create the custom .deb(s) from the edited source .deb(s), rather than the tarball, but I suppose either would do.