doe300/VC4CL #49: "clinfo" can't open /dev/mem but "sudo clinfo" worked

nakarin, 2018-12-13:
I installed VC4CL and try to run "clinfo" ,its show like these:
""
[VC4CL] can't open /dev/mem
[VC4CL] This program should be run as root. Try prefixing command with: sudo
terminate called after throwing an instance of 'std::system_error'
  what():  Failed to open /dev/mem: Permission denied
Aborted
""

When i try to run "sudo clinfo", its OK.
When i test with PlaidML by "plaidml-setup", the same error:
""
[VC4CL] can't open /dev/mem
[VC4CL] This program should be run as root. Try prefixing command with: sudo
terminate called after throwing an instance of 'std::system_error'
  what():  Failed to open /dev/mem: Permission denied
Aborted
""

When i test with PlaidML by "sudo plaidml-setup", its cannot found device:
""
No supported devices found. Run 'clinfo' and file an issue containing the full output.
""

Anyone can help me.
Thanks.

nakarin, 2018-12-13:
When i use PlaidML by plaidml-setup command its show

`
 [VC4CL] can't open /dev/mem
[VC4CL] This program should be run as root. Try prefixing command with: sudo
terminate called after throwing an instance of 'std::system_error'
what(): Failed to open /dev/mem: Permission denied
Aborted
`
but when i try "sudo plaidml-setup'  like  your suggest, , its cannot found device:
`No supported devices found. Run 'clinfo' and file an issue containing the full output.`

doe300, 2018-12-15:
I don't know why the device is not found. Did you run both commands with the same environment variables (e.g. `LD_LIBRARYPATH`)?

credman0, 2018-12-16:
So I ran into the same issue. After a quick peek at the plaidml setup code (mostly looking around what is currently line 49 in [plaidml_setup.py](https://github.com/plaidml/plaidml/blob/master/plaidml/plaidml_setup.py)) it looks like this is the message that happens when at least one device is found, but no supported devices are found. I think that what is missing is a line in their configuration file, i.e. something needs to be added in [experimental.json](https://github.com/plaidml/plaidml/blob/master/plaidml/configs/experimental.json).

credman0, 2018-12-16:
I also filed an issue with PlaidML ([#214](https://github.com/plaidml/plaidml/issues/214))

nakarin, 2018-12-17:
@doe300 
> I don't know why the device is not found. Did you run both commands with the same environment variables (e.g. `LD_LIBRARYPATH`)?

I set same LD_LIBRARY_PATH,but result still the same.

@credman0 
> So I ran into the same issue. After a quick peek at the plaidml setup code (mostly looking around what is currently line 49 in [plaidml_setup.py](https://github.com/plaidml/plaidml/blob/master/plaidml/plaidml_setup.py)) it looks like this is the message that happens when at least one device is found, but no supported devices are found. I think that what is missing is a line in their configuration file, i.e. something needs to be added in [experimental.json](https://github.com/plaidml/plaidml/blob/master/plaidml/configs/experimental.json). But that is as far as I got because even after reading the descriptions of the GPU I just don't know enough to be sure which values are supposed to be the vector size, etc.

Its may miss config of support VC4.

nakarin, 2018-12-18:
I will try to config experimental.json of PlaidML.
@doe300 , @credman0 
Thanks you for yours suuport.

closed 2018-12-18 by nakarin