doe300/VC4CL #54: Bug in sin()

WrathfulSpatula, 2019-02-01:
When I use the device-side `sin()` function, no matter how I try to isolate it or go around it, I'm getting results that don't make sense. Here's my original kernel:

```
void kernel decomposeamp(global real1* stateProb, global real1* stateAngle, constant bitCapInt* bitCapIntPtr, global cmplx* nStateVec)
{
    bitCapInt ID, Nthreads, lcv;
    
    ID = get_global_id(0);
    Nthreads = get_global_size(0);
    bitCapInt maxQPower = bitCapIntPtr[0];
    real1 angle;
    for (lcv = ID; lcv < maxQPower; lcv += Nthreads) {
        angle = stateAngle[lcv];
        nStateVec[lcv] = sqrt(stateProb[lcv]) * sin((float2)(angle + M_PI_2_F, angle));
    }
}
```

In the cases I'm testing, `stateProb` should always be an array of values that are (at least extremely close to) exactly 1 or 0. If I comment out the multiplication by the `sin` of the vector, my components all do come out 1 or 0, via the `sqrt`. However, the values returned by `sin` seem to be not even close, like cos^2(x) + sin^(x) > 2.7. In fact, the vectorized `sin()` above should be equivalent to components of `(float2)(cos(angle), sin(angle))` by trigonometric identities, but I'm finding that the resulting norm (sum of the squared components) is around 2.7 or larger for a `stateProb[lcv]` of 1.

This doesn't seem to be limited to the vectorized version of `sin()`, either, if I replace this with two separate calls to construct two components separately. It also happens if I switch `M_PI_2_F` out with a known good value for pi/2.

The kernel is not particularly complicated, and if I implement it host-side at the same point in the code, and use the host `sin()` instead, it works correctly.

(Thank you for the awesome compiler, by the way.) Any idea what's going on? Am I doing something wrong?

doe300, 2019-02-02:
> Any idea what's going on?

Yes, the `sin()` implementation (and the implementations for some more mathematical functions) is not yet correct;)
I didn't find yet and efficient and vectorizable implementation for most of the trigonometric functions, which can be implemented on the limited hardware of the VC4 QPUs.

Terminus-IMRC, 2019-02-02:
As for `sin()` and `cos()`, Mendenhall's equation can be easily ported.  It is based on Taylor series and is being used by the Raspberry Pi's firmware for computing sine/cosine in GLSL.

References:
- https://github.com/broadcomCM/android_hardware_broadcom_brcm_usrlib/blob/ics/dag/vmcsx/middleware/khronos/glsl/glsl_mendenhall.c
- https://github.com/broadcomCM/android_hardware_broadcom_brcm_usrlib/blob/ics/dag/vmcsx/middleware/khronos/glsl/glsl_mendenhall.h
- https://arxiv.org/abs/cs/0406049

WrathfulSpatula, 2019-02-02:
Thank you, folks! I figured there might not be a hardware implementation for trigonometry. This actually isn't a blocking issue for our project over at https://github.com/vm6502q/qrack, at the moment, but it would be nice to have, soon.

If the series expansion is accurate for [-pi/2, pi/2], might the implementation be able to adjust any input to this range before expanding? If performance would suffer significantly, I can make sure our values are clamped, but I'm not sure what the OpenCL specification expects.

Thank you again!

doe300, 2019-02-02:
> I figured there might not be a hardware implementation for trigonometry

No, not even hardware support for floating point division (or integer multiplication/division for that matter).

> If the series expansion is accurate for [-2pi, 2pi], might the implementation be able to adjust any input to this range before expanding?

It does, actually it moves all inputs to the range [-pi/2, pi/2]. But the argument reduction is inaccurate (meaning it exceeds the allowed error, for some applications it might still be good enough) for values `|x| > 2pi`. So it will accept any value, will just not yet produce a result as exact as the OpenCL standard mandates. The standard does not specify any value input range, which means all float values are valid.

WrathfulSpatula, 2019-02-18:
I just noticed this commit, and I can easily test whether this fixes Qrack's issue, when I have an opportunity, later.

By the way, doesn't the minimum allowed domain have to have a width of 2*pi, since that's the period of all the basic trigonometric functions? Maybe error is larger outside of [-pi/2, pi/2], but I think you'd still be forced to take the range [-pi, pi], in that case, just to cover the period as best as possible. (I'm just going by the comments; I haven't traced the code.)

doe300, 2019-02-19:
> By the way, doesn't the minimum allowed domain have to have a width of 2*pi, since that's the period of all the basic trigonometric functions? Maybe error is larger outside of [-pi/2, pi/2], but I [...]

I'm not sure if I understood you correctly. Using a range of [-pi/2, pi/2] works, if we swap the sign accordingly, since .e.g. `sin([-pi, -pi/2])` is `-sin([0, pi/2])`. And for the range of [-pi/2, pi/2] the approximation is better, so I move all values to that range and invert the result where necessary.

WrathfulSpatula, 2019-02-24:
That works, too. Yes, in any case, it looks like this passes all the unit tests of our Qrack library, now. I'm waiting for review, to merge in the deprecation of the "shunt" past the OpenCL trigonometry, which is no longer necessary. I'm not sure whether you'll do more work on the trigonometry, if you need to in order to satisfy the OpenCL specification, but it already works for our purposes. Thank you!

doe300, 2019-02-25:
Glad it works for you

closed 2019-02-25 by doe300