doe300/VC4CL #55: Segfault on kernel waiting for list of events

WrathfulSpatula, 2019-02-02:
doe300, as you know, back on the Qrack project's Raspberry Pi 3 support pull request at https://github.com/vm6502q/qrack/pull/143, we get a segfault if our kernel calls receive a large list of events to wait on, but not if the wait list is manually waited on before calling the kernel.

The wait list is daisy-chained. We pass only the direct wait list dependencies for the kernel, while the events in the list can depend on their own separate sets of dependencies that aren't in the kernel list.

We have this pattern, (which works):

```
#if ENABLE_RASPBERRYPI
    clFinish();
#endif

    // Dispatch the primary kernel, to apply the gate.
    cl::Event kernelEvent;
    std::vector<cl::Event> kernelWaitVec = device_context->ResetWaitEvents();
    queue.enqueueNDRangeKernel(ocl.call, cl::NullRange, // kernel, offset
        cl::NDRange(workItemCount), // global number of work items
        cl::NDRange(localGroupSize), // local number (per group)
        &kernelWaitVec, // vector of events to wait for
        &kernelEvent); // handle to wait for the kernel

    queue.flush();

#if ENABLE_RASPBERRYPI
    clFinish();
#endif
```

The `clFinish()` method is a user code "soft finish" implementation; all it does is manually  wait on the list of events returned by `device_context->ResetWaitEvents()`, and it clears the list. In the simplest case we've tested, the list of events just read into the set of buffers needed before calling the kernel. We pass these events in the list to the kernel, ans we manually wait  on the same buffer loads before exiting our method, to make sure that the loads complete before we could lose ephemeral inputs for the buffers.

This implementation has been tested extensively on NVIDIA GTX cards, Intel Core HDs, Intel Cores, and and Intel Xeons, and it's never been an issue anywhere except on the Raspberry Pi.

I think this kind of fall-through asynchronous behavior is asking a _lot_ of the Raspberry Pi's resources, but I don't think there's a logical problem with the chain of dependencies, if we remove the `clFinish()` calls for the Raspberry Pi.

doe300, 2019-02-02:
Lets see, if I understand this correctly:
* with `clFinish()` waiting on all events which belong to this objects, everything works okay (except that the current thread has to wait, which is not good performance wise).
* without `clFinish()` you get a SEGFAULT with VC4CL?

Could you run the execution with `gdb` or `valgrind` (preferably with debug build, `CMAKE_BUILD_TYPE=Debug`) and attach the outputs?

WrathfulSpatula, 2019-02-02:
You are correct, that this is what is happening. It will take me some time to get the profiling output, (our unit tests take quite some time on the Raspberry Pi 3,) but I will get that for you today.

WrathfulSpatula, 2019-02-02:
I built with `CMAKE_BUILD_TYPE=Debug` and ran our unit tests with valgrind. Execution seemed to hang for a long time before I `^Z^C`-ed on the command line. It looks like it hung at about the point I'd expect, in building the wait list dependency graph. The output is here:

https://gist.github.com/WrathfulSpatula/55253877a3b29ae7fc3a5bbc1329e8e4

I'm running this again, and I'm letting it go for as long as it wants, hours if necessary. I'll revise the gist above, if anything turns up.

WrathfulSpatula, 2019-02-03:
Thank you. I'm using the OpenCL C++ headers, so I don't directly call `clRetainEvent()` or `clReleaseEvent(),` but something I am doing is probably triggering this. The kernel needs to wait on the buffer loading events, but those also need to be (separately) guaranteed to complete before we exit the same method, so that we don't lose ephemeral array parameters to load into those buffers, if the kernel is queued but not run before finishing the method.

I realize now, I keep two variable references to these wait events, to cover both of the above dependency conditions, but  the C++ headers assume that most OpenCL objects are unique, such that a deallocator/destructor is probably called twice on each buffer load event.

If keeping the wait objects unique fixes the problem, I'll let you know soon, and we can close the issue. Thank you again!