doe300/VC4CL #56: VC4CL installation error

aaguiar96, 2019-02-13:
Hello.

I was able to build VC4CLStdLib, VC4C and VC4CL.
I installed both VC4CLStdLib and VC4C by:
* sudo dpkg -i /tmp/vc4cl-stdlib.deb
* sudo dpkg -i /tmp/vc4c.deb 

But, when I install VC4CL by
* sudo dpkg -i /tmp/vc4cl.deb
I get: 
* dependency problems prevent configuration of vc4cl:
*   vc4cl depends on opencl-c-headers; however Package opencl-c-headers is not installed.

This is strange because I've installed it by:
* sudo apt-get install opencl-headers

Any ideas? 
Thanks in advance.

doe300, 2019-02-13:
It says requires `opencl-c-headers` (note the **C**), these this is a different package than `opencl-headers`. One is run-time (host-side) the other kernel-side headers.

aaguiar96, 2019-02-13:
Thanks for the answer.
Do you know how to install it?

I've tried 
```
sudo apt install opencl-c-headers
``'
But without success.

aaguiar96, 2019-02-13:
Maybe ubuntu 16.04 does not support opencl-c-headers package?

doe300, 2019-02-13:
[Debian](https://packages.debian.org/stretch/opencl-c-headers) provides them, so they should be in the repositories... If not, you can manually download them from the link I provided.

aaguiar96, 2019-02-13:
I've installed opencl-c-headers from the link but it currupts the installation of opencl-headers. I'm really confused...

Also, clinfo returns 0 platforms.
Can't you provide a full installation guide?

doe300, 2019-02-13:
For Raspbian, the full installation guide is as follows:

1. Install `vc4cl-stdlib` package
2. Install dependencies `clang-3.9, llvm-3.9, llvm-3.9-dev` and `vc4c` package
3. Install dependencies `opencl-c-headers, ocl-icd-opencl-dev` and `vc4cl` package

At least that is the guide for Raspbian. For any other Debian it should be similar.

`clinfo` will return 0 platforms if you did not yet install the `vc4cl` package successfully and have no other OpenCL run-time installed.

doe300, 2019-02-13:
Also note, for other OSes (other than Raspbian), you will need to make sure, the required kernel modules and the [Raspberry Pi userland](https://github.com/raspberrypi/userland) libraries are available.
See also #53 for how to port to other distributions

aaguiar96, 2019-02-14:
Thanks for all the support.
I'll try to install Raspbian instead of Ubuntu and see how it goes.

Just to check. To install VC4C/VC4CL-StdLib/VC4CL I should do
``` 
cmake ..
sudo make install
```
in each build directory, right?

doe300, 2019-02-14:
Well, you should build without root rights and only install with root rights. But otherwise yes. The order is important though, you need to install VC4CLStdLib before building VC4C!

aaguiar96, 2019-02-14:
Installing raspbian do the trick, thank you very much!

Now, I will try to link VC4CL with ROS.
Any suggestion?

doe300, 2019-02-14:
That depends on what ROS is?

aaguiar96, 2019-02-14:
ROS = Robot Operating System.
My main objective is to optimize the processing time of a Visual Odometry algorithm that runs over ROS.

aaguiar96, 2019-02-14:
Note that ROS compiles using Cmake!

doe300, 2019-02-14:
If the ROS supports OpenCL and supports using the OpenCL ICD loader, then it should work out-of-the-box.

aaguiar96, 2019-02-14:
I'm not sure... But I'll give it a try.

Thanks for all the support.
I think you can close the issue.

closed 2019-02-18 by doe300