doe300/VC4CL #57: Precompilation error on tests

aaguiar96, 2019-02-18:
Hello!

I installed all the setup correctly (clinfo runs fine) but when I compile the tests in VC4C/example folder I always get precompile errors.

The compilation command I'm using is:
```
VC4C --hex -o /dev/null ./example/file.cl
```
I'm running this in a Raspberry Pi 3B.

Any ideas?
Thanks in advance.

doe300, 2019-02-18:
What errors do you get?

If you want to use the manually built and installed binary, then you need to install `VC4CLStdLib` first and make sure the `VC4C` project uses the installed location (the location used is printed in the cmake configuration output).

aaguiar96, 2019-02-18:
I get the following errors.

```
[D] Mon Feb 18 13:24:44 2019: Compiling './example/test_instructions.cl' into '/dev/null' with optimization level 2 and options '' ...
[D] Mon Feb 18 13:24:44 2019: Temporary file '/tmp/vc4c-g5SGbq' created
[D] Mon Feb 18 13:24:44 2019: Temporary file '/tmp/vc4c-UPLL4F' created
[I] Mon Feb 18 13:24:44 2019: Compiling OpenCL to LLVM-IR with: /usr/bin/clang -cc1 -triple spir-unknown-unknown  -I ./example -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /usr/local/include/vc4cl-stdlib/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-UPLL4F ./example/test_instructions.cl
[I] Mon Feb 18 13:24:49 2019: Linking LLVM-IR modules with: /usr/bin/llvm-link -only-needed -internalize -o=/tmp/vc4c-g5SGbq /tmp/vc4c-UPLL4F /usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc
[E] Mon Feb 18 13:24:49 2019: Errors in precompilation:
[E] Mon Feb 18 13:24:49 2019: Expected no forward declarations!
!3 = <temporary!> !{}
incorrect number of operands in llvm.ident metadata
!0 = !{i32 0, i32 1}
/usr/bin/llvm-link: /tmp/vc4c-UPLL4F: error: input module is broken!

[E] Mon Feb 18 13:24:49 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x76b000fc]
[E] Mon Feb 18 13:24:49 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x82293c [0x76e1793c]
[E] Mon Feb 18 13:24:49 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::linkLLVMModules(std::vector<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>, std::allocator<vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3> > >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x350 [0x76e18988]
[E] Mon Feb 18 13:24:49 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::linkInStdlibModule(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x78 [0x76e18150]
[E] Mon Feb 18 13:24:49 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), void (*)(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x60 [0x76e1cc40]
[E] Mon Feb 18 13:24:49 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76e1bd00]
[E] Mon Feb 18 13:24:49 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0xd0 [0x76e1abec]
[E] Mon Feb 18 13:24:49 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilatio [... cut ...]
/usr/bin/llvm-link: /tmp/vc4c-UPLL4F: error: input module is broken!

terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: Expected no forward declarations!
!3 = <temporary!> !{}
incorrect number of operands in llvm.ident metadata
!0 = !{i32 0, i32 1}
/usr/bin/llvm-link: /tmp/vc4c-UPLL4F: error: input module is broken!

Aborted
```

I've installed VC4CLStdLib already.
Is this compilation command the right way to test VC4CL?

doe300, 2019-02-18:
Which version of clang do you have installed?

aaguiar96, 2019-02-18:
Here it is.

```
$ clang --version
clang version 3.9.1-9+rpi1 (tags/RELEASE_391/rc2)
Target: armv6-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
```

aaguiar96, 2019-02-19:
Any ideas? Am I compiling it correctly?
Thanks in advance.

doe300, 2019-02-19:
The only thing I can think of is a version difference between your clang front-end and your llvm-opt program.
Make sure you only have a single version of clang/llvm and all its required tools installed and try to re-build and re-install the VC4CLStdLib pre-built module again (see [here](https://github.com/doe300/VC4C#known-issues)).
Also, if your change the version of llvm, you will need to re-configure and re-build VC4C.

aaguiar96, 2019-02-19:
I installed llvm by
```
sudo apt install llvm-dev
```
Is this correct?

doe300, 2019-02-19:
Sure. Did you already have clang or "base" llvm installed?

Also you could try to install `clang` and `llvm` pacakges.

aaguiar96, 2019-02-19:
Ok. I reinstalled all the setup and now i'm getting this error:
```
[D] Tue Feb 19 11:08:48 2019: Compiling './example/test_instructions.cl' into '/dev/null' with optimization level 2 and options '' ...
[D] Tue Feb 19 11:08:48 2019: Temporary file '/tmp/vc4c-Kzrvdc' created
[D] Tue Feb 19 11:08:48 2019: Temporary file '/tmp/vc4c-KuQxbe' created
[I] Tue Feb 19 11:08:48 2019: Compiling OpenCL to LLVM-IR with: /usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown  -I ./example -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-KuQxbe ./example/test_instructions.cl
[E] Tue Feb 19 11:08:50 2019: Errors in precompilation:
[E] Tue Feb 19 11:08:50 2019: <built-in>:2:10: fatal error: '/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h' file not found
#include "/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h"
         ^
1 error generated.

[E] Tue Feb 19 11:08:50 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x76b2de50]
[E] Tue Feb 19 11:08:50 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x7d2ff8 [0x76dfaff8]
[E] Tue Feb 19 11:08:50 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0x7d3360 [0x76dfb360]
[E] Tue Feb 19 11:08:50 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLWithDefaultHeader(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x64 [0x76dfb724]
[E] Tue Feb 19 11:08:50 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), void (*)(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x60 [0x76dfece4]
[E] Tue Feb 19 11:08:50 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76dfdc88]
[E] Tue Feb 19 11:08:50 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0xa0 [0x76dfcdf4]
[E] Tue Feb 19 11:08:50 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType) [... cut ...]
#include "/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h"
         ^
1 error generated.

terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: <built-in>:2:10: fatal error: '/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h' file not found
#include "/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h"
         ^
1 error generated.

Aborted
```

This means that VC4C is not finding VC4CLStdLib. How to fix it?

doe300, 2019-02-19:
The way that VC4C finds the VC4CLStdLib headers is currently still very static (see https://github.com/doe300/VC4C/issues/111). You have to make sure, it finds the headers from the installed directory, e.g. by specifying the cmake variable `VC4CL_STDLIB_HEADER_SOURCE` when configuring VC4C.

aaguiar96, 2019-02-19:
Ok I only have to recomplie VC4C using that flag?
```
cmake -DVC4CL_STDLIB_HEADER_SOURCE="path" ..
make
sudo make install
```
Right?

doe300, 2019-02-19:
Yes, should be enough

aaguiar96, 2019-02-19:
Ok, it work. But now another problem appears. Sorry for being annoying...
```
[D] Tue Feb 19 12:08:45 2019: Compiling './example/hello_world.cl' into '/dev/null' with optimization level 2 and options '' ...
[D] Tue Feb 19 12:08:45 2019: Temporary file '/tmp/vc4c-CYcLrp' created
[D] Tue Feb 19 12:08:45 2019: Temporary file '/tmp/vc4c-u8v8MF' created
[I] Tue Feb 19 12:08:45 2019: Compiling OpenCL to LLVM-IR with: /usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown  -I ./example -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-u8v8MF ./example/hello_world.cl
[I] Tue Feb 19 12:08:46 2019: Linking LLVM-IR modules with: /usr/bin/llvm-link-3.9 -only-needed -internalize -o=/tmp/vc4c-CYcLrp /tmp/vc4c-u8v8MF /home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.bc
[D] Tue Feb 19 12:08:47 2019: Temporary file '/tmp/vc4c-u8v8MF' deleted
[I] Tue Feb 19 12:08:47 2019: Compilation complete!
[D] Tue Feb 19 12:08:47 2019: Reading LLVM module from bit-code...
[D] Tue Feb 19 12:08:47 2019: Found SPIR kernel-function: hello_world
[D] Tue Feb 19 12:08:47 2019: Reading function void hello_world(...)
[D] Tue Feb 19 12:08:47 2019: Reading parameter (g) <16 x i8>* %out
[D] Tue Feb 19 12:08:47 2019: Mapping function 'hello_world'...
[D] Tue Feb 19 12:08:47 2019: Generating label label %tmp.0
[D] Tue Feb 19 12:08:47 2019: Generating immediate call to get_global_idj -> i32
[D] Tue Feb 19 12:08:47 2019: Generating calculating index i32 %call of (g) <16 x i8>* %out into (g) <16 x i8>* %arrayidx
[D] Tue Feb 19 12:08:47 2019: Generating writing of <16 x i8> container with 16 elements into (g) <16 x i8>* %arrayidx
[D] Tue Feb 19 12:08:47 2019: Generating return nothing
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: EliminatePhiNodes
[I] Tue Feb 19 12:08:47 2019: -----
[I] Tue Feb 19 12:08:47 2019: Inlining functions for kernel: hello_world
[I] Tue Feb 19 12:08:47 2019: -----
[D] Tue Feb 19 12:08:47 2019: -----
[I] Tue Feb 19 12:08:47 2019: Running normalization passes for: hello_world
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: MapMemoryAccess
[D] Tue Feb 19 12:08:47 2019: Determining memory access for kernel: hello_world
[D] Tue Feb 19 12:08:47 2019: Global parameter '(g) <16 x i8>* %out' which is written to will be stored in RAM and accessed via VPM
[D] Tue Feb 19 12:08:47 2019: Generating memory access which cannot be optimized: store <16 x i8> container with 16 elements into (g) <16 x i8>* %arrayidx
[D] Tue Feb 19 12:08:47 2019: Found base address - with offset -1 for writing into memory
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: ResolveStackAllocations
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: Intrinsics
[D] Tue Feb 19 12:08:47 2019: Intrinsifying multiplication with left-shift: i32 %index_offset.1 = mul i32 %call, i32 16
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: EliminateReturns
[D] Tue Feb 19 12:08:47 2019: Replacing return in kernel-function with branch to end-label
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: HandleLiteralVector
[D] Tue Feb 19 12:08:47 2019: Rewriting move from container register vpm = <16 x i8> container with 16 elements
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: MapGlobalDataToAddress
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: HandleImmediates
[D] Tue Feb 19 12:08:47 2019: Mapping constant for immediate value 4 to: 4 (4)
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 72
[D] Tue Feb 19 12:08:47 2019: Mapping constant for immediate value 0 to: 0 (0)
[D] Tue Feb 19 12:08:47 2019: Mapping constant for immediate value 0 to: 0 (0)
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 72
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 101
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 108
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 108
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 111
[D] Tue Feb 19 12:08:47 2019: Calculating immediate value 32 with operation 'ftoi' and immediate value 32.000000 (37)
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 87
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 111
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 114
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 108
[D] Tue Feb 19 12:08:47 2019: Calculating immediate value 100 with operation 'mul24' and immediate value 10 (10)
[D] Tue Feb 19 12:08:47 2019: Loading immediate value: 33
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: PropagateGroupUniformValues
[D] Tue Feb 19 12:08:47 2019: 
[D] Tue Feb 19 12:08:47 2019: Running pass: CheckNormalized
[E] Tue Feb 19 12:08:47 2019: Failed to in-line or intrinsify function-call: i32 %call = i32 get_global_idj(i32 0) (group_uniform)
[W] Tue Feb 19 12:08:47 2019: Candidates:
[W] Tue Feb 19 12:08:47 2019: void hello_world((g) <16 x i8>* %out
[E] Tue Feb 19 12:08:47 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string [... cut ...]
[E] Tue Feb 19 12:08:47 2019: Background worker threw error: Normalizer: Not normalized instruction found: i32 %call = i32 get_global_idj(i32 0) (group_uniform)
[E] Tue Feb 19 12:08:47 2019: While running worker task: Normalization
[D] Tue Feb 19 12:08:47 2019: Temporary file '/tmp/vc4c-CYcLrp' deleted
[E] Tue Feb 19 12:08:47 2019: Compiler threw exception: Normalizer: Not normalized instruction found: i32 %call = i32 get_global_idj(i32 0) (group_uniform)
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Normalizer: Not normalized instruction found: i32 %call = i32 get_global_idj(i32 0) (group_uniform)
Aborted
```

doe300, 2019-02-19:
the linker still uses the local (not the installed) VC4CLStdLib file (`/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.bc`), which it shouldn't if you want to use the installed package.

aaguiar96, 2019-02-19:
Ok. How can I change it?

aaguiar96, 2019-02-19:
I think you're right.
I get:
``` 
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: /usr/bin/llvm-link-3.9: /home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.bc: error: Could not open input file: No such file or directory
/usr/bin/llvm-link-3.9: error loading file '/home/pi/rpi3-vc4cl/VC4C/../VC4CLStdLib/include/VC4CLStdLib.bc
```
but I don't know how to fix it...

aaguiar96, 2019-02-19:
This is strange because the flag `VC4CL_STDLIB_MODULE` is set to `"/usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc"`.

doe300, 2019-02-19:
Did you rebuild and re-install it with the flag set?

aaguiar96, 2019-02-19:
Yes, I did:
`cmake -DVC4CL_STDLIB_MODULE="/usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc"`. when building `VC4C`but it didn't solve.

doe300, 2019-02-19:
That is the same error from above. This is an LLVM internal error, idk what it is exactly, but it looks like some incompatibility between the LLVM version used to read the module and the LLVM version used to generate it.

aaguiar96, 2019-02-19:
Yes, I agree.
I'm not able to solve it...

Thanks anyway.

closed 2019-02-20 by doe300