doe300/VC4CL #58: Illegal Instruction error

aaguiar96, 2019-03-07:
Hello.
I'm using a Raspberry Pi 3 to optimise some code. I've already run some test programs like summing two arrays, etc.
Now, I'm stuck with the following error.

```
[D] Thu Mar  7 17:04:25 2019: Compiling 'src/libviso2/libviso2/kernels/get_inliers.cl' into 'out' with optimization level 2 and options '' ...
[D] Thu Mar  7 17:04:25 2019: Temporary file '/tmp/vc4c-EQ5dwJ' created
[D] Thu Mar  7 17:04:25 2019: Temporary file '/tmp/vc4c-i0dILi' created
[I] Thu Mar  7 17:04:25 2019: Compiling OpenCL to LLVM-IR with: /usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown  -I src/libviso2/libviso2/kernels -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant  -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /usr/local/include/vc4cl-stdlib/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-i0dILi src/libviso2/libviso2/kernels/get_inliers.cl
[E] Thu Mar  7 17:04:25 2019: Errors in precompilation:
[E] Thu Mar  7 17:04:25 2019: Stack dump:
0.	Program arguments: /usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown -I src/libviso2/libviso2/kernels -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /usr/local/include/vc4cl-stdlib/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-i0dILi src/libviso2/libviso2/kernels/get_inliers.cl 
Segmentation fault

[E] Thu Mar  7 17:04:25 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x76aa9d44]
[E] Thu Mar  7 17:04:25 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x8200ec [0x76dc00ec]
[E] Thu Mar  7 17:04:25 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : +0x820498 [0x76dc0498]
[E] Thu Mar  7 17:04:25 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLWithDefaultHeader(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x64 [0x76dc0860]
[E] Thu Mar  7 17:04:25 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), void (*)(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x60 [0x76dc525c]
[E] Thu Mar  7 17:04:25 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76dc4200]
[E] Thu Mar  7 17:04:25 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0xa0 [0x76dc336c]
[E] Thu Mar  7 17:04:25 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&), std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> vc4c::precompilation::chainSteps<(vc4c::SourceType)3, (vc4c::SourceType)1, (vc4c::SourceType)3>(std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&, std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)3>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)> const&)::{lambda(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)#1}>::_M_invoke(std::_Any_data const&, vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x5c [0x76dc5630]
[E] Thu Mar  7 17:04:25 2019:  (9) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)>::operator()(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&) const+0x78 [0x76dc4200]
[E] Thu Mar  7 17:04:25 2019:  (10) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x608 [0x76dca260]
[E] Thu Mar  7 17:04:25 2019:  (11) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x164 [0x76dc7fd8]
[E] Thu Mar  7 17:04:25 2019:  (12) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::alloca [... cut ...]
0.	Program arguments: /usr/bin/clang-3.9 -cc1 -triple spir-unknown-unknown -I src/libviso2/libviso2/kernels -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -finclude-default-header -include /usr/local/include/vc4cl-stdlib/defines.h -x cl -S -emit-llvm-bc -o /tmp/vc4c-i0dILi src/libviso2/libviso2/kernels/get_inliers.cl 
Segmentation fault

Aborted
```

This happens even in a kernel that had already worked.
The error returned by OpenCL is CL_COMPILE_FAILURE (-15).

Any ideas?
Thanks in advance.

doe300, 2019-03-07:
The error is thrown inside of clang/LLVM.

- What version of clang/LLVM do you have?
- What does `VC4C --version` report?
- Did you upgrade an existing version of Clang/LLVM or install a new version recently?

aaguiar96, 2019-03-07:
* Running `clang-3.9 --version throws the following exception.
```
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Stack dump:
0.	Program arguments: clang-3.9 --version 
Aborted
```

* `VC4C --version``` returns
``` 
Running VC4C in version: 0.4
Build configuration:
debug mode; multi-threaded optimization; clang 3.9+ OpenCL features; clang in /usr/bin/clang-3.9; LLVM library front-end with libLLVM 3.9; VC4CL standard-library in /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch; vc4asm verification
```

* No, I did not upgrade Clang/LLVM.

doe300, 2019-03-07:
> Should I reinstall clang-3.9(-dev)?

You may try this. Otherwise, the clang error seems very strange...

aaguiar96, 2019-03-07:
It worked.
I just hope it doesn't magically happen again...

Thanks for the support.

closed 2019-03-07 by doe300