doe300/VC4CL #61: Performance questions

aaguiar96, 2019-03-13:
Hello.

I'm optimizing a RANSAC algorithm using Videocore IV and VC4CL.
However, the performance of the GPU for my kernel is worst than CPU's.

So, I need some help with some performance issues.

* Since VC4CL only allows 12 work group size, I'm using a three dimensional work group (12x12x12 -> global) with local size (1x1x1 -> local). Is this a good option or should, for example, (6x6x6 -> global) and (2x2x2 -> local) be better?

* Is there any data type that improves the performance, like uint8, uint16, float8, float16, etc...

* I notice that sums, multiplications and divisions take some time. Is better to perform this operations using only integers or should I use floats?

Thanks in advance.

doe300, 2019-03-13:
> Since VC4CL only allows 12 work group size, [...]

A local size of 1 runs only on 1 QPU. You want to have a local size of 12 or as close to 12 as possible, e.g. 12x1x1, 4x3x1 or 3x2x2. The global size/number of work-groups is unlimited.

> Is there any data type that improves the performance, [...]

All calculations are done with 16-element SIMD vectors, so any `xxx16` type will be good. Also, depending on the type of calculation, explicitly choosing between signed/unsigned and char/short/int might give you some performance gains (e.g. division is unsigned faster than signed and `uchar` division is way faster than with `int`s, similar for multiplication).

> I notice that sums, multiplications and divisions take some time. [...]

Additions only take 1 cycle, independent of the type. For multiplications it depends on the type (1 cycle for `floatn`, `ucharn` and `ushortn`, some more for `charn`, `shortn` and `uintn`, most for `intn` (about 10-20 iirc). Divisions are very tricky, since the GPU does not have native support, so we need to emulate it. Smaller types (char, short) are faster than larger types (int), also unsigned is faster than signed. Floating point division takes a constant ~20 cycles.

aaguiar96, 2019-03-13:
Thank you!!

I only have one question. Everytime I set global size greater than 12 I get  `CL_INVALID_WORK_GROUP_SIZE`... So a group size of 512x512x512 and a local size of 12x1x1 is valid?

doe300, 2019-03-13:
The global size needs to be a multiple of the local size (in every dimension). This is mandated by the OpenCL standard. 512x512x512 is not a a multiple of 12x1x1. If you e.g. use a local size of 8x1x1, it should work.

aaguiar96, 2019-03-13:
Ok, I got it.
Thanks you so much!!

aaguiar96, 2019-03-22:
Hello again.

I've been trying to optimize a simple kernel using 16-element data types (simple sum between two floating point buffers). However, I'm not getting any gains...
* Should I declare them in the host side with 16 times less size than the original float buffers?
* How to properly access their elements in the kernel?

Thanks in advance.

doe300, 2019-03-22:
> Should I declare them in the host side with 16 times less size than the original float buffers?

The buffer size needs to be the same, doesn't it?

> How to properly access their elements in the kernel?

What do you mean by that?

I cannot really help if I don't know what the kernel code looks like

aaguiar96, 2019-03-22:
My thought was, if float16 type provides 16 floats per index, its size should be 16 times less than a float array. I am wrong?

The kernel looks like this:
```
#define LOCAL_SIZE 12

__atribute__ ((reqd_work_group_size(LOCAL_SIZE,1,1)))
__kernel void float_sum(__global const float16* A, __global const float16* B, __global float16* C) {
  int id = get_global_id(0);
  C[id] = A[id] + B[id];
}
```
What do you think?

doe300, 2019-03-22:
The union of `float[16]` and `float16` will probably be converted into some very inefficient code. Also, you don't need it.

The following kernel code should suffice:

    #define LOCAL_SIZE 12

    __attribute__ ((reqd_work_group_size(LOCAL_SIZE,1,1)))
    __kernel void float_sum(__global const float16* A, __global const float16* B, __global float16* C) {
      int id = get_global_id(0);
      C[id] = A[id] + B[id];
    }

OpenCL C (device side) code can apply arithmetic operations to vectors of values, so there is no need to manually unroll the `float16` vector.

aaguiar96, 2019-03-22:
I agree!

However, even with that approach, for 19200 size arrays with 192000 global work group size and 12 local group size I get:
* GPU = 0.0108 sec
* CPU = 0.0016 sec

Shouldn't GPU be faster?

doe300, 2019-03-22:
> [...] for 19200 size arrays with 19200 global work group size and 12 local group size [...]

Do you have 19200 `float` entries or 19200 `float16` entries? Since 19200 `float` entries would only need 1200 work-items to calculate, since a work-item handles 16 elements at once.

Also, does the measured time include the compilation time for the GPU kernel? And with such small times (1-10ms), GPU probably will never win, since it has a lot more initial overhead.

aaguiar96, 2019-03-22:
I have 1200 float16 entries.
The time is measured without the compilation. Ok, I think is fair to test in heavier kernels...

aaguiar96, 2019-03-22:
Ok, the global size was wrong!! I forgot to divide it by 16.
Now I've got:
* GPU = 0.0024
* COU = 0.0017
Much better, thank you!!

aaguiar96, 2019-04-06:
Hello again.
Another tricky thing is happening, I hope you can help me out...

I have this kernel running on RPi GPU:

```
#define LOCAL_SIZE 12
#define RANSAC_ITERS 122/16

struct match_t
{
  float u1p;
  float v1p;
  float u1c;
  float v1c;
};

__attribute__((reqd_work_group_size(LOCAL_SIZE,1,1)))
__kernel void get_inliers(const ushort matches_size,
			  const float16 threshold,
			  __global const struct match_t* restrict matches,
			  __global const float16*        restrict F,
			  __global uint16*               restrict best_inliers,
			  __global uint16*               restrict counter,
			  __global float16*              restrict debug_)
{
  uint id = get_global_id(0);
  uint offset = id * 9;

  __local float16 f[9];
  f[0] = F[0 + offset];
  f[1] = F[1 + offset];
  f[2] = F[2 + offset];
  f[3] = F[3 + offset];
  f[4] = F[4 + offset];
  f[5] = F[5 + offset];
  f[6] = F[6 + offset];
  f[7] = F[7 + offset];
  f[8] = F[8 + offset];

  float16 x2tFx1;
  float16 Fx1u,Fx1v,Fx1w;
  float16 Ftx2u,Ftx2v;
  float16 d;

  for(size_t i = 0; i < matches_size && id < RANSAC_ITERS; i++) {
    Fx1u   = f[0] * matches[i].u1p + f[1] * matches[i].v1p + f[2];
    Fx1v   = f[3] * matches[i].u1p + f[4] * matches[i].v1p + f[5];
    Fx1w   = f[6] * matches[i].u1p + f[7] * matches[i].v1p + f[8];
    Ftx2u  = f[0] * matches[i].u1c + f[3] * matches[i].v1c + f[6];
    Ftx2v  = f[1] * matches[i].u1c + f[4] * matches[i].v1c + f[7];
    x2tFx1 = matches[i].u1c * Fx1u + matches[i].v1c * Fx1v + Fx1w;

    d = x2tFx1 * x2tFx1 / (Fx1u * Fx1u + Fx1v * Fx1v + Ftx2u * Ftx2u + Ftx2v * Ftx2v);

    best_inliers[i + id * matches_size] = abs(fabs(d) < threshold);
    counter[id] = counter[id] + best_inliers[i + id * matches_size];
    debug_[id] = Fx1v;
  }
}
```			  

A strange thing is happening.
I check the values of all the private variables and for example `Fx1u` is right but `Fx1v` is wrong calculated. However, if I switch its position with `Fx1u` (calculate `Fx1v` first and then `Fx1u`) now `Fx1v` is correct and `Fx1u` is incorrect... Do you know what might be the problem? Is there some bug on my kernel?

Thanks in advance.

doe300, 2019-04-06:
> I check the values of all the private variables

How do you check the values of local variables in VC4CL?

aaguiar96, 2019-04-06:
I have the global variable `debug_` to output the local variables as it shown in the kernel.

doe300, 2019-04-06:
Oh right, missed that.
You declare `f` as `__local`, is that on purpose that it is shared between all work-items? If so, you should add some barrier to guarantee the values not be changed by another work-item.

aaguiar96, 2019-04-06:
It can be shared because the index o F dependes on de global id, so the same index is not accessed by different work item right?

doe300, 2019-04-07:
__local float16 f[9];
      f[0] = F[0 + offset];
      f[1] = F[1 + offset];
      f[2] = F[2 + offset];
      f[3] = F[3 + offset];
      f[4] = F[4 + offset];
      f[5] = F[5 + offset];
      f[6] = F[6 + offset];
      f[7] = F[7 + offset];
      f[8] = F[8 + offset];

This code is executed on all work-item writing into the same `__local` vector `f`. While the index of `F` is different per work-item, the indices of `f` are the same. So the work-items write different values into `f` at the same positions, making the final contents of `f` indeterministic.

aaguiar96, 2019-04-07:
Ok, that makes sense.
So, making `__private float16 f[9]` or using a `barrier` will solve the problem right?

closed 2019-04-09 by aaguiar96