doe300/VC4CL #62: Error running

aaguiar96, 2019-03-19:
Hello.

I installed all in ubuntu 18.04 and confirmed that VC4C is working.
However, while creating a cl::Buffer I get this error message:
```
[VC4CL] test_cl.o: ioctl_set_msg failed: -1
[VC4CL] Error in mbox_propertie: Inappropriate ioctl for device
```

Any idea what this may be?
Thanks in advance.

doe300, 2019-03-19:
Did you install Raspbian or a "vanilla" Ubuntu?

OSes other than Raspbian do not provide the mailbox driver and the accompanying character device to communicate with the firmware.

aaguiar96, 2019-03-19:
I installed Ubuntu because I need to work on it...

Is it possible to install those things in Ubuntu?

doe300, 2019-03-20:
You will need to provide the Raspberry PI [userland](https://github.com/raspberrypi/userland) and [firmware](https://github.com/raspberrypi/firmware) and also probably the correct kernel module, see [here](https://github.com/doe300/VC4CL/issues/53#issuecomment-457687444) and [here](https://github.com/doe300/VC4CL/issues/53#issuecomment-457706381).

closed 2019-03-22 by aaguiar96