doe300/VC4CL #67: "Execution: failed" when running simple dot product kernel

kernhanda, 2019-03-27:
Using the most recent commits from VC4CLStdLib, VC4C, and VC4CL (while commenting out the offending lines from #66 ), I'm running into a failure when running the code at https://github.com/kernhanda/opencl_dot_product.

To repro:
```shell
git clone https://github.com/kernhanda/opencl_dot_product
make
sudo ./dot_product
```

Output:
```
$ sudo ./dot_product
On the host, the dot product took 0.013763 seconds.
[VC4CL](    dot_product): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x18bf75c, const char* "clIcdGetPlatformIDsKHR")
[VC4CL](    dot_product): get extension function address: clIcdGetPlatformIDsKHR
[VC4CL](    dot_product): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x18bf75c, const char* "clGetPlatformInfo")
[VC4CL](    dot_product): get extension function address: clGetPlatformInfo
[VC4CL](    dot_product): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 0, cl_platform_id* 0, cl_uint* 0x7e7e05c8)
[VC4CL](    dot_product): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 1, cl_platform_id* 0x18a9468, cl_uint* 0)
[VC4CL](    dot_product): API call: cl_int clGetPlatformInfo(cl_platform_id 0x18bf75c, cl_platform_info 2308, size_t 0, void* 0, size_t* 0x7e7e0558)
[VC4CL](    dot_product): API call: cl_int clGetPlatformInfo(cl_platform_id 0x18bf75c, cl_platform_info 2308, size_t 159, void* 0x18c02f8, size_t* 0)
[VC4CL](    dot_product): API call: cl_int clGetPlatformInfo(cl_platform_id 0x18bf75c, cl_platform_info 2336, size_t 0, void* 0, size_t* 0x7e7e0558)
[VC4CL](    dot_product): API call: cl_int clGetPlatformInfo(cl_platform_id 0x18bf75c, cl_platform_info 2336, size_t 6, void* 0x18b7548, size_t* 0)
[VC4CL](    dot_product): API call: cl_int clGetDeviceIDs(cl_platform_id 0x18bf75c, cl_device_type 4, cl_uint 1, cl_device_id* 0x7e7e067c, cl_uint* 0)
[VC4CL](    dot_product): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0x7e9e066c, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0x7e7e05a8, void* 0, cl_int* 0x7e9e0660)
[VC4CL](    dot_product): Tracking live-time of object: cl_context
[VC4CL](    dot_product): API call: cl_int clGetDeviceInfo(cl_device_id 0x18bf770, cl_device_info 4100, size_t 4, void* 0x7e9e0668, size_t* 0)
[VC4CL] base=0x3fc00000, mem=0x76feb000
[VC4CL] V3D base: 0x76feb000
[VC4CL](    dot_product): API call: cl_program clCreateProgramWithSource(cl_context 0x18bffa4, cl_uint 1, const char** 0x7e7e0678, const size_t* 0x7e7e0674, cl_int* 0x7e7e066c)
[VC4CL](    dot_product): Tracking live-time of object: cl_program
[VC4CL](    dot_product): API call: cl_int clBuildProgram(cl_program 0x1893d4c, cl_uint 0, const cl_device_id* 0, const char* (null), void(CL_CALLBACK*)(cl_program program, void* user_data) 0x7e7e0650, void* 0)
[VC4CL](    dot_product): Precompiling source with: 
[VC4CL](    dot_product): Dumping program sources to /tmp/vc4cl-source-15489783.cl
[VC4CL](    dot_product): Precompilation complete with status: 0
[VC4CL](    dot_product): Compilation log: [W] Wed Mar 27 07:48:17 2019: Warnings in precompilation:
[W] Wed Mar 27 07:48:17 2019: <stdin>:22:1: warning: null character ignored
<U+0000>
^
1 warning generated.

[VC4CL](    dot_product): Compiling source with: 
[VC4CL](    dot_product): Compilation complete with status: 0
[VC4CL](    dot_product): Dumping program sources to /tmp/vc4cl-binary-1608230074.bin
[VC4CL](    dot_product): API call: cl_kernel clCreateKernel(cl_program 0x1893d4c, const char* "dot_product", cl_int* 0x7e9e0660)
[VC4CL](    dot_product): Tracking live-time of object: cl_kernel
[VC4CL](    dot_product): API call: cl_mem clCreateBuffer(cl_context 0x18bffa4, cl_mem_flags 36, size_t 1048512, void* 0x7e8e0678, cl_int* 0x7e9e0660)
[VC4CL] Mailbox file descriptor opened: 4
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 3b400000 04c00000 00000000
[VC4CL](    dot_product): Mailbox request: succeeded
[VC4CL](    dot_product): Tracking live-time of object: cl_mem
[VC4CL](    dot_product): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 000fffc0 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 00000019 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000019 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be3a9000 00000000 00000000
[VC4CL] base=0x3e3a9000, mem=0x70d00000
[VC4CL](    dot_product): Allocated 1048512 bytes of buffer: handle 25, device address 0xbe3a9000, host address 0x70d00000
[VC4CL](    dot_product): API call: cl_mem clCreateBuffer(cl_context 0x18bffa4, cl_mem_flags 33, size_t 1048512, void* 0x7e7e06b8, cl_int* 0x7e9e0660)
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 3b400000 04c00000 00000000
[VC4CL](    dot_product): Mailbox request: succeeded
[VC4CL](    dot_product): Tracking live-time of object: cl_mem
[VC4CL](    dot_product): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 000fffc0 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 0000001a 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 0000001a 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be2a8000 00000000 00000000
[VC4CL] base=0x3e2a8000, mem=0x70c00000
[... 477 lines cut ...]
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CUSTOM)  No devices found in platform
[VC4CL](         clinfo): API call: cl_context clCreateContextFromType(const cl_context_properties* 0, cl_device_type 4294967295, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0x7e81f0d0, void* 0, cl_int* 0x7e81f0e0)
[VC4CL](         clinfo): API call: cl_int clGetDeviceIDs(cl_platform_id 0x6c7e14, cl_device_type 4294967295, cl_uint 1, cl_device_id* 0x7e81f098, cl_uint* 0x7e81f09c)
[VC4CL](         clinfo): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0x7e81f098, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0x7e81ef78, void* 0, cl_int* 0x7e81f0e0)
[VC4CL](         clinfo): Tracking live-time of object: cl_context
[VC4CL](         clinfo): API call: cl_int clGetContextInfo(cl_context 0x697144, cl_context_info 4225, size_t 0, void* 0, size_t* 0x7e81f170)
[VC4CL](         clinfo): API call: cl_int clGetContextInfo(cl_context 0x697144, cl_context_info 4225, size_t 32, void* 0x6c7e50, size_t* 0)
[VC4CL](         clinfo): API call: cl_int clGetDeviceInfo(cl_device_id 0x6c7e28, cl_device_info 4145, size_t 4, void* 0x7e81f16c, size_t* 0)
[VC4CL](         clinfo): API call: cl_int clGetDeviceInfo(cl_device_id 0x6c7e28, cl_device_info 4139, size_t 867, void* 0x6cb16d, size_t* 0x7e81f14c)
[VC4CL](         clinfo): API call: cl_int clReleaseContext(cl_context 0x697144)
[VC4CL](         clinfo): Releasing live-time of object: cl_context
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_ALL)  Success (1)
    Platform Name                                 OpenCL for the Raspberry Pi VideoCore IV GPU
    Device Name                                   VideoCore IV GPU

ICD loader properties
  ICD loader Name                                 OpenCL ICD Loader
  ICD loader Vendor                               OCL Icd free software
  ICD loader Version                              2.2.11
  ICD loader Profile                              OpenCL 2.1

[VC4CL](         clinfo): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000000 00000000 00000000
[VC4CL](         clinfo): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 00000000 00000000 00000000
[VC4CL] Mailbox file descriptor closed: 3
```

doe300, 2019-03-27:
Unless you explicitly disabled `REGISTER_POKE_KERNEL`, the "Execution: failed" error means that the kernel execution timed out.

This could be due to one of two problems:

- The execution of a single kernel instance takes longer than the timeout (currently 30 seconds)
- The execution somehow hangs the QPU or runs into an infinite loop (more likely)

Can you run the kernel in emulation mode and send the output (or at least the error)?
To do this, you will have to recompile `VC4CL` with the `MOCK_HAL` option enabled and execute the code as before. Instead of the GPU, this will execute the code in the built-in emulator which will take longer than the normal execution but should provide more detailed error information.

kernhanda, 2019-03-27:
Building VC4CL with `cmake .. -DCMAKE_BUILD_TYPE=Debug -DBUILD_DEBUG=ON -DMOCK_HAL=ON` and then running `clinfo` results in a segfault.

The crash is in `src/extensions.cpp`:
```cpp
65| void* clGetExtensionFunctionAddress(const char* name)
66| {
67+>    return VC4CL_FUNC(clGetExtensionFunctionAddressForPlatform)(Platform::getVC4CLPlatform().toBase(), name);
68| }
```

I believe it's because `VC4CL_FUNC(clGetExtensionFunctionAddressForPlatform)` isn't found, so it just segfaults.

Let me know if I'm building VC4CL incorrectly.

doe300, 2019-03-28:
When `VC4CL` is built in emulation mode (`MOCK_HAL` enabled), it is built without ICD loader support.

Thus, to run `clinfo` with the emulation version, you need to make sure, `clinfo` loads the `libVC4CL.so` as `libOpenCL.so`:

1. Create a symlink `ln --symbolic libVC4CL.so libOpenCL.so` within the `build/src` directory
2. Run clinfo with `LD_PRELOAD=path/to/libOpenCL.so clinfo`

Any program which relies on the ICD loader (any program linking in `libOpenCL.so` and not directly `libVC4CL.so`) will need to be started with the `LD_PRELOAD` environment variable.

kernhanda, 2019-03-28:
Thanks! I'll give this a shot.

kernhanda, 2019-03-28:
Alright, with `MOCK_HAL` enabled, it seems that the kernel ran to almost the end, but then some unknown error happened.

Compared to running directly on the device, with emulation, all work groups ran.

Any other ideas about what could be going on here?

Snipped output below (removed the middle part that was repetitive, full output [here](https://gist.github.com/kernhanda/3204874ebdcafa72f558756de096d8cf))
```
pi@raspberrypi:~/gpu-on-pi/opencl_dot_product$ sudo LD_PRELOAD=$PWD/../VC4CL/build/src/libOpenCL.so ./dot_product
On the host, the dot product took 0.000456 seconds.
[VC4CL](    dot_product): API call: cl_int clGetPlatformIDs(cl_uint 1, cl_platform_id* 0x7ee5b9c0, cl_uint* 0)
[VC4CL](    dot_product): API call: cl_int clGetDeviceIDs(cl_platform_id 0xad15bc, cl_device_type 4, cl_uint 1, cl_device_id* 0x7ee5b9bc, cl_uint* 0)
[VC4CL](    dot_product): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0x7ee6e62c, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0x7ee5b8e8, void* 0, cl_int* 0x7ee6e620)
[VC4CL](    dot_product): Tracking live-time of object: cl_context
[VC4CL](    dot_product): API call: cl_int clGetDeviceInfo(cl_device_id 0xad15cc, cl_device_info 4100, size_t 4, void* 0x7ee6e628, size_t* 0)
[VC4CL] base=0x1f6db000, mem=(nil)
[VC4CL] V3D base: 0xeef
[VC4CL](    dot_product): API call: cl_program clCreateProgramWithSource(cl_context 0xad1d4c, cl_uint 1, const char** 0x7ee5b9b8, const size_t* 0x7ee5b9b4, cl_int* 0x7ee5b9ac)
[VC4CL](    dot_product): Tracking live-time of object: cl_program
[VC4CL](    dot_product): API call: cl_int clBuildProgram(cl_program 0xad322c, cl_uint 0, const cl_device_id* 0, const char* (null), void(CL_CALLBACK*)(cl_program program, void* user_data) 0x7ee5b990, void* 0)
[VC4CL](    dot_product): Precompiling source with:
[VC4CL](    dot_product): Dumping program sources to /tmp/vc4cl-source-1863408870.cl
[VC4CL](    dot_product): Precompilation complete with status: 0
[VC4CL](    dot_product): Compilation log: [W] Thu Mar 28 20:50:32 2019: Warnings in precompilation:
[W] Thu Mar 28 20:50:32 2019: <stdin>:22:1: warning: null character ignored
<U+0000>
^
1 warning generated.

[W] Thu Mar 28 20:50:33 2019: Warnings in precompilation:
[W] Thu Mar 28 20:50:33 2019:

[VC4CL](    dot_product): Compiling source with:
[VC4CL](    dot_product): Compilation complete with status: 0
[VC4CL](    dot_product): Dumping program sources to /tmp/vc4cl-binary-1149089301.bin
[VC4CL](    dot_product): API call: cl_kernel clCreateKernel(cl_program 0xad322c, const char* "dot_product", cl_int* 0x7ee6e620)
[VC4CL](    dot_product): Tracking live-time of object: cl_kernel
[VC4CL](    dot_product): API call: cl_mem clCreateBuffer(cl_context 0xad1d4c, cl_mem_flags 36, size_t 38400, void* 0x7ee64ff8, cl_int* 0x7ee6e620)
[VC4CL] Mailbox file descriptor opened: 48879
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00030012 00000008 00000004 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00010006 00000008 00000000 00000000 00800000 00000000
[VC4CL](    dot_product): Mailbox request: succeeded
[VC4CL](    dot_product): Tracking live-time of object: cl_mem
[VC4CL](    dot_product): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00009600 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 00000004 00000001 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000001 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 0003000d 00000008 00000004 00000000 00000000 00000000
[VC4CL] base=0x0, mem=0xaef2b0
[VC4CL](    dot_product): Allocated 38400 bytes of buffer: handle 1, device address 0x0, host address 0xaef2b0
[VC4CL](    dot_product): API call: cl_mem clCreateBuffer(cl_context 0xad1d4c, cl_mem_flags 33, size_t 38400, void* 0x7ee5b9f8, cl_int* 0x7ee6e620)
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 00010006 00000008 00000000 00000000 00800000 00000000
[VC4CL](    dot_product): Mailbox request: succeeded
[VC4CL](    dot_product): Tracking live-time of object: cl_mem
[VC4CL](    dot_product): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00009600 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 00000004 00000002 00001000 0000000c 00000000
[VC4CL](    dot_product): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000002 00000000 00000000
[VC4CL](    dot_product): Mailbox buffer after: 00000020 80000000 0003000d 00000008 00000004 00800000 00000000 00000000
[VC4CL] base=0x800000, mem=0xb05b30
[VC4CL](    dot_product): Allocated 38400 bytes of buffer: handle 2, device address 0x800000, host address 0xb05b30
[VC4CL](    dot_product): API call: cl_mem clCreateBuffer(cl_context 0xad1d4c, cl_mem_flags 2, size_t 800, void* 0, cl_int* 0x7ee6e620)
[... cut ...]
[VC4CL](    dot_product): Deallocated 800 bytes of buffer: handle 3, device address 0x1000000, host address 0xb346b8
[VC4CL](    dot_product): API call: cl_int clReleaseKernel(cl_kernel 0xb34b8c)
[VC4CL](    dot_product): API call: cl_int clReleaseCommandQueue(cl_command_queue 0xad1dc4)
[VC4CL](    dot_product): API call: cl_int clReleaseProgram(cl_program 0xad322c)
[VC4CL](    dot_product): API call: cl_int clReleaseContext(cl_context 0xad1d4c)
terminate called without an active exception
Aborted
```

doe300, 2019-03-29:
I tried it out myself too yesterday and it took a while but finished. Which is bad in this case, since it doesn't help to find the problem.

So from the `Dot product failed.` line, it looks like the result of the execution is wrong. I will look into this, but it still does not explain why the original execution failed to execute at all.

The `terminate called without an active exception` and `Aborted` occurs if a VC4CL internal thread had no time to shut down before the program exits, but this has no effect on the result. I will also try to fix this one.

One thing you could check:
On the Raspberry Pi (with native execution), does the time the execution takes before failing correspond to the timeout? E.g. if you increase/decrease the timeout in https://github.com/doe300/VC4CL/blob/master/src/executor.cpp#L28, does the actual time taken also increase/decrease?

kernhanda, 2019-03-29:
Your suspicions seem to be correct.

Without any changes:
```
pi@raspberrypi:~/gpu-on-pi/opencl_dot_product$ time sudo ./dot_product
<snip/>

real    0m34.003s
user    0m33.750s
sys     0m0.140s
```

Changing the 30s timeout to 60s:
```
pi@raspberrypi:~/gpu-on-pi/opencl_dot_product$ time sudo ./dot_product
<snip/>

real    1m3.891s
user    1m3.760s
sys     0m0.110s
```

Changing the 60s timeout to 120s:
```
pi@raspberrypi:~/gpu-on-pi/opencl_dot_product$ time sudo ./dot_product
<snip/>

real    2m3.949s
user    2m3.720s
sys     0m0.130s
```

kernhanda, 2019-04-03:
Ping. :) Any luck?

doe300, 2019-04-03:
Sorry, didn't have time to look into that.
Since the error is not reproducible on the emulator, I will have to take a look at the generated source code, but I'm not sure when I will have time to do that.

doe300, 2019-05-04:
So it looks like for a work-group size of 12 (the maximum), the execution hangs, probably somewhere in `barrier(...)`. For a work-group-size of less than 12, it at least passes, though the result is wrong on the Raspberry Pi, but correct on the emulator. This looks like there is a dead-lock/infinite stall somewhere in `barrier(...)` which the emulator does not catch...

A side note: `clGetEventProfilingInfo` can only be queried if the command-queue is created with `CL_QUEUE_PROFILING_ENABLE`.
Also you need to call `clReleaseEvent(prof_event)` on cleanup, this also removes the termination and abortion issue.

doe300, 2019-05-04:
So, turns out I did semaphore access wrong, that is why the execution timed out.
Now the execution passes, but the result seems to still be wrong.

rNoz, 2019-12-21:
Hi. 
First, thank you for you effort with this project. It is a massive complex work.

I tried to reproduce this bug, and I find that VC4CL in my Raspberry Pi 3 B+, has some float point precision errors.

I did 2 tests: dot_product (the same as this issue) and a basic saxpy program ( dest[i] = src[i] * 3.14 ). In both cases I get different values than in the host:

[0] Host: 3313.5571  Device: 3313.5571
[1] Host: 6579.5039  Device: 6579.5034
[FAILURE] at index 1:  6579.5039 != 6579.5034
[2] Host: 6708.3955  Device: 6708.3950
[FAILURE] at index 2:  6708.3955 != 6708.3950
(as you can see, in many indexes but not all)

The dot product has problems with local memory and computation. I don't know why. I checked your other issues but I didn't find a workaround. For example, if I do a simple operation, the final debug values are just wrong:

```
    __local float4 partial_dot[12]; // instead of __local in parameters
   if (BUG){
      partial_dot[lid] = a_vec[gid] + (float4)1.0f;
   }else{
      partial_dot[lid] = a_vec[gid];
   }
   output_debug[lid] = partial_dot[lid];
```

Without BUG:
[4] 40.00000
[5] 50.00000
[6] 60.00000
[7] 70.00000

With BUG (the same with multiplication, etc):
[4] -360777220096.00000
[5] -360777220096.00000
[6] -360777220096.00000
[7] -360777220096.00000

Results:
571.67017 (check: 19.49584)
683.61450 (check: 16.47419)
781.39270 (check: 16.03123)
661.78967 (check: 13.76786)
Total: 2698.46704 (check: 68.02542)
Dot product failed.

What can be wrong?

doe300, 2019-12-22:
The difference between `6579.5039` and `6579.5034` is 1 ULP, same for `6708.3955` and `6708.3950`. The problem here is probably, how the constant `3.14` gets stored (`3.140000` vs. `3.1399998`). Can you given me the full kernel code you used?

For the second part:
With `-DBUG=0`, the code is compiled to a simple memory copy and with `-DBUG=1` it is compiled to read-add-write. So yes, there seems to be something wrong with the lowered `__local` buffer.

rNoz, 2019-12-23:
I have created this repo: https://github.com/rNoz/opencl_embedded_tests

**Precision**

See the differences when using FACTOR=3.1415 or FACTOR=2.0.

It is the expected behavior? It is the first time I read about ULP. I checked that OpenCL is less restrictive (less precision) than CUDA, and I assume in Embedded Profile it is even further. Any advice regarding doing "unit test"/bench check for embedded? 
I was surprise the dot_product OpenCL repo has a final `if (abs(opencl_value - host_value) > 10.0)) failure()`, because 10.0 seems quite a lot of error for that calc.

```console
$ FACTOR=3.1415 VECTOR=12 CHECK=1 sudo -E ./build/saxpy saxpy.cl
vector_len: 12
check results: true
factor: 3.141500
using platform.device: 0.0
saxpy.cl
operation: saxpy
=== 1 OpenCL platform(s) found: ===
  -- 0 --
  PROFILE = EMBEDDED_PROFILE
  VERSION = OpenCL 1.2 VC4CL 0.4.9999
  NAME = OpenCL for the Raspberry Pi VideoCore IV GPU
  VENDOR = doe300
  EXTENSIONS = cl_khr_il_program cl_khr_spir cl_khr_create_command_queue cl_altera_device_temperature cl_altera_live_object_tracking cl_khr_icd cl_vc4cl_performance_counters
=== 2 OpenCL device(s) found on platform:
  -- 0 --
  DEVICE_NAME = VideoCore IV GPU
  DEVICE_VENDOR = Broadcom
  DEVICE_VERSION = OpenCL 1.2 VC4CL 0.4.9999
  DRIVER_VERSION = 0.4.9999
  DEVICE_MAX_COMPUTE_UNITS = 1
  DEVICE_MAX_CLOCK_FREQUENCY = 300
  DEVICE_GLOBAL_MEM_SIZE = 67108864
  DEVICE_MAX_WG_SIZE X=12,Y=12,Z=12
Creating context...
Creating command queue...
Creating program...
Building program from source...
attempting to create input buffer
attempting to create output buffer
attempting to create kernel
setting up kernel args cl_mem: 0x7f4800bc38
attempting to enqueue write buffer
attempting to enqueue kernel
Enqueue'd kerenel
time(ns):590937
Result:
[0] Host: 263.944977  Device: 263.944977
[1] Host: 123.895401  Device: 123.895393
[FAILURE] at index 1:  123.895401 != 123.895393
[2] Host: 246.010620  Device: 246.010605
[FAILURE] at index 2:  246.010620 != 246.010605
[FAILURE] at index 4:  286.394043 != 286.394012
[FAILURE] at index 6:  105.310226 != 105.310219
[FAILURE] at index 9:  174.029678 != 174.029663

computed 12 elements

$ FACTOR=2.0 VECTOR=12 CHECK=1 sudo -E ./build/saxpy saxpy.cl
vector_len: 12
check results: true
factor: 2.000000
using platform.device: 0.0
saxpy.cl
operation: saxpy
=== 1 OpenCL platform(s) found: ===
  -- 0 --
  PROFILE = EMBEDDED_PROFILE
  VERSION = OpenCL 1.2 VC4CL 0.4.9999
  NAME = OpenCL for the Raspberry Pi VideoCore IV GPU
[... 131 lines cut ...]
operation: vecmul
max wg size: 12
[0] OpenCL (2.00000) Host (2.00000)
[1] OpenCL (8.00000) Host (8.00000)
[2] OpenCL (18.00000) Host (18.00000)
[3] OpenCL (32.00000) Host (32.00000)
[20] OpenCL (882.00000) Host (882.00000)
[21] OpenCL (968.00000) Host (968.00000)
[22] OpenCL (1058.00000) Host (1058.00000)
[23] OpenCL (1152.00000) Host (1152.00000)
Everything seems to work fine!
```

It is interesting (but chaotic):
- Two executions doesn't produce the same results (previously was Ok, now it Fails, but eventually it will work in some execution - eg. tested with random executions of 12,24,128,1024)
- It fails at a random index
- If the length is small, it continues working.

Maybe you know what is going on here.

**Local memory**

I understood that is a bug and you will work on it. If you need any more test from my side, please, let me know. Also, I don't know if the kernel/device/building process would affect.

Thank you.

doe300, 2019-12-23:
Thanks for investing time to better analyse this error.

> See the differences when using FACTOR=3.1415 or FACTOR=2.0.

The difference here is that `2.0` can be exactly represented as `float`, while `3.1415` cannot. So if or `3.1415` the VC4C compiler chooses the wrong of the two adjacent actually representable values, the result will be off by this one ULP (which then can expand to a larger error depending on the calculations done on the value).

I don't know what is going wrong with the indeterministic results, I have to look further into that.

doe300, 2019-12-23:
So I can partially reproduce the behavior you are seeing:
Up to 10240, everything is okay, for 102400 I get a lot of errors.
When I get back to <=10240 samples, the successful results are reported again, so I don't get you indeterministic behavior.
The pattern of the wrong values also seems to suggest that there is a off-by-one-ULP error somewhere.

rNoz, 2019-12-23:
What can I do to help detecting the issue? Maybe executing low-level tests regarding memory? float ops? 
Do you think the distro (Manjaro Arm), kernel version (latest 4.19), device (Rpi 3B+) or anything in the building process (VC4CL opts) would affect?
If you provide me more advices or guidelines I can do further tests.

doe300, 2019-12-23:
So for the float multiplication, I did some testing and it looks like the `fmul` instruction does different rounding than the host CPU. I will have to try to find out which rounding is done by the CPU/GPU, which rounding is allowed by OpenCL and how to round correctly (if we have to change something).

No, I don't think the build process has anything to do with that.

About the memory issue: Did you run the code with the latest VC4C/VC4CL version? Or which version are you using?

doe300, 2019-12-24:
So it very much looks like the CPU uses the "normal" IEEE 754 round-to-nearest-even rounding mode, which is also the default rounding modes for the OpenCL 1.2 full profile.
The `fmul` operation seems to use the IEEE 754 round-to-zero rounding mode, which is why for some calculations, the absolute value of the result for the GPU is 1 ULP less than the CPU.
According to the OpenCL 1.2 standard, for the embedded profile (which is the one supported by the VC4CL implementation) an implementation is allowed to either use the round-to-nearest-even or round-to-zero rounding modes, so the floating point multiplication is **correct according to the specification**.

So to test for correctness, instead of the equality comparison a  comparison with 1 ULP of allowed error has to be used.

rNoz, 2019-12-24:
Thank you for your time.

I contribute with further tests:
- Rpi3B: Raspbian Stretch fully updated, VC4CL nightly build installed (Wiki, deb packages from circleci). 
- Rpi3B+: Manjaro Arm fully updated, VC4CL from git, built and installed with -DCMAKE_BUILD_TYPE=Release but also with ` -DCMAKE_BUILD_TYPE=Debug -DBUILD_DEBUG=On -DBUILD_TESTING=On -DCMAKE_VERBOSE_MAKEFILE=On` to dump logs.

They differ partially. **Important note**: I started writting this issue 6h ago, but I wanted to be completely sure everything I wrote is correct, so, I re-run again the experiments.... oh my bad. Now, they differ, just like yesterday with the non deterministic behavior. Finally, I have decided to skip the inconsistent results from this morning and I will explore further.

The main thing was that this morning I saw how it gave failures at the 11584 index, but I cannot reproduce anymore. I write the two sections with the only relevant (bus error, ULP):

### 6h ago experiments

Rpi3B:

```console
$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 2 )) sudo -E ./build/vectors vecmul.cl
vector: 11586
...
[11584] OpenCL (268424448.00000) Host (268424448.00000)
[11585] OpenCL (268470784.00000) Host (268470784.00000)
Everything seems to work fine!

$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 3 )) sudo -E ./build/vectors vecmul.cl
...
[11585] OpenCL (268470784.00000) Host (268470784.00000)
[11586] OpenCL (268517120.00000) Host (268517152.00000)
[FAILURE] [11586] OpenCL (268517120.00000) Host (268517152.00000)

$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 4 )) sudo -E ./build/vectors vecmul.cl
vector: 11588
...
[11586] OpenCL (268517120.00000) Host (268517152.00000)
[FAILURE] [11586] OpenCL (268517120.00000) Host (268517152.00000)
[11587] OpenCL (268563488.00000) Host (268563488.00000)
```

Rpi3B+:

```console
$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 )) sudo -E ./build/vectors vecmul.cl
vector: 11584
...
[11582] OpenCL (268331776.00000) Host (268331776.00000)
[11583] OpenCL (268378112.00000) Host (268378112.00000)
Everything seems to work fine!

$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 1)) sudo -E ./build/vectors vecmul.cl
vector: 11585
check results: true
using platform.device: 0.0
operation: vecmul
max wg size: 12
[1]    1802 bus error  CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 1)) sudo -E  vecmul.c

$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 2 )) sudo -E ./build/vectors vecmul.cl
vector: 11586
...
[11584] OpenCL (268424448.00000) Host (268424448.00000)
[11585] OpenCL (268470784.00000) Host (268470784.00000)
Everything seems to work fine!

$ CHECK=1 VECTOR=$(( 10240 * 1 + 512 * 2 + 128 * 2 + 64 + 4 )) sudo -E ./build/vectors vecmul.cl
vector: 11588
...
[11585] OpenCL (268470784.00000) Host (268470784.00000)
[11586] OpenCL (268517120.00000) Host (268517152.00000)
[FAILURE] [11586] OpenCL (268517120.00000) Host (268517152.00000)
[11587] OpenCL (268563488.00000) Host (268563488.00000)
```

[... 161 lines cut ...]
>   Core Temperature (Altera)                       54 C
53c54
<   Global memory size                              79691776 (76MiB)
---
>   Global memory size                              67108864 (64MiB)
55c56
<   Max memory allocation                           79691776 (76MiB)
---
>   Max memory allocation                           67108864 (64MiB)
64c65
<   Local memory size                               79691776 (76MiB)
---
>   Local memory size                               67108864 (64MiB)
66c67
<   Max constant buffer size                        79691776 (76MiB)
---
>   Max constant buffer size                        67108864 (64MiB)
87a89
>
```

My next test will be to install a full raspbian in the Rpi3B+ to discard issues with Manjaro Arm or its config files/drivers.
Also, I will try again the local memory tests `__local`.

I have a question for you, since I was having a look to some parts of your code. How did you do this massive work? I couldn't find any funding/project info. Do you work for free on VC4CL? Or did you extracted this as an open-source project but it is backed up/funded by real industry/projects where you get your salary/money from? To me, it seems a full-time job that you carried out for 2 years...

doe300, 2019-12-24:
> You refer to fmul but in the Built-in Functions it does not appear. Maybe I miss something.

You didn't miss something, I did. The floating point multiplication has to be correctly rounded (0 ULP), similar to the addition and subtraction according to the OpenCL 1.2 specification. I will update the wiki page with the correct allowed and the newly tested behavior (e.g. rounding mode).

> 1. When it fetches the values, what is the value that the GPU "sees"?

The GPU sees (as values read from memory) the values you give into the buffer, they are only memory copied, so no conversion there.

> 2. You specify the relative ULP is 2^-24. Should I apply it to the fetched values and then to the multiplication itself?

Its the error of the operation (or of the result of the operation).

> 3. Imagine that instead of a multiplication I want to do the division. I see x / y allowed ULP = 3. How can be applied? I assume I need to propagate the errors for every operation I add inside the [...]

About the huge run-time difference: Maybe the debug build is so much slower? But the difference is really huge, so I am not sure this is the sole reason.

Yes, the total error of e.g. an algorithm is calculated by adding up the allowed/actual errors of the operations performed.
So having 2 multiplications gives you an allowed error of (2 * 0 =) 0 ULP. And having a multiplication and an exp gives you (0 + 4 = ) 4 ULP allowed error.

> How did you do this massive work? I couldn't find any funding/project info. Do you work for free on VC4CL?

It started out as the programming part of my Masters Thesis pretty much 3 years ago, 2 years ago I published it first on github with the finalization of my thesis and from then I basically continue it as hobby. I sadly do not get paid for it and have to work an actual job, this is also why sometimes the progress is slower than I hope ;)