doe300/VC4CL #69: Error creating makefile

seyo-IV, 2019-04-02:
Linux: Kali Architecture: ARMv6l

root@kali:/home/seyo/VC4CL# make
[  3%] Built target generate_icd
[ 78%] Built target VC4CL
[ 82%] Linking CXX executable v3d_profile
/usr/bin/ld: ../src/libVC4CL.so.0.4.9999: undefined reference to `bcm_host_deinit'
/usr/bin/ld: ../src/libVC4CL.so.0.4.9999: undefined reference to `bcm_host_init'
/usr/bin/ld: ../src/libVC4CL.so.0.4.9999: undefined reference to `bcm_host_get_peripheral_address'                                                                collect2: error: ld returned 1 exit status
make[2]: *** [tools/CMakeFiles/v3d_profile.dir/build.make:88: tools/v3d_profile] Error 1
make[1]: *** [CMakeFiles/Makefile2:214: tools/CMakeFiles/v3d_profile.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

doe300, 2019-04-02:
The Raspberry Pi host libraries are missing on your machine, see https://github.com/doe300/VC4CL/issues/53#issuecomment-457687444.

seyo-IV, 2019-04-03:
Okay i haven't been sucessfull yet. I'l try installing raspian on the raspberry, then i should have all the librares.

seyo-IV, 2019-04-04:
Successfully installed everything on my pi with Kali. Still get "CL_PLATFORM_NOT_FOUND_KHR" when I'm using hashcat.

doe300, 2019-04-04:
Did you run the  program as root?

seyo-IV, 2019-04-08:
yes i was loged in as root

closed 2019-05-02 by seyo-IV