doe300/VC4CL #72: Runtime error when calling clBuildProgram, prompting symbol lookup error: /usr/local/lib/libVC4CL.so: undefined symbol: _ZN4vc4c11Precompiler10precompileERSiRSt10unique....

zhaodongmcw, 2019-05-15:
My machine is Raspberry PI 3 Model B Plus Rev 1.3.

I have installed VC4C, VC4CL and VC4CLStdLib according to the guide from https://github.com/doe300/VC4CL/wiki/How-to-get. I found the error when calling function clBuildProgram to build the .cl file on runtime, but I only use vc4c to compile the same .cl file, it is successful. The following is detailed error information of calling function clBuildProgram.

symbol lookup error: /usr/local/lib/libVC4CL.so: undefined symbol: _ZN4vc4c11Precompiler10precompileERSiRSt10unique_ptrISiSt14default_deleteISiEENS_13ConfigurationERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8OptionalISD_EESH_

However I find the symbol in libVC4CC.so and libVC4CL.so.

pi@raspberrypi:~ $ nm /usr/local/lib/libVC4CL.so | grep _ZN4vc4c11Precompiler10precompileERSiRSt10unique

         U _ZN4vc4c11Precompiler10precompileERSiRSt10unique_ptrISiSt14default_deleteISiEENS_13ConfigurationERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8OptionalISD_EESH_

pi@raspberrypi:~ $ nm /usr/local/lib/libVC4CC.so | grep _ZN4vc4c11Precompiler10precompileERSiRSt10unique

00842780 T _ZN4vc4c11Precompiler10precompileERSiRSt10unique_ptrISiSt14default_deleteISiEENS_13ConfigurationERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_8OptionalISD_EESJ_

zhaodongmcw, 2019-05-15:
The OS is Raspbian Stretch with desktop and recommended software from official website(https://www.raspberrypi.org/downloads/raspbian/).

zhaodongmcw, 2019-05-15:
pi@raspberrypi:~/workspace/test $ gcc -w -g -I/usr/include main.c -o main -lOpenCL -lVC4CL -lVC4CC -L/usr/local/lib
/usr/local/lib/libVC4CL.so：‘vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)’ undefined references
/usr/local/lib/libVC4CL.so：对‘vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)’undefined references
collect2: error: ld returned 1 exit status

I hope the following information can be useful.

CL_PLATFORM_VENDOR, vendor: doe300
GPU Name: VideoCore IV GPU

doe300, 2019-05-15:
Can't see the issue from the output.
Do you have the same problem, if you build the VC4CL library from source using the existing VC4C library to link against?
What GCC version are you running?

zhaodongmcw, 2019-05-16:
Thank you very much for your reply.

GCC version is:
pi@raspberrypi:~ $ gcc --version
gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I use the existing VC4CC and VC4CL library instead of building the VC4CL library from source code. I find the existing libVC4CL.so generated on 29th March, I will try to download the latest code from github and build libVC4CL to link the existing VC4CC library. Check whether there is still the same problem.

zhaodongmcw, 2019-05-21:
I have downloaded the VC4CL source code from github, commit NO 7ac88be. I build and install libVC4CL on Raspberry PI 3 B+, and successfully run OpenCL to link the existing VC4CC library. 

Thank you very much, doe300.

doe300, 2019-05-21:
Then I guess the latest VC4CL library is too old (since there haven't been any new commits in a while) and something changed in the VC4C public interface which made changes in the available symbols.  This should be fixed with the next commit on VC4CL.

closed 2019-06-01 by doe300