doe300/VC4CL #77: Problem with opencv 4.0.1 and openalpr

ant2n, 2019-06-11:
Hi, I compiled and installed the VC4CL on my raspberry pi 3 B+. My goal is to improve the openalpr library performance on my pi 3. I also compiled the opencv 4.0.1 and made the proper changes on the openalpr to run with this opencv version (small changes mostly on the names of some #define). Everything seems fine, the output of clinfo is ok. However, I'm having the following output when I execute my binary (my project has a .so with the openalpr and a binary that uses that library.):

```
sudo ./alpr_exe -c br -p br /home/pi/chapas/br/PJI7589.jpg 
i32 addrspace(3)* getelementptr inbounds ([97 x i32], [97 x i32] addrspace(3)* @runLBPClassifierStump.lstore, i32 0, i32 1)
OpenCL program build log: objdetect/cascadedetect
Status -11: CL_BUILD_PROGRAM_FAILURE
-D LOCAL_SIZE_X=8 -D LOCAL_SIZE_Y=8 -D SPLIT_STAGE=1 -D N_STAGES=18 -D MAX_FACES=10000 -D LBP
[E] Tue Jun 11 16:41:03 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x6ef34acc]
[E] Tue Jun 11 16:41:03 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::precalculateConstantExpression(vc4c::Module&, llvm::ConstantExpr const*)+0x394 [0x6f10d6b8]
[E] Tue Jun 11 16:41:03 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::toConstant(vc4c::Module&, llvm::Value const*)+0x558 [0x6f10cea8]
[E] Tue Jun 11 16:41:03 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::toValue(vc4c::Method&, llvm::Value const*)+0x540 [0x6f10c5b0]
[E] Tue Jun 11 16:41:03 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseInstruction(vc4c::Module&, vc4c::Method&, std::vector<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> >, std::allocator<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> > > >&, llvm::Instruction const&)+0xd8c [0x6f109898]
[E] Tue Jun 11 16:41:03 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseFunctionBody(vc4c::Module&, vc4c::Method&, std::vector<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> >, std::allocator<std::unique_ptr<vc4c::llvm2qasm::LLVMInstruction, std::default_delete<vc4c::llvm2qasm::LLVMInstruction> > > >&, llvm::Function const&)+0x174 [0x6f1085a8]
[E] Tue Jun 11 16:41:03 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parseFunction(vc4c::Module&, llvm::Function const&)+0x50c [0x6f108384]
[E] Tue Jun 11 16:41:03 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : vc4c::llvm2qasm::BitcodeReader::parse(vc4c::Module&)+0xf8 [0x6f106494]
[E] Tue Jun 11 16:41:03 2019:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::convert()+0xf4 [0x6ef35a98]
[E] Tue Jun 11 16:41:03 2019:  (10) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x220 [0x6ef36314]
[E] Tue Jun 11 16:41:03 2019:  (11) /usr/local/lib/libVC4CL.so : +0x56c20 [0x6f40ec20]
[E] Tue Jun 11 16:41:03 2019:  (12) /usr/local/lib/libVC4CL.so : vc4cl::Program::link(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, void (*)(_cl_program*, void*), void*, std::vector<vc4cl::Program*, std::allocator<vc4cl::Program*> > const&)+0x12c [0x6f414918]
[E] Tue Jun 11 16:41:03 2019:  (13) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x504 [0x6f414e6c]
[E] Tue Jun 11 16:41:03 2019:  (14) /usr/local/lib/libopencv_core.so.4.0 : +0x1a63c0 [0x76d333c0]
[E] Tue Jun 11 16:41:03 2019:  (15) /usr/local/lib/libopencv_core.so.4.0 : +0x1af1e0 [0x76d3c1e0]
[E] Tue Jun 11 16:41:03 2019:  (16) /usr/local/lib/libopencv_core.so.4.0 : +0x1ba7b4 [0x76d477b4]
[E] Tue Jun 11 16:41:03 2019:  (17) /usr/local/lib/libopencv_core.so.4.0 : cv::ocl::Program::create(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x70 [0x76d47c24]
[E] Tue Jun 11 16:41:03 2019:  (18) /usr/local/lib/libopencv_core.so.4.0 : cv::ocl::Program::Program(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x14 [0x76d47d98]
[E] Tue Jun 11 16:41:03 2019:  (19) /usr/local/lib/libopencv_core.so.4.0 : cv::ocl::Context::Impl::getProg(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x2f0 [0x76d48090]
[E] Tue Jun 11 16:41:03 2019:  (20) /usr/local/lib/libopencv_core.so.4.0 : cv::ocl::Context::getProg(cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)+0x24 [0x76d48378]
[E] Tue Jun 11 16:41:03 2019:  (21) /usr/local/lib/libopencv_core.so.4.0 : cv::ocl::Kernel::create(char const*, cv::ocl::ProgramSource const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)+0xa0 [0x76d48434]
[E] Tue Jun 11 16:41:03 2019:  (22) /usr/local/lib/libopencv_objdetect.so.4.0 : +0x163c4 [0x761a73c4]
[E] Tue Jun 11 16:41:03 2019:  (23) /usr/local/lib/libopencv_objdetect.so.4.0 : +0x18564 [0x761a9564]
[E] Tue Jun 11 16:41:03 2019:  (24) /usr/local/lib/libopencv_objdetect.so.4.0 : +0x18884 [0x761a9884]
[E] Tue Jun 11 16:41:03 2019:  (25) /usr/local/lib/libopencv_objdetect.so.4.0 : +0x18a30 [0x761a9a30]
[E] Tue Jun 11 16:41:03 2019:  (26) /usr/local/lib/libopencv_objdetect.so.4.0 : cv::CascadeClassifier::detectMultiScale(cv::_InputArray const&, std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > >&, double, int, int, cv::Size_<int>, cv::Size_<int>)+0x94 [0x761a66d0]
[E] Tue Jun 11 16:41:03 2019:  (27) /opt/fsa/chapas/libopenalpr_m.so : alpr::DetectorOCL::find_plates(cv::Mat, cv::Mat, cv::Size_<int>, cv::Size_<int>, bool&)+0x178 [0x760d0bcc]
[E] Tue Jun 11 16:41:03 2019:  (28) /opt/fsa/chapas/libopenalpr_m.so : alpr::Detector::detect(cv::Mat, cv::Mat, std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > >, bool&)+0x7a8 [0x760c98c0]
[E] Tue Jun 11 16:41:03 2019:  (29) /opt/fsa/chapas/libopenalpr_m.so : alpr::AlprImpl::analyzeSingleCountry(cv::Mat, cv::Mat, std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > >, bool&)+0x2e0 [0x760dadb4]
[E] Tue Jun 11 16:41:03 2019:  (30) /opt/fsa/chapas/libopenalpr_m.so : alpr::AlprImpl::recognizeFullDetails(cv::Mat, std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > >)+0x9f8 [0x760dcf50]
[E] Tue Jun 11 16:41:03 2019:  (31) /opt/fsa/chapas/libopenalpr_m.so : alpr::AlprImpl::recognize(cv::Mat, std::vector<cv::Rect_<int>, std::allocator<cv::Rect_<int> > >)+0xe4 [0x760df818]
[E] Tue Jun 11 16:41:03 2019:  (32) /opt/fsa/chapas/libopenalpr_m.so : alpr::AlprImpl::recognize(unsigned char*, int, int, int, std::vector<alpr::AlprRegionOfInterest, std::allocator<alpr::AlprRegionOfInterest> >)+0x1e4 [0x760dfa78]
[E] Tue Jun 11 16:41:03 2019:  (33) /opt/fsa/chapas/libopenalpr_m.so : alpr::Alpr::recognize(unsigned char*, int, int, int, std::vector<alpr::AlprRegionOfInterest, std::allocator<alpr::AlprRegionOfInterest> >)+0xd4 [0x760d2204]
[E] Tue Jun 11 16:41:03 2019:  (34) ./alpr_exe() [0x1e2fc]
[E] Tue Jun 11 16:41:03 2019:  (35) ./alpr_exe : main+0x1b1c [0x19d54]
[E] Tue Jun 11 16:41:03 2019:  (36) /lib/arm-linux-gnueabihf/libc.so.6 : __libc_start_main+0x114 [0x75cf0678]
[E] Tue Jun 11 16:41:03 2019: Compiler threw exception: Parser: Only constant getelementptr without offsets are supported for now: 

plate0: 2 results
    - PJI7589	 confidence: 85.8618	 pattern_match: 1
    - PJI7588	 confidence: 81.3419	 pattern_match: 1
terminate called without an active exception
Aborted 
```
I don't know what's wrong, it seems to be a problem with the LBP classifier ?

doe300, 2019-06-30:
Looks like the OpenCL code you are compiling uses some features which are not yet implemented in th VC4C compiler

TMekelenkamp, 2020-06-19:
I see that the issue is still open. I too have an issue regarding this post and it seems that @ant2n has fixed it somehow in the mean time. Maybe @ant2n could help me out on the alpr + opencv4 implementation, would love to get in touch 😄

ant2n, 2020-06-20:
Sadly I haven't solved the problem yet. I'm using the alpr without the VC4CL. 😕

doe300, 2020-06-20:
This particular issue should be fixed by now, see the linked commit. @TMekelenkamp, what kind of issues do you have?

TMekelenkamp, 2020-07-20:
Thanks for the replies! I'm stuck at the same point as @ant2n. Not as mentioned in this issue but with the ALPR package with opencv 4 to run on the openCL integration. And since I saw him working on an implementation of openALPR on a pi with openCL. And I figured this was the only way to contact him to maybe exchange info and work on it... 😅 

So maybe it would be better if I get in touch with @ant2n to maybe check if we can fix it. And then this issue can be closed 😄

closed 2021-02-09 by doe300