doe300/VC4CL #79: OpenCL on the VideoCore VI

Bobrow, 2019-06-25:
I was wondering as the rpi 4 has a VC6 GPU, would VC4CL work with it or does someone need to make a version for it?

nktl, 2019-06-25:
Just tried this, unfortunately precompiled dpkgs and build process itself are rather broken since Pi4 uses Debian Buster, which for instance does not have clang 3.9 any more

It does look like it will require some work - and who knows at this point how much VideoCore VI is different from IV...

Terminus-IMRC, 2019-06-26:
Although TMU, uniform, VPM and so forth is still there, the instruction set of VC6 has been renewed from VC4 and some new functions such as MMU is introduced. Some of the changes and instruction sets are open-sourced in [https://cgit.freedesktop.org/mesa/mesa/tree/src/broadcom](https://cgit.freedesktop.org/mesa/mesa/tree/src/broadcom), but a detailed reference manual is not published yet. Gordon Hollingworth says that it would be available [https://twitter.com/gsholling/status/1143069972106817536](https://twitter.com/gsholling/status/1143069972106817536).

doe300, 2019-06-26:
> Although TMU, uniform, VPM and so forth is still there, the instruction set of VC6 has been renewed from VC4 and some new functions such as MMU is introduced.

Exactly. I don't know that much details about the VC6, but at least the VC5 (from what I can see from the mesa driver) has a lot of changes which make it a lot easier to work with but would also need a lot of rewriting/additional support.

closed 2019-07-21 by doe300

hagom, 2021-07-27:
What's the status of the VC6 support for pi4 ?

doe300, 2021-07-27:
There won't be any, at least not via this library. For the VideoCore VI, probably your best chance is the official mesa graphics library and either use OpenGL/Vulkan directly or try one of https://github.com/google/clspv or https://github.com/kpet/clvk as wrappers.

hagom, 2021-07-29:
this is sad because the pi 4 has more than a year and it is incredible that the SBC doesn't have fully support for the GPU

commit c3f2da18 (2019-07-21): Adds non-support for Raspberry Pi 4

Fixes #79, #80