doe300/VC4CL #80: Error Building Kernel.cpp.o on Raspberry Pi 4

BoxWizard000000, 2019-07-21:
I'm trying a compile on raspberry pi 4. It seems to find all libraries fine; however, it errors out at this phase. Here's the log.

```
/home/pi/Downloads/VC4CL/src/icd_loader.cpp:200:5: error: invalid conversion from ‘_cl_command_queue* (*)(cl_context, cl_device_id, const cl_queue_properties_khr*, cl_int*)’ {aka ‘_cl_command_queue* (*)(_cl_context*, _cl_device_id*, const long long unsigned int*, int*)’} to ‘cl_int (*)()’ {aka ‘int (*)()’} [-fpermissive]
     &VC4CL_clCreateCommandQueueWithPropertiesKHR, /* clCreateCommandQueueWithProperties */
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/Downloads/VC4CL/src/icd_loader.cpp:223:5: error: invalid conversion from ‘_cl_program* (*)(cl_context, const void*, size_t, cl_int*)’ {aka ‘_cl_program* (*)(_cl_context*, const void*, unsigned int, int*)’} to ‘cl_int (*)()’ {aka ‘int (*)()’} [-fpermissive]
     &VC4CL_clCreateProgramWithILKHR, /* clCreateProgramWithIL */
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/VC4CL.dir/build.make:180: src/CMakeFiles/VC4CL.dir/icd_loader.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/pi/Downloads/VC4CL/src/Kernel.cpp:7:
/home/pi/Downloads/VC4CL/src/Kernel.h:54:12: warning: function might be candidate for attribute ‘const’ [-Wsuggest-attribute=const]
     struct KernelArgument
            ^~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:155: src/CMakeFiles/VC4CL.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
```

alanmikhak, 2019-07-21:
Please see minute 3:00 of video I uploaded to YouTube showing these two compiler errors on line 200 and 223 of icd_loader.cpp and how I resolved them by reinterpret_cast.

Raspberry Pi 4: Compiled OpenCL 1.2 VC4CL On Raspbian 10 (buster):
https://www.youtube.com/watch?v=YbzZgnz_uDU

doe300, 2019-07-21:
@alanmikhak's video looks to fix the compilation error, not sure though whether it is also logically correct...

**Anyway, VC4CL won't work on the Raspberry pi 4 at all, since it has a different GPU!** See #79

alanmikhak, 2019-07-22:
I also observe the same two compiler errors on my Raspberry Pi 3 Model B+ running Raspbian 10 (buster).

[ 32%] Building CXX object src/CMakeFiles/VC4CL.dir/executor.cpp.o
/home/pi/src/github.com/doe300/VC4CL/src/executor.cpp: In function ‘unsigned int AS_GPU_ADDRESS(const unsigned int*, vc4cl::DeviceBuffer*)’:
/home/pi/src/github.com/doe300/VC4CL/src/executor.cpp:40:24: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     const char* tmp = *reinterpret_cast<const char**>(&ptr);
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 35%] Building CXX object src/CMakeFiles/VC4CL.dir/extensions.cpp.o
[ 39%] Building CXX object src/CMakeFiles/VC4CL.dir/icd_loader.cpp.o
/home/pi/src/github.com/doe300/VC4CL/src/icd_loader.cpp:200:5: error: invalid conversion from ‘_cl_command_queue* (*)(cl_context, cl_device_id, const cl_queue_properties_khr*, cl_int*)’ {aka ‘_cl_command_queue* (*)(_cl_context*, _cl_device_id*, const long long unsigned int*, int*)’} to ‘cl_int (*)()’ {aka ‘int (*)()’} [-fpermissive]
     &VC4CL_clCreateCommandQueueWithPropertiesKHR, /* clCreateCommandQueueWithProperties */
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/src/github.com/doe300/VC4CL/src/icd_loader.cpp:223:5: error: invalid conversion from ‘_cl_program* (*)(cl_context, const void*, size_t, cl_int*)’ {aka ‘_cl_program* (*)(_cl_context*, const void*, unsigned int, int*)’} to ‘cl_int (*)()’ {aka ‘int (*)()’} [-fpermissive]
     &VC4CL_clCreateProgramWithILKHR, /* clCreateProgramWithIL */
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/VC4CL.dir/build.make:180: src/CMakeFiles/VC4CL.dir/icd_loader.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:188: src/CMakeFiles/VC4CL.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

$ cd ~

$ uname -a
Linux rpi7 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

$ clang --version
clang version 7.0.1-8+rpi2 (tags/RELEASE_701/final)
Target: armv6k-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin

$ gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cmake --version
cmake version 3.13.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ free -h
              total        used        free      shared  buff/cache   available
Mem:          874Mi       191Mi       236Mi       9.0Mi       445Mi       612Mi
Swap:         2.0Gi        29Mi       2.0Gi

doe300, 2019-07-22:
Can you give me the versions of the following debian packages that are installed on your system?

- ocl-icd-opencl-dev
- ocl-icd-libopencl1

alanmikhak, 2019-07-22:
$ sudo apt install ocl-icd-opencl-dev ocl-icd-libopencl1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ocl-icd-libopencl1 is already the newest version (2.2.12-2).
ocl-icd-libopencl1 set to manually installed.
ocl-icd-opencl-dev is already the newest version (2.2.12-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

commit cbccf311 (2019-07-22): Fixes some clang compiler warnings and errors

Fixes #80
closed 2019-07-22 by doe300