doe300/VC4CL #81: Hashcat problems on Raspberry Pi 3 with VC4CL

lingylol, 2019-08-03:
When I try to run any Hashcat attack on a raspberry pi 3 with VC4CL, I get the error `clBuildProgram(): CL_COMPILE_PROGRAM_FAILURE`, I have installed VC4CL according to here, https://github.com/aaguiar96/VC4CL-Install. Can anybody help me figure out what went wrong?
```pi@raspberrypi:~/hashcat $ sudo hashcat -m 0 -a 3 hash.txt
hashcat (v5.1.0-1321-ge74396d5) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 76/76 MB allocatable, 1MCU

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates

Applicable optimizers:
* Zero-Byte
* Early-Skip
* Not-Salted
* Not-Iterated
* Single-Hash
* Single-Salt
* Brute-Force
* Raw-Hash

Minimum password length supported by kernel: 0
Maximum password length supported by kernel: 256

ATTENTION! Pure (unoptimized) backend kernels selected.
Using pure kernels enables cracking longer passwords but for the price of drastically reduced performance.
If you want to switch to optimized backend kernels, append -O to your commandline.
See the above message to find out about the exact limits.

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

clBuildProgram(): CL_COMPILE_PROGRAM_FAILURE

[E] Fri Aug  2 19:27:45 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xb1 [0x7235faa2]
[E] Fri Aug  2 19:27:45 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x725c88 [0x72586c88]
[E] Fri Aug  2 19:27:45 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::findStandardLibraryFiles(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x5b [0x72586e40]
[E] Fri Aug  2 19:27:45 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLToLLVMIR(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x23 [0x725800dc]
[E] Fri Aug  2 19:27:45 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x4c7 [0x72587870]
[E] Fri Aug  2 19:27:45 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x121 [0x72585320]
[E] Fri Aug  2 19:27:45 2019:  (7) /usr/local/lib/libVC4CL.so : +0x444b2 [0x726ef4b2]
[E] Fri Aug  2 19:27:45 2019:  (8) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&, void (*)(_cl_program*, void*), void*)+0x8f [0x726efc78]
[E] Fri Aug  2 19:27:45 2019:  (9) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x11d [0x726f2ef6]

* Device #1: Kernel /usr/local/share/hashcat/OpenCL/m00000_a3-pure.cl build failed - proceeding without this device.

Started: Fri Aug  2 19:27:39 2019
Stopped: Fri Aug  2 19:27:45 2019
terminate called without an active exception
Aborted
```

doe300, 2019-08-03:
From the error location, it looks like the precompiled VC4CL-StdLib files were not found.

Can you list the contents of:

- /usr/local/include/vc4cl-stdlib/
- /usr/include/vc4cl-stdlib/
- ~/.cache/vc4c

lingylol, 2019-08-03:
In /usr/local/include/vc4cl-stdlib/ I have 
```
_async.h        defines.h      _integer.h     _overloads.h        VC4CLStdLib.h
_atomics.h      _extensions.h  _intrinsics.h  _printf.h           _vector.h
_common.h       _flags.h       _math.h        _relational.h       _work_items.h
_config.h       _geometric.h   opencl-c.h     _spir_mangling.h
_conversions.h  _images.h      _operators.h   _synchronization.h
```
However I am missing the directories

/usr/include/vc4cl-stdlib/,
~/.cache/vc4c 
Do I need to copy some files over?

doe300, 2019-08-03:
No you don't need to. But you should have these two files also in `/usr/local/include/vc4cl-stdlib/`:

- VC4CLStdLib.h.pch
- VC4CLStdLib.bc

Can you rerun the `make install ` steps for `vc4cl-stdlib` and `vc4c` and post the output?

lingylol, 2019-08-03:
For vc4cl-stdlib I get
```
[100%] Built target VC4CLStdLib
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/include/vc4cl-stdlib
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_synchronization.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_math.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_operators.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_vector.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_atomics.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_geometric.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_printf.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_integer.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_common.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_relational.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/opencl-c.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_config.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_overloads.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_spir_mangling.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/defines.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_images.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_async.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_conversions.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_flags.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_intrinsics.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_work_items.h
-- Up-to-date: /usr/local/include/vc4cl-stdlib/_extensions.h
```
For vc4c I get
```
[  1%] Performing update step for 'variant-project'
From https://github.com/mpark/variant
 * branch                HEAD       -> FETCH_HEAD
Already up to date.
[  1%] Performing configure step for 'variant-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VC4CLStdLib/VC4C/variant
[  2%] Performing build step for 'variant-project'
[  4%] Built target variant-project-build
[  6%] Performing update step for 'vc4asm-project'
From https://github.com/maazl/vc4asm
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[  7%] Performing configure step for 'vc4asm-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VC4CLStdLib/VC4C/vc4asm/src/vc4asm-project-build
[  7%] Performing build step for 'vc4asm-project'
[ 22%] Built target objvc4asm
[ 45%] Built target objvc4dis
[ 72%] Built target objvc4base
[ 81%] Built target vc4dis
[ 90%] Built target vc4asm
[ 95%] Built target libvc4asm-static
[100%] Built target libvc4asm-shared
[  9%] Built target vc4asm-project-build
[ 11%] Performing update step for 'cpplog-project'
From https://github.com/doe300/cpplog
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[ 12%] Performing configure step for 'cpplog-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VC4CLStdLib/VC4C/cpplog
[ 12%] Performing build step for 'cpplog-project'
[ 50%] Built target cpplog-static
[100%] Built target cpplog
[ 14%] Built target cpplog-project-build
[ 76%] Built target VC4CC
[ 79%] Built target VC4C
[ 80%] Built target qpu_emulator
[ 80%] Performing update step for 'cpptest-lite-project'
From https://github.com/doe300/cpptest-lite
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[ 81%] Performing configure step for 'cpptest-lite-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/VC4CLStdLib/VC4C/cpptest-lite/src/cpptest-lite-project-build
[ 82%] Performing build step for 'cpptest-lite-project'
[100%] Built target cpptest-lite
[ 85%] Built target cpptest-lite-project-build
[100%] Built target TestVC4C
Install the project...
-- Install configuration: "Debug"
-- Up-to-date: /usr/local/lib/libVC4CC.so.0.4.9999
-- Up-to-date: /usr/local/lib/libVC4CC.so.1.2
-- Up-to-date: /usr/local/lib/libVC4CC.so
-- Up-to-date: /usr/local/bin/vc4c
-- Up-to-date: /usr/local/share/vc4cc/VC4CC-targets.cmake
-- Installing: /usr/local/share/vc4cc/VC4CC-targets-debug.cmake
-- Up-to-date: /usr/local/include/vc4cc
-- Up-to-date: /usr/local/include/vc4cc/config.h
-- Up-to-date: /usr/local/include/vc4cc/Optional.h
-- Up-to-date: /usr/local/include/vc4cc/tools.h
-- Up-to-date: /usr/local/include/vc4cc/Precompiler.h
-- Up-to-date: /usr/local/include/vc4cc/CompilationError.h
-- Up-to-date: /usr/local/include/vc4cc/Compiler.h
-- Up-to-date: /usr/local/include/vc4cc/VC4C.h
```

lingylol, 2019-08-03:
I also get errors when trying to compile the example. 
```
vc4c --hex /home/pi/VC4CLStdLib/VC4C/example/hello_world.cl
[D] Sat Aug  3 11:53:55 2019: Compiling '/home/pi/VC4CLStdLib/VC4C/example/hello_world.cl' into '/home/pi/VC4CLStdLib/VC4C/example/hello_world.cl.hex' with optimization level 2 and options '' ...
[D] Sat Aug  3 11:53:55 2019: Temporary file '/tmp/vc4c-BhugNj' created
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /usr/local/include/vc4cl-stdlib//VC4CLStdLib.bc (No such file or directory)
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /usr/include/vc4cl-stdlib//VC4CLStdLib.bc (No such file or directory)
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /home/pi/.cache/vc4c/VC4CLStdLib.bc (No such file or directory)
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /usr/local/include/vc4cl-stdlib//VC4CLStdLib.h.pch (No such file or directory)
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /usr/include/vc4cl-stdlib//VC4CLStdLib.h.pch (No such file or directory)
[D] Sat Aug  3 11:53:55 2019: Could not access VC4CL standard-library file: /home/pi/.cache/vc4c/VC4CLStdLib.h.pch (No such file or directory)
[E] Sat Aug  3 11:53:56 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xb1 [0x76c14aa2]
[E] Sat Aug  3 11:53:56 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x725c88 [0x76e3bc88]
[E] Sat Aug  3 11:53:56 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::findStandardLibraryFiles(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&)+0x5b [0x76e3be40]
[E] Sat Aug  3 11:53:56 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLToLLVMIR(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x23 [0x76e350dc]
[E] Sat Aug  3 11:53:56 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x4c7 [0x76e3c870]
[E] Sat Aug  3 11:53:56 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x121 [0x76e3a320]
[E] Sat Aug  3 11:53:56 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Compiler::compile(std::istream&, std::ostream&, vc4c::Configuration const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0xa7 [0x76c15a44]
[E] Sat Aug  3 11:53:56 2019:  (8) vc4c : main+0xc31 [0x429dd6]
[E] Sat Aug  3 11:53:56 2019:  (9) /lib/arm-linux-gnueabihf/libc.so.6 : __libc_start_main+0x10c [0x7310f718]
[D] Sat Aug  3 11:53:56 2019: Temporary file '/tmp/vc4c-BhugNj' deleted
[E] Sat Aug  3 11:53:56 2019: Compiler threw exception: Pre-compilation: Required VC4CL standard library file not found in any of the provided paths: /usr/local/include/vc4cl-stdlib/, /usr/include/vc4cl-stdlib/, /home/pi/.cache/vc4c
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Required VC4CL standard library file not found in any of the provided paths: /usr/local/include/vc4cl-stdlib/, /usr/include/vc4cl-stdlib/, /home/pi/.cache/vc4c
Aborted
```
It seems I have it configured incorrectly somehow.

doe300, 2019-08-03:
Yes, this command generates the precompiled headers required:

    path/to/vc4c --precompile-stdlib -o /output/directory/ path/to/VC4CLStdLib/include/VC4CLStdLib.h

Example (when running from VC4C project directory:

    ./build/src/vc4c --precompile-stdlib -o /home/pi/.cache/vc4c/ ../VC4CLStdLib/include/VC4CLStdLib.h

This is also the command that should have been executed automatically. I will need to have a look as to why it was not run.

lingylol, 2019-08-03:
When I run that command I get the error 
```
 vc4c --precompile-stdlib -o /home/pi/.cache/vc4c/ ../VC4CLStdLib/include/VC4CLStdLib.h
[D] Sat Aug  3 13:15:45 2019: Pre-compiling '../VC4CLStdLib/include/VC4CLStdLib.h' into '/home/pi/.cache/vc4c/'...
[I] Sat Aug  3 13:15:45 2019: Pre-compiling standard library with: /usr/bin/clang -cc1 -triple spir-unknown-unknown -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -Wno-all -Wno-gcc-compat -Wdouble-promotion -Wno-undefined-inline -Wno-unknown-attributes -x cl -emit-pch -o /home/pi/.cache/vc4c//VC4CLStdLib.h.pch ../VC4CLStdLib/include/VC4CLStdLib.h
[E] Sat Aug  3 13:15:46 2019: Errors in precompilation:
[E] Sat Aug  3 13:15:46 2019: error: error reading '../VC4CLStdLib/include/VC4CLStdLib.h'
1 error generated.

[E] Sat Aug  3 13:15:46 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xb1 [0x76bdeaa2]
[E] Sat Aug  3 13:15:46 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x14f [0x76dfdcc4]
[E] Sat Aug  3 13:15:46 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x1b1 [0x76e060f2]
[E] Sat Aug  3 13:15:46 2019:  (4) vc4c : main+0x79d [0x443942]
[E] Sat Aug  3 13:15:46 2019:  (5) /lib/arm-linux-gnueabihf/libc.so.6 : __libc_start_main+0x10c [0x730d9718]
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: error: error reading '../VC4CLStdLib/include/VC4CLStdLib.h'
1 error generated.

Aborted
```

doe300, 2019-08-03:
Does the relative path `../VC4CLStdLib/include/VC4CLStdLib.h` point to the existing header file?

lingylol, 2019-08-03:
Yes, the file exists in that directory

doe300, 2019-08-04:
Does the user (`pi` I assume) has read rights on that file?
Does the file contain content of a C header?

lingylol, 2019-08-07:
Thank you, I reinstalled VC4C and now the files VC4CLStdLib.h.pch and VC4CLStdLib.bc are 
in the /home/pi/VC4CLStdLib/include folder. I can now run the `hello_world` example successfully, however, when running hashcat I get the error 
```
hashcat (v5.1.0-1321-ge74396d5) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 76/76 MB allocatable, 1MCU

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates

Applicable optimizers:
* Zero-Byte
* Early-Skip
* Not-Salted
* Not-Iterated
* Single-Hash
* Single-Salt
* Brute-Force
* Raw-Hash

Minimum password length supported by kernel: 0
Maximum password length supported by kernel: 256

ATTENTION! Pure (unoptimized) backend kernels selected.
Using pure kernels enables cracking longer passwords but for the price of drastically reduced performance.
If you want to switch to optimized backend kernels, append -O to your commandline.
See the above message to find out about the exact limits.

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

clBuildProgram(): CL_BUILD_PROGRAM_FAILURE

[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Wed Aug  7 09:40:12 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[... cut ...]
[E] Wed Aug  7 09:40:26 2019:  (41) /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 : +0x986ea [0x71dc16ea]
[E] Wed Aug  7 09:40:29 2019: Compiler threw exception: Optimizer: Copying dynamically sized memory within RAM is not yet implemented: copy i32 %tmp.6 entries from (g) i8* %scevgep5 into (p) i8* %tmp.0

* Device #1: Kernel /usr/local/share/hashcat/OpenCL/m00000_a3-pure.cl build failed - proceeding without this device.

Started: Wed Aug  7 09:38:01 2019
Stopped: Wed Aug  7 09:40:29 2019
terminate called without an active exception
Aborted
```

lingylol, 2019-08-09:
I managed to clear some errors by increasing the memory of the gpu, however I still get the error
```
hashcat (v5.1.0-1321-ge74396d5) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 448/448 MB allocatable, 1MCU

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates

Applicable optimizers:
* Zero-Byte
* Early-Skip
* Not-Salted
* Not-Iterated
* Single-Hash
* Single-Salt
* Brute-Force
* Raw-Hash

Minimum password length supported by kernel: 0
Maximum password length supported by kernel: 256

ATTENTION! Pure (unoptimized) backend kernels selected.
Using pure kernels enables cracking longer passwords but for the price of drastically reduced performance.
If you want to switch to optimized backend kernels, append -O to your commandline.
See the above message to find out about the exact limits.

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

clBuildProgram(): CL_COMPILE_PROGRAM_FAILURE

[E] Thu Aug  8 19:35:22 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xb1 [0x723b6d52]
[E] Thu Aug  8 19:35:22 2019:  (2) /usr/local/lib/libVC4CC.so.1.2 : +0x5539a8 [0x723e09a8]
[E] Thu Aug  8 19:35:22 2019:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::runProcess(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*, std::ostream*)+0x2c9 [0x723e1312]
[E] Thu Aug  8 19:35:22 2019:  (4) /usr/local/lib/libVC4CC.so.1.2 : runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x3d [0x725e4efa]
[E] Thu Aug  8 19:35:22 2019:  (5) /usr/local/lib/libVC4CC.so.1.2 : +0x758320 [0x725e5320]
[E] Thu Aug  8 19:35:22 2019:  (6) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLWithPCH(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0xa5 [0x725e561a]
[E] Thu Aug  8 19:35:22 2019:  (7) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::compileOpenCLToLLVMIR(vc4c::precompilation::PrecompilationSource<(vc4c::SourceType)1>&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::precompilation::PrecompilationResult<(vc4c::SourceType)3>&)+0x55 [0x725e6456]
[E] Thu Aug  8 19:35:22 2019:  (8) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::run(std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::SourceType, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >)+0x4c7 [0x725edbcc]
[E] Thu Aug  8 19:35:22 2019:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompile(std::istream&, std::unique_ptr<std::istream, std::default_delete<std::istream> >&, vc4c::Configuration, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, vc4c::Optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&)+0x121 [0x725eb668]
[E] Thu Aug  8 19:35:22 2019:  (10) /usr/local/lib/libVC4CL.so : +0x444b2 [0x7275a4b2]
[E] Thu Aug  8 19:35:22 2019:  (11) /usr/local/lib/libVC4CL.so : vc4cl::Program::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, vc4cl::object_wrapper<vc4cl::Program>, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, vc4cl::object_wrapper<vc4cl::Program> > > > const&, void (*)(_cl_program*, void*), void*)+0x8f [0x7275ac78]
[E] Thu Aug  8 19:35:22 2019:  (12) /usr/local/lib/libVC4CL.so : VC4CL_clBuildProgram+0x11d [0x7275def6]

* Device #1: Kernel /usr/local/share/hashcat/OpenCL/m00000_a3-pure.cl build failed - proceeding without this device.

Started: Thu Aug  8 19:35:14 2019
Stopped: Thu Aug  8 19:35:22 2019
terminate called without an active exception
Aborted
```

doe300, 2019-08-09:
> [E] Wed Aug  7 09:40:29 2019: Compiler threw exception: Optimizer: Copying dynamically sized memory within RAM is not yet implemented: copy i32 %tmp.6 entries from (g) i8* %scevgep5 into (p) i8* %tmp.0

This error is due to some feature not yet implemented in VC4C.

The other error I cannot really say without an error message, although it looks like it originates in the clang-frontend. Can you run the code in gdb, break on exceptions via `catch throw` and post the stack-trace and error message?

lingylol, 2019-08-09:
Sorry, I'm quite new to this. How do I run hashcat in gdb and find the errors?

doe300, 2019-08-09:
The command that you use to run hashcat.

Assuming you run hashcat with the program `hashcat`, then instead you execute `gdb hashcat` and in the interactive gdb shell, `catch throw` and then `run`. If the first exception is thrown, it will return into the interactive gdb shell where you can type `info stack` to get the stack trace. Also you probably want to go `up` to see the error message.

lingylol, 2019-08-09:
Here is the result I get from gdb
```
sudo gdb --args hashcat -a3 -m0 hash.txt
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hashcat...(no debugging symbols found)...done.
(gdb) catch throw
Catchpoint 1 (throw)
(gdb) run
Starting program: /home/pi/hashcat/hashcat -a3 -m0 hash.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
hashcat (v5.1.0-1321-ge74396d5) starting...

[New Thread 0x6cb56460 (LWP 10063)]
[New Thread 0x6c1ff460 (LWP 10064)]
[New Thread 0x6b9fe460 (LWP 10065)]
[New Thread 0x6b1fd460 (LWP 10066)]
[New Thread 0x6a9fc460 (LWP 10067)]
[Thread 0x6a9fc460 (LWP 10067) exited]
OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 512/512 MB allocatable, 1MCU

[New Thread 0x6a9fc460 (LWP 10068)]
Generated bitmap tables...[Thread 0x6a9fc460 (LWP 10068) exited]
[Thread 0x6b1fd460 (LWP 10066) exited]
[Thread 0x6b9fe460 (LWP 10065) exited]
[Thread 0x6c1ff460 (LWP 10064) exited]
[Thread 0x6cb56460 (LWP 10063) exited]

Program terminated with signal SIGKILL, Killed.
The program no longer exists.
```
`info stack` and `up` only return `No stack.`

lingylol, 2019-08-10:
What happens to me is interesting. When running hashcat, I get my second error from before. When I run it in gdb, I get the following
```
sudo gdb --args hashcat -a3 -m0 hash.txt
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hashcat...(no debugging symbols found)...done.
(gdb) catch throw
Catchpoint 1 (throw)
(gdb) run
Starting program: /home/pi/hashcat/hashcat -a3 -m0 hash.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
hashcat (v5.1.0-1321-ge74396d5) starting...

[New Thread 0x6cb56460 (LWP 1280)]
[New Thread 0x6c1ff460 (LWP 1281)]
[New Thread 0x6b9fe460 (LWP 1282)]
[New Thread 0x6b1fd460 (LWP 1283)]
[New Thread 0x6a9fc460 (LWP 1284)]
OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 448/448 MB allocatable, 1MCU

[Thread 0x6a9fc460 (LWP 1284) exited]
[New Thread 0x6a9fc460 (LWP 1285)]
Generated bitmap tables...[Thread 0x6a9fc460 (LWP 1285) exited]
[Thread 0x6b1fd460 (LWP 1283) exited]
[Thread 0x6b9fe460 (LWP 1282) exited]
[Thread 0x6c1ff460 (LWP 1281) exited]
[Thread 0x6cb56460 (LWP 1280) exited]

Program terminated with signal SIGKILL, Killed.
The program no longer exists.
(gdb)
```
`info stack` and `up` now return `No stack.` When running hashcat again normally, I get 
```sudo  hashcat -a3 -m0 hash.txt
hashcat (v5.1.0-1321-ge74396d5) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 448/448 MB allocatable, 1MCU

Generated bitmap tables...Killed
```
This resets when I reboot however.

ling1729, 2020-01-05:
Thanks! Hashcat now runs, but during runtime I get this error
```
[E] Sat Jan  4 21:38:21 2020:  (30) /usr/local/lib/libVC4CC.so.1.2 : +0x61ee58 [0x709cce58]
[E] Sat Jan  4 21:38:21 2020:  (31) /usr/local/lib/libVC4CC.so.1.2 : +0x61fc80 [0x709cdc80]
[E] Sat Jan  4 21:38:21 2020:  (32) /usr/local/lib/libVC4CC.so.1.2 : +0x61f94c [0x709cd94c]
[E] Sat Jan  4 21:38:21 2020:  (33) /usr/local/lib/libVC4CC.so.1.2 : +0x61ffd8 [0x709cdfd8]
[E] Sat Jan  4 21:38:21 2020:  (34) /usr/local/lib/libVC4CC.so.1.2 : +0x61ff38 [0x709cdf38]
[E] Sat Jan  4 21:38:21 2020:  (35) /usr/local/lib/libVC4CC.so.1.2 : +0x61ff10 [0x709cdf10]
[E] Sat Jan  4 21:38:21 2020:  (36) /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 : +0x9d9b0 [0x702da9b0]
[E] Sat Jan  4 21:41:45 2020: Compiler threw exception: Normalizer: Accessing memory through a phi-node is not implemented yet: (p) i32* %call49.%call49.%call.i.%call49.%call.i.%call26.%buf1.sink77

* Device #1: Kernel /usr/local/share/hashcat/OpenCL/m01000_a0-optimized.cl build failed - proceeding without this device.

```

ling1729, 2020-01-17:
I also needed to increase my swap file to 1 gb for hashcat to compile the opencl runtime binaries

mspinolo, 2020-01-20:
Hi there, is there any update on this? trying to run hashcat on my RPI but without success (using latest nightly VC4CL) and wondering whether is a user error or still lacking Hashcat support.

Kind regards,
Michele

doe300, 2020-01-20:
The above error (`Normalizer: Accessing memory through a phi-node is not implemented yet`) is not yet solved. I did some improvements, but an actual solution is way more complex than I would like. 

> I also needed to increase my swap file to 1 gb for hashcat to compile the opencl runtime binaries

@ling1729, what compilation step exactly requires that much memory?

ling1729, 2020-01-20:
When running hashcat, it gets to here 
```
hashcat (v5.1.0-1561-gd7de083a) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 12/76 MB (76 MB allocatable), 1MCU

Minimum password length supported by kernel: 0
Maximum password length supported by kernel: 27

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates

Applicable optimizers:
* Optimized-Kernel
* Zero-Byte
* Precompute-Init
* Meet-In-The-Middle
* Early-Skip
* Not-Salted
* Not-Iterated
* Single-Hash
* Single-Salt
* Brute-Force
* Raw-Hash

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

Initializing backend runtime for device #1...
```

After increasing my GPU memory and letting it run for an hour or so it gets to the above error.

doe300, 2020-02-01:
Can you check which threads are running for your hashcat program when the memory usage increased that dramatically? Also, which of the threads exactly take up all the CPU?

This can e.g. be done by installing `htop` and configuring it to [show thread names](https://stackoverflow.com/questions/4461289/change-thread-name-on-linux-htop). This way I can determine whether the problem is in compilation or execution.

ling1729, 2020-02-02:
It doesn't seem to use as much memory as before now, when running a wordlist it first uses around 200mb of ram running `11279 root       20   0  205M  133M 53832 R 100. 14.2  0:24.86 /usr/bin/clang -cc1 -triple spir-unknown-unknown -D KERNEL_STATIC -I OpenCL -I /usr/local/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D HAS_ADD=0 -D HAS_ADDC=0 -D HAS_SUB=0 -D HAS_SUBC=0 -D HAS_VADD=0 -D HAS_VADDC=0 -D HAS_VADD_CO=0 -D HAS_VADDC_CO=0 -D HAS_VSUB=0 -D HAS_VSUBB=0 -D HAS_VSUB_CO=0 -D HAS_VSUBB_CO=0 -D HAS_VPERM=0 -D HAS_VADD3=0 -D HAS_VBFE=0 -D HAS_BFE=0 -D HAS_LOP3=0 -D HAS_MOV64=0 -D HAS_PRMT=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=1000 -D ATTACK_EXEC=11 -D ATTACK_KERN=3 -D _unroll -w -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -fgnu89-inline -Wno-undefined-inline -Wno-unused-parameter -Wno-unused-local-typedef -Wno-gcc-compat -include-pch /home/pi/VC4C/../VC4CLStdLib/include//VC4CLStdLib.h.pch -x cl -S -emit-llvm-bc -o /tmp/vc4c-lPBFQs -`, using 1 core. After a bit, it starts running `hashcat -m 1000 -a 0 hash.txt rockyou.txt -r hob064.rule -O`, which uses around 200 mb more ram. Hashcat has been stuck on 
```
hashcat (v5.1.0-1615-gcf4cee2f) starting...

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999) - Platform #1 [doe300]
=============================================================
* Device #1: VideoCore IV GPU, 0/64 MB (64 MB allocatable), 1MCU

Minimum password length supported by kernel: 0
Maximum password length supported by kernel: 27

Hashes: 1 digests; 1 unique digests, 1 unique salts
Bitmaps: 16 bits, 65536 entries, 0x0000ffff mask, 262144 bytes, 5/13 rotates
Rules: 64

Applicable optimizers:
* Optimized-Kernel
* Zero-Byte
* Precompute-Init
* Meet-In-The-Middle
* Early-Skip
* Not-Salted
* Not-Iterated
* Single-Hash
* Single-Salt
* Raw-Hash

Watchdog: Hardware monitoring interface not found on your system.
Watchdog: Temperature abort trigger disabled.

Initializing backend runtime for device #1...
```
 this whole time. After around 10 min, it outputs 
https://pastebin.com/7dvhEgLY.

When running in brute force mode, it also runs clang for a bit, then the cpu gets stuck on 100% on all four cores from `hashcat -m 1000 -a 3 hash.txt -w 4 -O`.

doe300, 2020-02-06:
I looked into this the last few days and the same hashcat command running on my host machine reports up to 1GB of memory used!

Some analysis showed that hashcat itself seem to allocate >700MB on my machine of that at various positions, e.g.
- 270MB at https://github.com/hashcat/hashcat/blob/master/src/mpsp.c#L1396
- 140MB at https://github.com/hashcat/hashcat/blob/master/src/mpsp.c#L740
- 140MB at https://github.com/hashcat/hashcat/blob/master/src/mpsp.c#L676

I still will be trying to reduce the memory usage of different types and algorithms, but that might not be enough to be able to run that command on a Raspberry Pi at all, at least not without tuning the hashcat parameters and/or build options.

ling1729, 2020-02-06:
Any idea what causes the cpu to be stuck on 100% while not progressing in brute force mode?

doe300, 2020-02-06:
Some:
- some parts of the compilation (esp. for such big kernels like in the hashcat) use a lot of CPU time
- if the memory usage gets too high, the CPU utilization might be caused by swapping

telsch, 2020-09-18:
@ling1729 could you run without errors?
`hashcat -b`

I got following errors:
```
hashcat -b
hashcat (v6.1.1) starting in benchmark mode...

Benchmarking uses hand-optimized kernel code by default.
You can use it in your cracking session by setting the -O option.
Note: Using optimized kernel code limits the maximum supported password length.
To disable the optimized kernel code in benchmark mode, use the -w option.

OpenCL API (OpenCL 1.2 VC4CL 0.4.9999 (76edc68)) - Platform #1 [doe300]
=======================================================================
* Device #1: VideoCore IV GPU, 320/384 MB (384 MB allocatable), 1MCU

Benchmark relevant options:
===========================
* --optimized-kernel-enable

Hashmode: 0 - MD5

clBuildProgram(): CL_BUILD_PROGRAM_FAILURE

Compilation error:
        Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
[E] Fri Sep 18 15:49:58 2020: i32 %use_with_literal.605 = i32 %gid_max.upper
[E] Fri Sep 18 15:49:58 2020: i32 %use_with_literal.763 = i32 %gid_max.upper
[E] Fri Sep 18 15:49:58 2020: i32 %use_with_literal.754 = i32 %gid_max.upper
[E] Fri Sep 18 15:49:58 2020: Compiler threw exception: Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper

* Device #1: Kernel /usr/share/hashcat/OpenCL/shared.cl build failed.

Started: Fri Sep 18 15:46:51 2020
Stopped: Fri Sep 18 15:49:58 2020
```

ling1729, 2020-09-19:
I can confirm that I am getting the same error on the latest commit
```
OpenCL API (OpenCL 1.2 VC4CL 0.4.244 (76edc68)) - Platform #1 [doe300]
======================================================================
* Device #1: VideoCore IV GPU, 12/76 MB (76 MB allocatable), 1MCU

Benchmark relevant options:
===========================
* --optimized-kernel-enable

Hashmode: 0 - MD5

clBuildProgram(): CL_BUILD_PROGRAM_FAILURE

[W] Fri Sep 18 21:40:32 2020: Warnings in precompilation:
[W] Fri Sep 18 21:40:32 2020: remark: <unknown>:0:0: loop not vectorized: loop control flow is not understood by vectorizer
remark: <unknown>:0:0: loop not vectorized: loop control flow is not understood by vectorizer
remark: <unknown>:0:0: loop not vectorized: value that could not be identified as reduction is used outside the loop
remark: <unknown>:0:0: loop not vectorized: loop control flow is not understood by vectorizer

Compilation error:
        Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
```

PinkFreud, 2020-09-22:
I'm seeing a slightly different error with hashcat on a Raspberry Pi 2:

```
hashcat (v5.1.0) starting in benchmark mode...

Benchmarking uses hand-optimized kernel code by default.
You can use it in your cracking session by setting the -O option.
Note: Using optimized kernel code limits the maximum supported password length.
To disable the optimized kernel code in benchmark mode, use the -w option.

OpenCL Platform #1: doe300
==========================
* Device #1: VideoCore IV GPU, 256/256 MB allocatable, 1MCU

Benchmark relevant options:
===========================
* --optimized-kernel-enable

Hashmode: 0 - MD5

clBuildProgram(): CL_BUILD_PROGRAM_FAILURE

Compilation error:
	Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
[E] Mon Sep 21 20:30:11 2020: i32 %use_with_literal.13024 = i32 %gid_max.upper
[E] Mon Sep 21 20:30:11 2020: i32 %use_with_literal.13048 = i32 %gid_max.upper
[E] Mon Sep 21 20:30:11 2020: i32 %use_with_literal.12948 = i32 %gid_max.upper
[E] Mon Sep 21 20:30:11 2020:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x7214217c]
[E] Mon Sep 21 20:30:11 2020:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::qpu_asm::GraphColoring::createGraph()+0x530 [0x72321cd8]
[E] Mon Sep 21 20:30:11 2020:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::qpu_asm::GraphColoring::colorGraph()+0x194 [0x72322cdc]
[E] Mon Sep 21 20:30:11 2020:  (4) /usr/local/lib/libVC4CC.so.1.2 : vc4c::qpu_asm::CodeGenerator::generateInstructions(vc4c::Method&)+0x35c [0x72310870]
[E] Mon Sep 21 20:30:11 2020:  (5) /usr/local/lib/libVC4CC.so.1.2 : vc4c::qpu_asm::CodeGenerator::toMachineCode(vc4c::Method&)+0x30 [0x723122c4]
[E] Mon Sep 21 20:30:11 2020:  (6) /usr/local/lib/libVC4CC.so.1.2 : +0x75c03c [0x7214303c]
[E] Mon Sep 21 20:30:11 2020:  (7) /usr/local/lib/libVC4CC.so.1.2 : +0x75cf84 [0x72143f84]
[E] Mon Sep 21 20:30:11 2020:  (8) /usr/local/lib/libVC4CC.so.1.2 : std::function<void (vc4c::Method* const&)>::operator()(vc4c::Method* const&) const+0x54 [0x72148ea4]
[E] Mon Sep 21 20:30:11 2020:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::ThreadPool::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, logging::Logger*)::{lambda()#1}::operator()() const+0x2c [0x7214669c]
[E] Mon Sep 21 20:30:11 2020:  (10) /usr/local/lib/libVC4CC.so.1.2 : std::_Function_handler<void (), vc4c::ThreadPool::scheduleAll<vc4c::Method*, std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > >(std::vector<vc4c::Method*, std::allocator<vc4c::Method*> > const&, std::function<void (vc4c::Method* const&)> const&, logging::Logger*)::{lambda()#1}>::_M_invoke(std::_Any_data const&)+0x24 [0x7214ad70]
[E] Mon Sep 21 20:30:11 2020:  (11) /usr/local/lib/libVC4CC.so.1.2 : std::function<void ()>::operator()() const+0x40 [0x72191668]
[E] Mon Sep 21 20:30:11 2020:  (12) /usr/local/lib/libVC4CC.so.1.2 : +0x7d1758 [0x721b8758]
[E] Mon Sep 21 20:30:11 2020:  (13) /usr/local/lib/libVC4CC.so.1.2 : +0x7d3a38 [0x721baa38]
[E] Mon Sep 21 20:30:11 2020:  (14) /usr/local/lib/libVC4CC.so.1.2 : +0x7d36fc [0x721ba6fc]
[E] Mon Sep 21 20:30:11 2020:  (15) /usr/local/lib/libVC4CC.so.1.2 : +0x7d347c [0x721ba47c]
[E] Mon Sep 21 20:30:11 2020:  (16) /usr/local/lib/libVC4CC.so.1.2 : +0x7d3e74 [0x721bae74]
[E] Mon Sep 21 20:30:11 2020:  (17) /usr/local/lib/libVC4CC.so.1.2 : +0x7d3ad4 [0x721baad4]
[E] Mon Sep 21 20:30:11 2020:  (18) /usr/local/lib/libVC4CC.so.1.2 : std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const+0x48 [0x721bce48]
[E] Mon Sep 21 20:30:11 2020:  (19) /usr/local/lib/libVC4CC.so.1.2 : std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*)+0x28 [0x721bc374]
[E] Mon Sep 21 20:30:11 2020:  (20) /usr/local/lib/libVC4CC.so.1.2 : void std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)+0xb0 [0x721bee94]
[E] Mon Sep 21 20:30:11 2020:  (21) /usr/local/lib/libVC4CC.so.1.2 : std::__invoke_result<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>::type std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr< [... cut ...]
[E] Mon Sep 21 20:32:37 2020:  (36) /usr/local/lib/libVC4CC.so.1.2 : +0x7d36a8 [0x721ba6a8]
[E] Mon Sep 21 20:32:37 2020:  (37) /usr/local/lib/libVC4CC.so.1.2 : +0x7d344c [0x721ba44c]
[E] Mon Sep 21 20:32:37 2020:  (38) /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 : +0x9d9b0 [0x719139b0]
[E] Mon Sep 21 20:32:46 2020: Compiler threw exception: Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper

Started: Mon Sep 21 20:25:25 2020

Stopped: Mon Sep 21 20:32:46 2020

```

doe300, 2020-09-22:
Can anyone of you run the hashcat command with the `VC4CL_DEBUG=code` envrionment variable set and post the last OpenCL C source code before the compilation error (i.e. the contents of the last line with `Dumping program sources to <some-file>` before the compilation error) as well as the associated compilation flags (they should be in the line above, something like `Precompiling source with: <compilation flags>`)?

PinkFreud, 2020-09-22:
```
[VC4CL](        hashcat): Precompiling source with: -cl-std=CL1.2 -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D AMD_ROCM=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D _unroll
Dumping program sources to /tmp/vc4cl-source-1671866731.cl
[VC4CL](        hashcat): Precompilation complete with status: 0
[VC4CL](        hashcat): Compiling source with: -cl-std=CL1.2 -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D AMD_ROCM=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D _unroll
[VC4CL](        hashcat): Compilation error: Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
[VC4CL](        hashcat): Compilation complete with status: -11
Compilation log: Compilation error:
        Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper

```

[vc4cl-source-1671866731.cl.txt](https://github.com/doe300/VC4CL/files/5262419/vc4cl-source-1671866731.cl.txt)

telsch, 2020-09-22:
```
[VC4CL](        hashcat): Precompiling source with: -D KERNEL_STATIC -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D HAS_ADD=0 -D HAS_ADDC=0 -D HAS_SUB=0 -D HAS_SUBC=0 -D HAS_VADD=0 -D HAS_VADDC=0 -D HAS_VADD_CO=0 -D HAS_VADDC_CO=0 -D HAS_VSUB=0 -D HAS_VSUBB=0 -D HAS_VSUB_CO=0 -D HAS_VSUBB_CO=0 -D HAS_VPERM=0 -D HAS_VADD3=0 -D HAS_VBFE=0 -D HAS_BFE=0 -D HAS_LOP3=0 -D HAS_MOV64=0 -D HAS_PRMT=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D ATTACK_EXEC=11 -D ATTACK_KERN=3 -w 
Dumping program sources to /tmp/vc4cl-source-361610549.cl
[VC4CL](        hashcat): Precompilation complete with status: 0
[VC4CL](        hashcat): Compiling source with: -D KERNEL_STATIC -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D HAS_ADD=0 -D HAS_ADDC=0 -D HAS_SUB=0 -D HAS_SUBC=0 -D HAS_VADD=0 -D HAS_VADDC=0 -D HAS_VADD_CO=0 -D HAS_VADDC_CO=0 -D HAS_VSUB=0 -D HAS_VSUBB=0 -D HAS_VSUB_CO=0 -D HAS_VSUBB_CO=0 -D HAS_VPERM=0 -D HAS_VADD3=0 -D HAS_VBFE=0 -D HAS_BFE=0 -D HAS_LOP3=0 -D HAS_MOV64=0 -D HAS_PRMT=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D ATTACK_EXEC=11 -D ATTACK_KERN=3 -w 
[VC4CL](        hashcat): Compilation error: Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
[VC4CL](        hashcat): Compilation complete with status: -11
Compilation log: Compilation error:
        Label/Register Mapping: Local is being read, but first and last occurrence are the same: i32 %gid_max.upper
```
[vc4cl-source-361610549.cl.txt](https://github.com/doe300/VC4CL/files/5263004/vc4cl-source-361610549.cl.txt)

doe300, 2020-09-23:
So the problem is that hashcat read 64-bit kernel parameters, which are not yet supported (like most 64-bit integer related stuff).

PinkFreud, 2020-09-23:
Ouch.  I'm guessing that will take some time to implement, if you're able to do it at all?

doe300, 2020-09-23:
Support for 64-bit integer parameters by itself won't take that much time, I will try to have a look at that in the next days.

But in general, also some things work with 64-bit integers (e.g. some bitwise operations, addition, memory access), they are actually not really supported so most likely fixing that particular error will only cause another compilation error for some other 64-bit operation...

PinkFreud, 2020-09-23:
Well, let us know when 64 bit integers are working.  I'll be happy to do further testing - I have a cluster of 5 RPi2s here I'm playing with, so I'll be happy to provide any bug reports I can.

doe300, 2020-10-03:
So the particular problem with not being able to pass in 64-bit integer directly to kernels should be fixed now. For the fix, all 3 repositories need to be updated.

telsch, 2020-10-04:
In particular this error from https://github.com/doe300/VC4CL/issues/81#issuecomment-694884286 seems to be fixed. But now hashcat got killed by oom_reaper.

doe300, 2020-10-05:
Yes, that is what I meant. For the out of memory, I also already debugged the reason for that in https://github.com/doe300/VC4CL/issues/81#issuecomment-583018345. So without modifying the hashcat code, I doubt it will actually work on a Raspberry Pi.

PinkFreud, 2020-10-05:
> In particular this error from [#81 (comment)](https://github.com/doe300/VC4CL/issues/81#issuecomment-694884286) seems to be fixed. But now hashcat got killed by oom_reaper.

Give your Pi 1G of swap.  hashcat seems to need ~ 1.2-1.3G of memory just for the MD5 benchmark.

PinkFreud, 2020-10-05:
Got much further this time, but still died during compilation (vc4cl-binary-1350418454.bin is empty, so I didn't attach it).

```
[VC4CL](        hashcat): Precompiling source with: -cl-std=CL1.2 -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D AMD_ROCM=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D _unroll
Dumping program sources to /tmp/vc4cl-source-1165735540.cl
[VC4CL](        hashcat): Dumping program IR to /tmp/vc4cl-ir-558203856.ll
[VC4CL](        hashcat): Precompilation complete with status: 0
[VC4CL](        hashcat): Compiling source with: -cl-std=CL1.2 -I OpenCL -I /usr/share/hashcat/OpenCL -D LOCAL_MEM_TYPE=2 -D VENDOR_ID=2147483648 -D CUDA_ARCH=0 -D AMD_ROCM=0 -D VECT_SIZE=16 -D DEVICE_TYPE=4 -D DGST_R0=0 -D DGST_R1=3 -D DGST_R2=2 -D DGST_R3=1 -D DGST_ELEM=4 -D KERN_TYPE=0 -D _unroll
[VC4CL](        hashcat): Compilation error: Label/Register Mapping: There are erroneous register-associations!
[VC4CL](        hashcat): Compilation complete with status: -11
Compilation log: Compilation error:
        Label/Register Mapping: There are erroneous register-associations!
[W] Mon Oct  5 15:02:56 2020: Register conflict resolver has exceeded its maximum rounds, there might still be errors!
[E] Mon Oct  5 15:04:02 2020: Error assigning local to register: %m00000s.2047037415.%add42
[E] Mon Oct  5 15:04:02 2020: Error assigning local to register: %m00000s.2047037415.%add56
[E] Mon Oct  5 15:04:02 2020: Error assigning local to register: %m00000s.2047037415.%add70
[E] Mon Oct  5 15:04:02 2020: Error assigning local to register: %m00000s.2047037415.%add74
...
```

[vc4cl-ir-558203856.ll.zip](https://github.com/doe300/VC4CL/files/5329716/vc4cl-ir-558203856.ll.zip)
[vc4cl-source-1165735540.cl.txt](https://github.com/doe300/VC4CL/files/5329722/vc4cl-source-1165735540.cl.txt)
[hashcat-20201005-01.err.txt](https://github.com/doe300/VC4CL/files/5329726/hashcat-20201005-01.err.txt)

PinkFreud, 2020-12-09:
@doe300, any followup here?

doe300, 2020-12-10:
I did not do anything specific in that directory and just tested with my latest development build, still getting the same errors.

PinkFreud, 2020-12-10:
I threw some unsalted MD5 hashes at it, and verified that the same register errors show up - it's not just the benchmark (which makes sense, but I figured I'd verify it anyway).

NetherStar64, 2021-01-10:
I'm also having the issue that Hashcat is not working. When I run hashcat with ```sudo ./hashcat -a 3 -m 0 hash.txt -O```, It hangs at ```Initializing backend runtime for device #1...``` and uses a lot of memory. I increased my swapfile cause I wanted to see how much, in total 1500 MB.

When running an example, it at first runs clang and then crashes with
[Crash.txt](https://github.com/doe300/VC4CL/files/5792642/Crash.txt)

I installed VC4CL normally following the guide and compiled hashcat from source normally.