doe300/VC4CL #82: BabelStream tested, bandwidth OK, validation fails

pfoof, 2019-08-17:
I don't know how to contribute to GitHub wiki, so just reporting in results from https://github.com/UoB-HPC/BabelStream

Triad works with `--float`, no error is reported, keep array size small (max tested `Array: 2048000  Iters: 12       Bandwidth (GB/s):  0.037`)

For all tests validation fails, but the speeds are reported:
```
Validation failed on sum. Error 75.5013
Sum was 150.991104125977 but should be 226.492431640625
Function    MBytes/sec  Min (sec)   Max         Average     
Copy        14.707      0.03481     0.03481     0.03481     
Mul         26.906      0.01903     0.01903     0.01903     
Add         39.416      0.01948     0.01948     0.01948     
Triad       39.135      0.01962     0.01962     0.01962     
Dot         79.948      0.00640     0.00640     0.00640
```

doe300, 2019-08-20:
Thanks for the report. I will have a look at the error and add the results to the wiki when I have some time.

What as the exact command that you used to run the test?

pfoof, 2019-08-21:
`sudo ./ocl-stream --float -n 12 --arraysize 640000`