doe300/VC4CL #83: Validation-error while running OpenCL kernel on Raspberry Pi 3 B+

Jiaqingmcw, 2019-08-23:
Hi, when I try to use VC4CL to run a OpenCL kernel on Raspberry Pi 3 B+, I met an error :

[W] Fri Aug 23 13:05:30 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Fri Aug 23 13:05:30 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Fri Aug 23 13:05:30 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[W] Fri Aug 23 13:05:30 2019: 64-bit operations are not supported by the VideoCore IV architecture, further compilation may fail!
[E] Fri Aug 23 13:05:36 2019: Validation-error 'Cannot read register ra0 because it just has been written by the previous instruction.' in: or r0, ra0, r3
[E] Fri Aug 23 13:05:36 2019: With reference to instruction: or.ifz ra0, r0, r0
[E] Fri Aug 23 13:05:36 2019:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xf4 [0x741efd40]

The kernel is like : 
__kernel void test_kernel(
    IMAGE_DECLARATION(in),
    IMAGE_DECLARATION(out),
    __global float matrix[9])
{
    Image in  = CONVERT_TO_IMAGE_STRUCT_NO_STEP(in);
    Image out = CONVERT_TO_IMAGE_STRUCT(out);
	
	float8 mtx = (float8)(matrix[0], matrix[1], matrix[2], matrix[3], matrix[4], matrix[5], 0.0, 0.0);

    float4 in_x_coords = (float4)(0, 1, 2, 3);

    float4 new_x = mad(in_x_coords, in_x_coords, in_x_coords);

    uchar4 newxcv = convert_uchar4_rtn(new_x);

    vstore4(newxcv, 0, out.ptr);
}

Do we support vector operation in VC4CL ? Why is there an error about "read register ra0" ?
Look forward to your reply.

doe300, 2019-08-23:
> Do we support vector operation in VC4CL ?

Yes, they are supported, since all operations are actually computed for 16-element SIMD anyway.

> Why is there an error about "read register ra0" ?

Probably some compiler error. Basically, the verifier checks some hardware constraints, which the generated code violate. I will have a look at this when I have some time.

doe300, 2019-08-24:
How are these macros/types defined?

- IMAGE_DECLARATION
- IMAGE_DECLARATION
- CONVERT_TO_IMAGE_STRUCT_NO_STEP
- CONVERT_TO_IMAGE_STRUCT
- Image

Jiaqingmcw, 2019-08-25:
Here is the definition of the macros/types:

#define IMAGE_DECLARATION(name)      \
    __global uchar *name##_ptr,      \
    uint        name##_stride_x, \
    uint        name##_step_x,   \
    uint        name##_stride_y, \
    uint        name##_step_y,   \
    uint        name##_offset_first_element_in_bytes
	
#define CONVERT_TO_IMAGE_STRUCT(name) \
    update_image_workitem_ptr(name##_ptr, name##_offset_first_element_in_bytes, name##_stride_x, name##_step_x, name##_stride_y, name##_step_y)

#define CONVERT_TO_IMAGE_STRUCT_NO_STEP(name) \
    update_image_workitem_ptr(name##_ptr, name##_offset_first_element_in_bytes, name##_stride_x, 0, name##_stride_y, 0)
	
/** Structure to hold Image information */
typedef struct Image
{
    __global uchar *ptr;                           /**< Pointer to the starting postion of the buffer */
    int             offset_first_element_in_bytes; /**< The offset of the first element in the source image */
    int             stride_x;                      /**< Stride of the image in X dimension (in bytes) */
    int             stride_y;                      /**< Stride of the image in Y dimension (in bytes) */
} Image;

doe300, 2019-08-26:
Now it complains about `update_image_workitem_ptr` being missing.
Can you please send me a fully compilable kernel source?

doe300, 2020-05-09:
Without having the full kernel code, there is not really much I can do here. Feel free to reopen when you send me a fully compilable kernel.

closed 2020-05-09 by doe300