doe300/VC4CL #84: 'std::runtime_error'   what():  -15:  Aborted When running on GPU

MaierF87, 2019-08-26:
Hey, though I had managed to get VC4CL setup correctly and tried confirm it using your Game of Life example. The example seems to work just fine when using the CPU, but thats not the case for the GPU.

Here are the outputs:
```
./GameOfLife GPU 50 50
Before:

0:	                                                  
1:	 xx                                               
2:	 xx                                               
3:	                                                  
4:	                                                  
5:	                    xx                            
6:	                    xx                            
7:	 x                    xx                          
8:	 x                    xx                          
9:	 x                                                
10:	                                                  
11:	             xx                                   
12:	            x  x                                  
13:	             xx                                   
14:	                                                  
15:	                                                  
16:	                                                  
17:	                                                  
18:	                                                  
19:	                                                  
20:	                                                  
21:	                                                  
22:	                                                  
23:	                                                  
24:	                                                  
25:	                                                  
26:	                                                  
27:	                             x                    
28:	                            x x                   
29:	                             x                    
30:	                                                  
31:	                                                  
32:	                                                  
33:	                                                  
34:	                                                  
35:	                                                  
36:	                                                  
37:	                                                  
38:	                                                  
39:	                                                  
40:	                                                  
41:	                                                  
42:	                                                  
43:	                                                  
44:	                                                  
45:	                                                  
46:	                                                  
47:	                                                  
48:	                                                  
49:	                                                  

Running on GPU with a 50 * 50 grid and 1024 steps
terminate called after throwing an instance of 'std::runtime_error'
  what():  -15: 
Aborted

pi@raspberrypi:~/GameOfLife/build $ ./GameOfLife CPU 50 50
Before:

0:	                                                  
1:	 xx                                               
2:	 xx                                               
3:	                                                  
[... 132 lines cut ...]
26:	                                                  
27:	                             x                    
28:	                            x x                   
29:	                             x                    
30:	                                                  
31:	                                                  
32:	                                                  
33:	                                                  
34:	                                                  
35:	                                                  
36:	                                                  
37:	                                                  
38:	                                                  
39:	                                                  
40:	                                                  
41:	                                                  
42:	                                                  
43:	                                                  
44:	                                                  
45:	                                                  
46:	                                                  
47:	                                                  
48:	                                                  
49:	
```

doe300, 2019-09-04:
Can you run the program with gdb, break on exceptions via `catch throw` and print the stack trace (`info stack`)?

MaierF87, 2019-09-13:
pi@raspberrypi:~/GameOfLife $ gdb GameOfLife 
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from GameOfLife...done.
(gdb) catch throw
Catchpoint 1 (throw)
(gdb) info stack
No stack.

I've tried debugging it with some printf lines in the GPU file, and as the error suggested its a problem building the Program (line 53).

I used the source code found at https://github.com/smistad/OpenCL-Getting-Started/ as a reference and got my OpenCL-Project to work, with very poor Kernel performance but it works.

closed 2021-02-09 by doe300