doe300/VC4CL #86: OpenCL support for Raspberry pi 4 GPU (VideoCore VI)

sandyz1000, 2019-11-17:
First of all thank you for building OpenCL compiler for VideoCoreIV GPUs. Just wanted to know is there is possibility to support for VC6 GPU devices.

doe300, 2019-11-18:
Short answer: no, see also #79.
The VideoCore VI GPU is a very different architecture (e.g. with MMU, without 2 separate register files AFAIK), so supporting it require rewriting most of the compiler.
Also, I don't know of any publicly available hardware and instruction set documentation.

antonakv, 2020-02-12:
@doe300 As I see Eben Upton co-founder of the Raspberry Pi Foundation, has announced open source Vulkan graphics driver for the Raspberry Pi 4 family. So I guess He can share VideoCore VI specs with you as well.

MartijnGevaert, 2022-01-01:
I tried to get this working:
https://github.com/kpet/clvk
But first off it needs a whole lot of ram to build (4gb ram + 6 gb swap for me). And unfortunately failed to build. Installing Vulkan on RPi isn't straightforward either.
I'm not in urgent need to run OpenCL on RPi, but wanted to post this to point out that this ticket probably can't be closed.
(Unless I missed something and recent developments have led to a better working approach?)

stolk, 2023-06-02:
I can [confirm](https://github.com/kpet/clvk/pull/543/files) that [CLVK](https://github.com/kpet/clvk) in combination with the latest version of Mesa's [v3dv](https://docs.mesa3d.org/drivers/v3d.html) Vulkan driver, you can get OpenCL on a rPi4.

Unfortunately, important extensions are missing. The v3dv Vulkan driver can't do FP16 (16 bit floats.) Nor can it do 8 bits integers. This limits the usefulness for me personally, as my kernels are written to operate on FP16 values.

If your kernels don't use FP16 / INT8, you may have better luck than me, though.

nullr0ute, 2023-06-08:
I suspect the best way to get a decent supported OpenCL will be using the mesa gallium driver in conjunction with the new mesa rusticl opencl stack. rusticl is being actively developed and a there has been support added for a bunch of other arm GPUs that support opencl (panfrost for MALI, etnaviv for Vivante).

stolk, 2023-06-09:
> the new mesa rusticl opencl stack.

It can find the rusticl platform.
```
$ clinfo --list
Platform #0: Clover
Platform #1: rusticl
```

But it cannot find any devices:
```
$ clinfo 
Number of platforms                               2
  Platform Name                                   Clover
  Platform Vendor                                 Mesa
  Platform Version                                OpenCL 1.1 Mesa 23.0.2
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions function suffix             MESA

  Platform Name                                   rusticl
  Platform Vendor                                 Mesa/X.org
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_icd
  Platform Extensions with Version                cl_khr_icd                                                       0x400000 (1.0.0)
  Platform Numeric Version                        0xc00000 (3.0.0)
  Platform Extensions function suffix             MESA
  Platform Host timer resolution                  0ns

  Platform Name                                   Clover
Number of devices                                 0

  Platform Name                                   rusticl
Number of devices                                 0
...
```
This is with Ubuntu 23.04 on rPi-400.

nullr0ute, 2023-06-10:
> > the new mesa rusticl opencl stack.
> 
> It can find the rusticl platform.
> 
> ```
> $ clinfo --list
> Platform #0: Clover
> Platform #1: rusticl
> ```
> 
> But it cannot find any devices:

There's probably some driver work required, but ultimately that will be the best/quickest way to get GPU HW OpenCL on the rpi4

Pepslee, 2023-06-16:
> I can [confirm](https://github.com/kpet/clvk/pull/543/files) that [CLVK](https://github.com/kpet/clvk) in combination with the latest version of Mesa's [...]

Can I make a build in docker on the amd64 platform with arm64 emulation ?  It might to decrease building time