doe300/VC4CL #88: Qrack support: aborted on 'mpark::bad_variant_access'

WrathfulSpatula, 2019-12-08:
Hey, maybe you remember us over at vm6502q/qrack. (We love your project!) We ultimately got the OpenCL parts of Qrack running on VC4CL, with your help, but something has since regressed. I updated my copies of your repos and tried 2 or 3 times to complete a fresh reinstallation from your cross-compiled Debian packages. VC4CL and its dependencies seem to compile and install just fine, but I keep running into `mpark::bad_variant_access` on Qrack OpenCL program compilation.

The OpenCL program for Qrack is very large, though I think I can say there's been little or no change in the program since the last time we successfully compiled with VC4C. I recall that was about a couple of months ago, though the version of VC4CL and its dependencies might have been much older than a couple of months.

Thanks for the help. I ran gdb with `catch throw` and printed the stack trace, and it doesn't seem to implicate anything specific to Qrack:

```
$ sudo gdb qrack_cl_precompile 
GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from qrack_cl_precompile...(no debugging symbols found)...done.
(gdb) catch throw
Catchpoint 1 (throw)
(gdb) run
Starting program: /home/pi/Github/qrack/_build/qrack_cl_precompile 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Precompiling OCL kernels...
Will save to: /root/.qrack/
[New Thread 0x73767450 (LWP 4168)]
Device #0, Built JIT.
[New Thread 0x72980450 (LWP 4173)]
[New Thread 0x71fff450 (LWP 4174)]
[New Thread 0x717fe450 (LWP 4175)]
[New Thread 0x70ffd450 (LWP 4176)]
[New Thread 0x704ba450 (LWP 4177)]
[New Thread 0x6fcb9450 (LWP 4178)]
[New Thread 0x6f4b8450 (LWP 4179)]
[New Thread 0x6ecb7450 (LWP 4180)]
[Switching to Thread 0x6f4b8450 (LWP 4179)]

Thread 9 "Normalization" hit Catchpoint 1 (exception thrown), 0x76e890c8 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
(gdb) info stack
#0  0x76e890c8 in __cxa_throw () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#1  0x7659c6e4 in mpark::throw_bad_variant_access () at /home/pi/Github/VC4C/_build/_deps/variant-src/include/mpark/variant.hpp:249
#2  0x765efbd8 in mpark::detail::generic_get<0u, mpark::variant<vc4c::Literal, vc4c::Register, vc4c::Local*, vc4c::SmallImmediate, vc4c::SIMDVector, mpark::monostate> const&> (v=...)
    at /home/pi/Github/VC4C/_build/_deps/variant-src/include/mpark/variant.hpp:1765
#3  0x765efc2c in mpark::get<0u, vc4c::Literal, vc4c::Register, vc4c::Local*, vc4c::SmallImmediate, vc4c::SIMDVector, mpark::monostate> (v=...) at /home/pi/Github/VC4C/_build/_deps/variant-src/include/mpark/variant.hpp:1785
#4  0x765efc54 in mpark::get<vc4c::Literal, vc4c::Literal, vc4c::Register, vc4c::Local*, vc4c::SmallImmediate, vc4c::SIMDVector, mpark::monostate> (v=...)
    at /home/pi/Github/VC4C/_build/_deps/variant-src/include/mpark/variant.hpp:1806
#5  0x765efc80 in vc4c::Value::literal (this=0x108c4d8) at /home/pi/Github/VC4C/src/Values.h:1023
#6  0x766d4e50 in vc4c::analysis::ValueRange::update (this=0x6f4b0fa8, constant=..., ranges=std::unordered_map with 11 elements = {...}, it=0x108c300, method=0xc14548) at /home/pi/Github/VC4C/src/analysis/ValueRange.cpp:296
#7  0x766d6250 in vc4c::analysis::ValueRange::<lambda(const LocalUser*)>::operator()(const vc4c::LocalUser *) const (__closure=0x6de9a738, writer=0x108c300) at /home/pi/Github/VC4C/src/analysis/ValueRange.cpp:445
#8  0x766da434 in std::_Function_handler<void(const vc4c::intermediate::IntermediateInstruction*), vc4c::analysis::ValueRange::updateRecursively(const vc4c::Local*, const vc4c::Method*, vc4c::FastMap<const vc4c::Local*, vc4c::analysis::ValueRange>&, vc4c::FastMap<const vc4c::intermediate::IntermediateInstruction*, vc4c::analysis::ValueRange>&, vc4c::FastMap<const vc4c::intermediate::IntermediateInstruction*, vc4c::Optional<vc4c::analysis::ValueRange> >&)::<lambda(const LocalUser*)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73d843>) (__functor=..., 
    __args#0=<unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73d843>) at /usr/include/c++/6/functional:1731
#9  0x765bbe88 in std::function<void (vc4c::intermediate::IntermediateInstruction const*)>::operator()(vc4c::intermediate::IntermediateInstruction const*) const (this=0x6f4b12b0, __args#0=0x108c300)
    at /usr/include/c++/6/functional:2127
#10 0x765b9af8 in vc4c::Local::forUsers(vc4c::LocalUse::Type, std::function<void (vc4c::intermediate::IntermediateInstruction const*)> const&) const (this=0xc1b384, type=vc4c::LocalUse::Type::WRITER, consumer=...)
    at /home/pi/Github/VC4C/src/Locals.cpp:59
#11 0x766d6900 in vc4c::analysis::ValueRange::updateRecursively (currentLocal=0xc1b384, method=0xc14548, ranges=std::unordered_map with 11 elements = {...}, closedSet=std::unordered_map with 11 elements = {...}, 
    openSet=std::unordered_map with 14 elements = {...}) at /home/pi/Github/VC4C/src/analysis/ValueRange.cpp:456
#12 0x766d5f88 in vc4c::analysis::ValueRange::<lambda(const LocalUser*)>::<lambda(const vc4c::Local*, vc4c::LocalUse::Type, const vc4c::intermediate::IntermediateInstruction&)>::operator()(const vc4c::Local *, vc4c::LocalUse::Type, const vc4c::intermediate::IntermediateInstruction &) const (__closure=0x6de9a720, loc=0xc1b384, type=vc4c::LocalUse::Type::READER, inst=...) at /home/pi/Github/VC4C/src/analysis/ValueRange.cpp:438
#13 0x766da2e0 in std::_Function_handler<void(const vc4c::Local*, vc4c::LocalUse::Type, const vc4c::intermediate::IntermediateInstruction&), vc4c::analysis::ValueRange::updateRecursively(const vc4c::Local*, const vc4c::Method*, vc4c::FastMap<const vc4c::Local*, vc4c::analysis::ValueRange>&, vc4c::FastMap<const vc4c::intermediate::IntermediateInstruction*, vc4c::analysis::ValueRange>&, vc4c::FastMap<const vc4c::intermediate::IntermediateInstruction*, vc4c::Optional<vc4c::analysis::ValueRange> >&)::<lambda(const LocalUser*)>::<lambda(const vc4c::Local*, vc4c::LocalUse::Type, const vc4c::intermediate::IntermediateInstruction&)> >::_M_invoke(const std::_Any_data &, <unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73f044>, <unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73f050>, const vc4c::intermediate::IntermediateInstruction &) (__functor=..., 
    __args#0=<unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73f044>, __args#1=<unknown type in /usr/local/lib/libVC4CC.so.1.2, CU 0x6ff820, DIE 0x73f050>, __args#2=...)
    at /usr/include/c++/6/functional:1731
#14 0x76750a54 in std::function<void (vc4c::Local const*, vc4c::LocalUse::Type, vc4c::intermediate::IntermediateInstruction const&)>::operator()(vc4c::Local const*, vc4c::LocalUse::Type, vc4c::intermediate::IntermediateInstruction const&) const (this=0x6f4b1920, __args#0=0xc1b384, __args#1=vc4c::LocalUse::Type::READER, __args#2=...) at /usr/include/c++/6/functional:2127
#15 0x7674e820 in vc4c::intermediate::IntermediateInstruction::forUsedLocals(std::function<void (vc4c::Local const*, vc4c::LocalUse::Type, vc4c::intermediate::IntermediateInstruction const&)> const&) const (this=0x108c568, 
    consumer=...) at /home/pi/Github/VC4C/src/intermediate/Instruction.cpp:448
[... 167 lines cut ...]
#116 0x765da5e0 in std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, [... cut ...]
#130 0x765d747c in std::thread::_State_impl<std::_Bind_simple<vc4c::ThreadPool::ThreadPool(const string&, unsigned int)::<lambda()>()> >::_M_run(void) (this=0x124de38) at /usr/include/c++/6/thread:197
#131 0x76eb39dc in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#132 0x76f7cfc4 in start_thread (arg=0x6f4b8450) at pthread_create.c:458
#133 0x76cfd038 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:76 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```

doe300, 2019-12-09:
Yeah, this is a bug in VC4C, nothing to do with your OpenCL client.

If you need the code working before I push the patch, you can replace https://github.com/doe300/VC4C/blob/master/src/analysis/ValueRange.cpp#L296 from

    extendBoundaries(0, static_cast<int64_t>(litArg->literal().unsignedInt()));

to

    extendBoundaries(0, static_cast<int64_t>(litArg->getLiteralValue()->unsignedInt()));

WrathfulSpatula, 2019-12-09:
Thank you! I will test with the suggested change.

closed 2019-12-15 by doe300