doe300/VC4CL #89: How to intergrate VC4CL in yocto build ?

bhargavthriler, 2020-01-27:
I wanted to prepare a raspberry pi 3 image (either 32 or 64 bit) with VC4CL integrated in the rootfs.
Any suggestion how to do it? 
I'm using yocto build system to generate the images.

doe300, 2020-01-27:
I don't know the first thing about yocto builds, but I assume you want to install it the same way that you install any other software that has to be build from sources.

bhargavthriler, 2020-01-28:
> I don't know the first thing about yocto builds, but I assume you want to install it the same way that you install any other software that has to be build from sources.

Ok that can be done.
So can you explain me like what is the flow of installation like: 
1. Cross compile and build in host
2. Copy the built binaries/ files to the target 
Or am I missing something ?

Actually I'm not clear how the installation process of the VC4CL is going on. If you could explain a bit more?

doe300, 2020-01-28:
If you cross-compile, then you should probably generate the debian packages and install them on the target via the package manager.

The installation process depends on the way you want to install it.

Using packages:
1. compile both VC4C and VC4CL and generate packages
2. install the `vc4cl-stdlib`, `vc4c` and `vc4cl` packages via apt in that order on the target
3. this will install into /usr/xxx

Manual install:
1. compile and install (`sudo make install`) both VC4C and VC4CL projects
2. this will install into /usr/local/xxx unless you specify a different install path when configuring the cmake projects via `CMAKE_INSTALL_PREFIX`.

bhargavthriler, 2020-02-24:
@doe300 

While cross compiling following the intruction from "https://github.com/doe300/VC4CL/wiki/How-to-get#built-it-yourself" I am getting the following erros. Can you please explain what I'm doing wrong here? 
First I'm exported the cross compilers

```
bhargavthriler@instance-3:~/VC4/arm64build/VC4C (master)$ echo $PATH
/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/sbin:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/bin:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/sbin:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/../x86_64-pokysdk-linux/bin:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux:/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux-musl:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
bhargavthriler@instance-3:~/VC4/arm64build/VC4C (master)$ 

```
## Building VC4C
**Generating makefiles**
While generating makefiles I'm getting LLVM package related errors
```
bhargavthriler@instance-3:~/VC4/arm64build/VC4C/build (master)$ cmake ..
CMake Deprecation Warning at CMakeLists.txt:4 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- VC4CL standard library headers found: /home/bhargavthriler/VC4/arm64build/VC4C/../VC4CLStdLib/include/
-- Toolchain file defaulted to '/opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/share/cmake/OEToolchainConfig.cmake'
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc
-- Check for working C compiler: /opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++
-- Check for working CXX compiler: /opt/poky/2.7+snapshot/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Enabling multi-threaded optimizations
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using CMake 3.14+ FetchContent to include dependencies...
-- CLang compiler found: /usr/bin/clang-9
-- Using enhanced OpenCL features of Clang 3.9+!
-- Clang opt found: /usr/bin/opt-9
-- LLVM-link found: /usr/bin/llvm-link-9
CMake Warning at cmake/libllvm.cmake:37 (find_package):
  By not providing "FindLLVM.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "LLVM", but
  CMake did not find one.

  Could not find a package configuration file provided by "LLVM" with any of
  the following names:

    LLVMConfig.cmake
    llvm-config.cmake

  Add the installation prefix of "LLVM" to CMAKE_PREFIX_PATH or set
  "LLVM_DIR" to a directory containing one of the above files.  If "LLVM"
  provides a separate development package or SDK, be sure it has been
[... 163 lines cut ...]
/home/bhargavthriler/VC4/arm64build/VC4C/test/emulation_helper.h:430:12: error: 'output' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  430 |     return output;
      |            ^~~~~~
In file included from /home/bhargavthriler/VC4/arm64build/VC4C/build/cpptest-lite/src/cpptest-lite-project/include/cpptest.h:11,
                 from /home/bhargavthriler/VC4/arm64build/VC4C/test/TestInstructions.h:10,
                 from /home/bhargavthriler/VC4/arm64build/VC4C/test/TestInstructions.cpp:7:
/home/bhargavthriler/VC4/arm64build/VC4C/test/TestInstructions.cpp: In member function 'void TestInstructions::testALUInstructions()':
/home/bhargavthriler/VC4/arm64build/VC4C/test/TestInstructions.cpp:1587:77: warning: potential null pointer dereference [-Wnull-dereference]
 1587 |     TEST_ASSERT(ins.toBinaryCode() == ins.as<ALUInstruction>()->toBinaryCode())
      |                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/bhargavthriler/VC4/arm64build/VC4C/build/cpptest-lite/src/cpptest-lite-project/include/asserts.h:29:22: note: in definition of macro 'TEST_ASSERT'
   29 |         if(false == (condition)) { \
      |                      ^~~~~~~~~
cc1plus: some warnings being treated as errors
test/CMakeFiles/TestVC4C.dir/build.make:218: recipe for target 'test/CMakeFiles/TestVC4C.dir/TestIntrinsics.cpp.o' failed
make[2]: *** [test/CMakeFiles/TestVC4C.dir/TestIntrinsics.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:380: recipe for target 'test/CMakeFiles/TestVC4C.dir/all' failed
make[1]: *** [test/CMakeFiles/TestVC4C.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
bhargavthriler@instance-3:~/VC4/arm64build/VC4C/build (master)$ 
```

Can you help me with this? Am I missing some building options while generating makefiles?

doe300, 2020-02-24:
From the CMake error message it looks like you do not have any supported compilers installed. Did you check requirements in the [Readme](https://github.com/doe300/VC4C#required-software)? If you installed the compiler into a non-standard location, you might need to set one of the CMake flags.

closed 2021-02-09 by doe300