doe300/VC4CL #90: Do you have any plan for VC6 ?

elensar92, 2020-05-25:
I want to study how to make opencl driver 

If you have any plan for VC6 ? 

Make Open source project or something else?

doe300, 2020-05-25:
No I don't. There is no hardware specification that I am aware of, describing what is different to the VC4, and there is too much difference to simply reuse the same code.
Also, since the Raspberry Pi Foundation announced [Vulkan support](https://www.raspberrypi.org/blog/vulkan-raspberry-pi-first-triangle/), they will already be doing most of the heavy lifting there and maybe OpenCL can then be run on a VC6 via [clvk](https://github.com/kpet/clvk) based on that Vulkan implementation.

pfoof, 2020-05-25:
There’s also support for compute shaders already: https://blogs.igalia.com/itoral/2020/01/17/raspberry-pi-4-v3d-driver-gets-opengl-es-3-1-conformance/
https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/2368

doe300, 2020-08-30:
Duplicate of #86

closed 2020-08-30 by doe300