doe300/VC4CL #92: Compilation fails on RPi2B + Arch + distcc

bronze51, 2020-06-20:
The first error I got could be fixed with this code:
https://github.com/doe300/VC4C/issues/145#issuecomment-647030474

But the second error:

```
[ 95%] Building CXX object test/CMakeFiles/TestVC4C.dir/TestMathFunctions.cpp.o
[E] Sat Jun 20 14:28:51 2020: Errors in precompilation:
[E] Sat Jun 20 14:28:51 2020: error: unknown argument: '-target'
distcc[26575] ERROR: compile (null) on localhost failed
distcc[26574] ERROR: compile (null) on localhost failed

[E] Sat Jun 20 14:28:51 2020:  (1) ~/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x2ac [0x76c81664]
[E] Sat Jun 20 14:28:51 2020:  (2) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x300 [0x76e42e48]
[E] Sat Jun 20 14:28:51 2020:  (3) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : vc4c::Precompiler::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x468 [0x76e4b75c]
[E] Sat Jun 20 14:28:51 2020:  (4) ~/dev/vc4cl-git/src/VC4C/build/src/vc4c : main+0x944 [0x4a9b4c]
[E] Sat Jun 20 14:28:51 2020:  (5) /usr/lib/libc.so.6 : __libc_start_main+0x110 [0x768c1bb8]
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: error: unknown argument: '-target'
distcc[26575] ERROR: compile (null) on localhost failed
distcc[26574] ERROR: compile (null) on localhost failed

[E] Sat Jun 20 14:28:51 2020: Received signal: SIGABRT
[E] Sat Jun 20 14:28:51 2020:  (1) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : +0xe33e8 [0x76cab3e8]
[E] Sat Jun 20 14:28:51 2020:  (2) /usr/lib/libc.so.6 : __default_rt_sa_restorer+0 [0x768d9300]
[E] Sat Jun 20 14:28:51 2020:  (3) /usr/lib/libc.so.6 : gsignal+0xb8 [0x768d7fcc]
make[2]: *** [src/CMakeFiles/VC4C.dir/build.make:108: src/vc4c] Error 6
make[2]: *** Deleting file 'src/vc4c'
make[1]: *** [CMakeFiles/Makefile2:2040: src/CMakeFiles/VC4C.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

```

doe300, 2020-06-20:
What clang version do you have? What does `<VC4C-build-directory>/src/vc4c --version` output?

Judging by the error it looks like the command for you would be:

    ~/dev/vc4cl-git/src/VC4C/build/src/vc4c --version

bronze51, 2020-06-20:
```
clang ~/dev/vc4cl-git/src/VC4C/build/src/vc4c --version
clang version 10.0.0
Target: armv7l-unknown-linux-gnueabihf
Thread model: posix
InstalledDir: /usr/bin
```

doe300, 2020-06-20:
And just running `~/dev/vc4cl-git/src/VC4C/build/src/vc4c --version` ?

bronze51, 2020-06-20:
Sorry, I didn't read that correctly:
There is no `vc4c` in `~/dev/vc4cl-git/src/VC4C/build/src/vc4c`

However, it was installed here:

```
whereis vc4c
vc4c: /usr/local/bin/vc4c
vc4c --version
Running VC4C in version: 0.4.9999
Build configuration:
multi-threaded optimization; clang 3.9+ OpenCL features; clang in /usr/lib/distcc/bin/clang; LLVM library front-end with libLLVM 10; LLVM linker; vc4asm verification

```

doe300, 2020-06-20:
Can distcc even compile to SPIR?

doe300, 2020-06-21:
To give a little bit more context:

VC4C(L) should be able to be compiled with any C++14 able compiler, including distcc.

But, additionally to that the VC4C compiler requires a clang (at runtime) to compile the OpenCL C sources to either a LLVM IR module or SPIR-V binary code (depending on the build configuration of VC4C). It looks like in your build, you also configured to use distcc here (see output `clang in /usr/lib/distcc/bin/clang` above, either specified manually or because it was detected as valid clang version). I don't know how/if distcc provides the complete functionality that clang provides, but I would guess therein lies the problem.

bronze51, 2020-06-21:
I successfully installed using pacman: VC4C and VC4CLstlib. With VC4CL I have a pre-compilation error?

I appreciate the above follow-up and the explanation. While right now the RPi is setup to invoke `disctcc` for  everything that needs compiling, this can be changed. As far as I understand, if distcc remote compilation fails, the system tries again locally (on the RPi).

The errors above: 

```
[E] Sat Jun 20 14:28:51 2020:  (1) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x2ac [0x76c81664]
[E] Sat Jun 20 14:28:51 2020:  (2) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x300 [0x76e42e48]
[E] Sat Jun 20 14:28:51 2020:  (3) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : vc4c::Precompiler::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x468 [0x76e4b75c]
[E] Sat Jun 20 14:28:51 2020:  (4) ~/dev/vc4cl-git/src/VC4C/build/src/vc4c : main+0x944 [0x4a9b4c]
[E] Sat Jun 20 14:28:51 2020:  (5) /usr/lib/libc.so.6 : __libc_start_main+0x110 [0x768c1bb8]
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: error: unknown argument: '-target'
```

and

```
[E] Sat Jun 20 14:28:51 2020: Received signal: SIGABRT
[E] Sat Jun 20 14:28:51 2020:  (1) ~/dev/vc4cl-git/src/VC4C/build/src/libVC4CC.so.1.2 : +0xe33e8 [0x76cab3e8]
[E] Sat Jun 20 14:28:51 2020:  (2) /usr/lib/libc.so.6 : __default_rt_sa_restorer+0 [0x768d9300]
[E] Sat Jun 20 14:28:51 2020:  (3) /usr/lib/libc.so.6 : gsignal+0xb8 [0x768d7fcc]
```

happen at VC4CL "compilation time" and not at "runtime"?

doe300, 2020-06-21:
> happen at VC4CL "compilation time" and not at "runtime"?

Yes, because at the end of the VC4C compilation, the `vc4c` executable itself is invoked to precompile the VC4CLStdLib standard library implementation.

The error looks like distcc does not know the `-target` compilation flag.

Can you run the following command and paste the output?

    <path/to/VC4C/executable>/vc4c --precompile-stdlib -o /tmp/ <path/to/VC4CLStdLib>/include/VC4CLStdLib.h

This is the command that will be invoked at the end of compilation. Maybe a more verbose log output gives is some useful information...

bronze51, 2020-06-21:
I got:

```
/usr/local/bin/vc4c --precompile-stdlib -o /tmp/ /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
[E] Sun Jun 21 15:04:19 2020: Errors in precompilation:
[E] Sun Jun 21 15:04:19 2020: error: unknown argument: '-target'
distcc[32110] ERROR: compile (null) on localhost failed
distcc[32109] ERROR: compile (null) on localhost failed

terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: error: unknown argument: '-target'
distcc[32110] ERROR: compile (null) on localhost failed
distcc[32109] ERROR: compile (null) on localhost failed

[E] Sun Jun 21 15:04:19 2020: Received signal: SIGABRT
```
I can try to disable `dictcc`.

bronze51, 2020-06-21:
Alternatively, it is possible to bypass any front-end of `clang` from within `vc4c`?

doe300, 2020-06-22:
Can you run `/usr/local/bin/vc4c --verbose --precompile-stdlib -o /tmp/ /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h`?

> Alternatively, it is possible to bypass any front-end of clang from within vc4c?

I don't understand what you mean. The clang front-end is hard-coded into the VC4C executable on compilation. It is used as I said before for OpenCL C -> LLVM IR compilation. If you at runtime never pass OpenCL C code to the VC4CL implementation, the front-end is never used.
If you can live with this limitation, you can also circumvent the above error by creating empty files for:

- `/usr/local/include/vc4cl-stdlib/VC4CLStdLib.h.pch`
- `/usr/local/include/vc4cl-stdlib/VC4CLStdLib.bc`

bronze51, 2020-06-22:
> I don't understand what you mean. The clang front-end is hard-coded into the VC4C executable on compilation. It is used as I said before for OpenCL C -> LLVM IR compilation.

Sorry, I was unclear. For me distcc is a front-end of clang. Can't you bypass any front-end of clang and invoke clang directly from vc4c (i.e. hard code a path? Or have an optoin to identify clang's path)?

And:

```
/usr/local/bin/vc4c --verbose --precompile-stdlib -o /tmp/ /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
[D] Mon Jun 22 16:53:58 2020: Pre-compiling '/usr/local/include/vc4cl-stdlib/VC4CLStdLib.h' into '/tmp/'...
[I] Mon Jun 22 16:53:58 2020: Pre-compiling standard library with: /usr/lib/distcc/bin/clang -cc1 -triple spir-unknown-unknown -O3 -ffp-contract=off -cl-std=CL1.2 -cl-kernel-arg-info -cl-single-precision-constant -fgnu89-inline -Wno-all -Wno-gcc-compat -Wdouble-promotion -Wno-undefined-inline -Wno-unknown-attributes -x cl -emit-pch -o /tmp//VC4CLStdLib.h.pch /usr/local/include/vc4cl-stdlib/VC4CLStdLib.h
[E] Mon Jun 22 16:54:00 2020: Errors in precompilation:
[E] Mon Jun 22 16:54:00 2020: error: unknown argument: '-target'
distcc[891] ERROR: compile (null) on localhost failed
distcc[890] ERROR: compile (null) on localhost failed

terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: error: unknown argument: '-target'
distcc[891] ERROR: compile (null) on localhost failed
distcc[890] ERROR: compile (null) on localhost failed

[E] Mon Jun 22 16:54:00 2020: Received signal: SIGABRT
```

doe300, 2020-06-23:
>  Can't you bypass any front-end of clang and invoke clang directly from vc4c (i.e. hard code a path? Or have an optoin to identify clang's path)?

I am using a hardcoded path to clang (see the output of `vc4c --version` above), but for your setup this clang is the distcc front-end.

Given the error message, that it fails on the `-target` argument, which is not explicitly passed by VC4C to the clang (distcc front-end), I assume that distcc adds it before actually calling clang.

Can you reconfigure and rebuild VC4C with the CMake option `-DCLANG_FOUND=<path/to/actual/clang/executable>`? This sets/updates the hardcoded clang path and should prevent from the VC4C using distcc on runtime.

bronze51, 2020-06-23:
I assume my clan (v10) is better than the required v3.9?
I got:

```
-- Using CMake 3.14+ FetchContent to include dependencies...
-- CLang compiler found: /usr/bin
CMake Error at cmake/clang.cmake:19 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  CMakeLists.txt:120 (include)

CMake Error at cmake/clang.cmake:21 (message):
  CLang version is not supported, update your CLang to version 3.9+!
Call Stack (most recent call first):
  CMakeLists.txt:120 (include)

-- Clang opt found: /usr/bin/opt
-- LLVM-link found: /usr/bin/llvm-link
-- Compiling LLVM library front-end with LLVM in version 10.0.0 located in '/usr/lib'
```

doe300, 2020-06-24:
Strange, manual tests using "clang version 10.0.0" as `clang_full_version_string` work as expected.

The CMake variable `clang_full_version` should not exist. Can you just add some prints for `clang_full_version_string` as well as `CLANG_VERSION_STRING`?

As a side note, in the code snippet you pasted, in the line

    string (REGEX REPLACE ".*clang version ([0-9]+\\.[0-9]+).*" "\\1" CLANG_VERSION_STRING {clang_full_version_string})

there should be a `$` before the opening bracket:

    string (REGEX REPLACE ".*clang version ([0-9]+\\.[0-9]+).*" "\\1" CLANG_VERSION_STRING ${clang_full_version_string})

bronze51, 2020-06-24:
I made a mistake when setting CLANG_FOUND variable. I initially set it to `-DCLANG_FOUND=/usr/bin/`
When setting it to `-DCLANG_FOUND=/usr/bin/clang` things work better ...

Compiling right now ...

bronze51, 2020-06-24:
OK, I did find this message that surprised me:

```
CMake Warning:
  Manually-specified variables were not used by the project:

    CLANG_FOUND

```

bronze51, 2020-06-24:
To be clear, I am compiling this package: https://aur.archlinux.org/packages/vc4c-git/

Looking at the error `error reading '/VC4CLStdLib.h'` I copied "VC4CLStdLib.h" to the root and that changed the error to:

```
[E] Wed Jun 24 16:39:52 2020: Errors in precompilation:
[E] Wed Jun 24 16:39:52 2020: /VC4CLStdLib.h:17:10: fatal error: '_config.h' file not found
#include "_config.h"
         ^~~~~~~~~~~
1 error generated.

terminate called after throwing an instance of 'vc4c::CompilationError'
```

So the following line in the cmake file seems to have an empty variable for `${VC4CL_STDLIB_DIR}`

```
install(CODE "execute_process(COMMAND ${VC4C_PROGRAM_LOCATION} --quiet --precompile-stdlib -o $DESTDIR/${CMAKE_INSTALL_PREFIX}/include/vc4cl-stdlib/ ${VC4CL_STDLIB_DIR}/VC4CLStdLib.h)")
```

doe300, 2020-06-25:
To be able to build VC4C, you also need the sources for VC4CLStdLib. There are several options where the VC4C CMake autodetects the location of the VC4CLStdLib sources:

- If you have a folder `<some/path>/VC4C` containing the VC4C sources and a folder `<some/path>/VC4CLStdLib` containing the VC4CLStdLib sources (i.e. both folders are siblings in the same parent folder)
- If you have the VC4CLStdLib source folder inside the VC4C source folder

In any other constellation, you need to specify the `VC4CL_STDLIB_DIR` CMake variable manually and set it to the `<path/to>/VC4CLStdLib/include` directory.

Note: This is only required for building, on "normal" runtime, the VC4C compiler also checks the `/usr/local/include/vc4cl-stdlib` and `/usr/include/vc4cl-stdlib` folders for the precompiled standard library files.