doe300/VC4CL #93: ocl-icd 2.2.12-4: ocl-icd>=1.3

bronze51, 2020-06-27:
On my raspberry pi, I have ocl-icd 2.2, but the build process fails with:

```
-- Checking for one of the modules 'ocl-icd>=1.3'
CMake Error at /usr/share/cmake-3.17/Modules/FindPkgConfig.cmake:745 (message):
  None of the required 'ocl-icd>=1.3' found
Call Stack (most recent call first):
  cmake/icd.cmake:6 (pkg_search_module)
  CMakeLists.txt:81 (include)
```
I am not sure whether I should get khronos-ocl-icd (1.2.11.0-1) or keep ocl-icd (2.2.12-4).

doe300, 2020-06-28:
Do you also have the development headers installed, i.e. `ocl-icd-dev` and `ocl-icd-opencl-dev`?

bronze51, 2020-06-30:
On Arch, the development files are included in the "base" package (`ocl-icd`). I found:

```
    usr/include/ocl_icd.h
    usr/lib/libOpenCL.so
    usr/lib/libOpenCL.so.1
    usr/lib/libOpenCL.so.1.0.0
    usr/lib/pkgconfig/OpenCL.pc
    usr/lib/pkgconfig/ocl-icd.pc
    usr/share/doc/ocl-icd/examples/ocl_icd_bindings.c
    usr/share/doc/ocl-icd/html/libOpenCL.html
    usr/share/licenses/ocl-icd/COPYING
    usr/share/man/man7/libOpenCL.7.gz
    usr/share/man/man7/libOpenCL.so.7.gz
```

doe300, 2020-06-30:
That is the file Cmake should find: `/usr/lib/pkgconfig/ocl-icd.pc`. So it looks like it is provided on your system...

1. Can you check the version reported in this file?
2. Also, can you try setting the `PKG_CONFIG_PATH` environment variable to `/usr/lib/pkgconfig/` before running the CMake command?

bronze51, 2020-07-01:
`/usr/lib/pkgconfig/ocl-icd.pc` :

```
prefix=/usr
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: ocl-icd
Description: Open Computing Language generic Installable Client Driver support
Version: 2.2.12
Libs: -L${libdir}
Cflags: -I${includedir}
```

I ran `cmake -DBUILD_TESTING=OFF -DSPIRV_FRONTEND=OFF -DCLANG_FOUND=/usr/bin/clang -DPKG_CONFIG_PATH=/usr/lib/pkgconfig/ .` :

```
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- ...
-- Detecting CXX compile features - done
-- Looking for CL_VERSION_2_2
-- Looking for CL_VERSION_2_2 - found
-- Found OpenCL: /usr/lib/libOpenCL.so (found version "2.2")
-- Found headers for OpenCL up to version 2.2 located in /usr/include
-- VC4C library found: /usr/local/lib/libVC4CC.so
-- VC4C compiler header found at: /home/jlbakker/dev/VC4C/include/VC4C.h
-- Building with ICD support
-- Checking for one of the modules 'ocl-icd>=1.3'
CMake Error at /usr/share/cmake-3.17/Modules/FindPkgConfig.cmake:745 (message):
  None of the required 'ocl-icd>=1.3' found
Call Stack (most recent call first):
  cmake/icd.cmake:6 (pkg_search_module)
  CMakeLists.txt:81 (include)
-- Found Khronos ICD Loader in version  in
-- Looking for clCreateProgramWithIL
-- Looking for clCreateProgramWithIL - found
-- Enabling register-poking to run kernels
-- ...
-- Configuring incomplete, errors occurred!
```

doe300, 2020-07-04:
Okay, so setting `PKG_CONFIG_EXECUTABLE` is wrong.

Can you try with the https://aur.archlinux.org/packages/khronos-ocl-icd/ package instead of the `ocl-icd`?

bronze51, 2020-07-15:
I have replaced ocl-icd with khronos-ocl-icd. Eventually, this allowed me to build VC4C! However, VC4CL fails with:

```
-- Checking for one of the modules 'ocl-icd>=1.3'
CMake Error at /usr/share/cmake-3.17/Modules/FindPkgConfig.cmake:745 (message):
  None of the required 'ocl-icd>=1.3' found
Call Stack (most recent call first):
  cmake/icd.cmake:6 (pkg_search_module)
  CMakeLists.txt:81 (include)

-- Found Khronos ICD Loader in version  in
-- Looking for clCreateProgramWithIL
-- Looking for clCreateProgramWithIL - not found
CMake Warning at cmake/icd.cmake:14 (message):
  clCreateProgramWithIL not found, strongly recommend to upgrade package
  opencl-headers!
Call Stack (most recent call first):
  CMakeLists.txt:81 (include)
```

doe300, 2020-07-15:
I would assume that is now since VC4CL only looks for the `ocl-icd` package and not for the `khronos-ocl-icd` package.

If you modify https://github.com/doe300/VC4CL/blob/master/cmake/icd.cmake#L6 to look for `khronos-ocl-icd`, that might do the trick.

mixeddmi, 2020-10-26:
I had a similar build issue with ocl-icd on Arch Linux.  Installing the pgkconfig  package ( pacman -S pkgconfig ) corrected the problem for me.

closed 2021-02-09 by doe300