doe300/VC4CL #94: [wiki] nighlty debian packages requires clang-3.9, not available anymore

shazz, 2020-07-11:
Hi,

I tried the "The easy way" but the nightly builds requries clang-3-9 which is not available anymore on latest raspbian distros:
````
dpkg: dependency problems prevent configuration of vc4c:
 vc4c depends on clang-3.9; however:
  Package clang-3.9 is not installed.
 vc4c depends on llvm-3.9; however:
  Package llvm-3.9 is not installed.
 vc4c depends on llvm-3.9-dev; however:
  Package llvm-3.9-dev is not installed.
````

````
sudo apt-get install clang-
clang-5.0
clang-8
clang-6.0              
clang-9
clang-7               
````

Would it be possible to update the CI top use clang-3.9 ?

Moreover in the wget/curl commands, `python` has to be replaced by `python3`

Thanks!

doe300, 2020-07-12:
See https://github.com/doe300/VC4C/issues/138 for the discussion on upgrading the clang/llvm versions.

shazz, 2020-07-13:
ok, looks complicated... I have the same problem with pocl, requires LLVM 3.x....

closed 2020-07-13 by shazz