doe300/VC4CL #97: support arm64

jasonwee, 2020-09-05:
Hi,

this documentation is helpful, https://github.com/doe300/VC4CL/wiki/How-to-get but there is something wrong with this command output.

```
curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C/$(python ./build_num.py /tmp/json)/artifacts" --output /tmp/dump
```

it return html, I have to change branch to devel and then able to proceed to the next line of documentation.

also, it seem that the package is build for armhf? is it possible to include support for arm64? That would be fantastic and my o/s is ubuntu server.

Thank you.

doe300, 2020-09-05:
Yeah, CircleCI by default only lists the last few builds, which may result in no "master" build being in the list if there was heavy activity on other branches, see also the latest VC4CL devel change [here](https://github.com/doe300/VC4CL/commit/76edc680a268afc6abf81a822c61f2a5f9eae2e9#diff-1d37e48f9ceff6d8030570cd36286a61R16).

I have no clue how good the support for the mailbox interface and the V3D hardware registers is on arm64. And only if these work properly, I might consider supporting 64-bit system. But this will probably be a lot of effort...

jasonwee, 2020-09-05:
yes, thanks, it works and so maybe should change the line in the doc from

```
curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C" --output /tmp/json
```

to 

```
curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C?limit=100&filter=successful" --output /tmp/json
```

and i got the debs.
```
ubuntu@ubuntu:/tmp$ sudo dpkg -i vc4cl-stdlib.deb
dpkg: warning: downgrading vc4cl-stdlib:armhf from 0.4-2020-09-02 to 0.4-2020-08-18
(Reading database ... 116299 files and directories currently installed.)
Preparing to unpack vc4cl-stdlib.deb ...
Unpacking vc4cl-stdlib:armhf (0.4-2020-08-18) over (0.4-2020-09-02) ...
Setting up vc4cl-stdlib:armhf (0.4-2020-08-18) ...
ubuntu@ubuntu:/tmp$ sudo dpkg -i vc4c.deb
Selecting previously unselected package vc4c:armhf.
(Reading database ... 116299 files and directories currently installed.)
Preparing to unpack vc4c.deb ...
Unpacking vc4c:armhf (0.4.2048-2020-08-18) ...
dpkg: dependency problems prevent configuration of vc4c:armhf:
 vc4c:armhf depends on clang-6.0.
 vc4c:armhf depends on llvm-6.0.
 vc4c:armhf depends on llvm-6.0-dev.

dpkg: error processing package vc4c:armhf (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 vc4c:armhf
ubuntu@ubuntu:/tmp$ 
ubuntu@ubuntu:/tmp$ dpkg --print-architecture
arm64
ubuntu@ubuntu:/tmp$ sudo dpkg --add-architecture armhf
ubuntu@ubuntu:/tmp$ uname -a
Linux ubuntu 5.4.0-1015-raspi #15-Ubuntu SMP Fri Jul 10 05:34:24 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:/tmp$ 
```
yes, if you need assistance, please let me know. coding, testing, just send me a message and I will do my best to help. But I need some guidance where to begin offer assistance.

doe300, 2020-09-05:
The VC4CC compiler does not care about 32-bit or 64-bit system, it also runs well on my 64-bit host machine.

For the VC4CL run-time, there are a few things have to be figured out/tested/fixed:
- [ ] Does the 64-bit OS even support access to the mailbox interface?
- [ ] Does VC4CL build on a 64-bit OS (without cross-compiling or mocked hardware)? Can it correctly link against the `/opt/vc/lib/bcm_host.so` library, i.e. is this library provided as 64-bit library?
- [ ] Does the syscall and hardware register access still work, e.g. can kernels be executed? There is a good chance that I have some unchecked assumption somewhere that the host-pointers are 32-bit or that some syscall does not work without changes.

jasonwee, 2020-09-05:
sorry, do you mean this?

```
ubuntu@ubuntu:~$ locate -b '\Mailbox.h'
ubuntu@ubuntu:~$ 
```

can you send the steps for the 3 points? I can paste you the output of the commands output.

doe300, 2020-09-05:
I don't know any fixed steps, at least not for points 1 and 3, but something like this:

1. So according to https://github.com/raspberrypi/userland/pull/455, the `bcm_host.so` library builds for arm64, but https://github.com/raspberrypi/userland/issues/391#issuecomment-363059735 lists a few problems to actually make it work. So the task here where to run some tests, whether it works as expected. If 2. works, this can also be combined with 3.
2. This is the most straight-forward. To test this, VC4CL needs to be just compiled natively on an arm64 system and checked whether the compilation works without a problem.
3. If 2. is done, at least the basic functionality can be tested by trying to run https://github.com/Oblomov/clinfo or https://github.com/krrishnarraj/clpeak with the natively compiled VC4CL library.

jasonwee, 2020-09-08:
i tried on another version of ubuntu, same error
```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.4.0-1015-raspi #15~18.04.1-Ubuntu SMP Fri Jul 10 09:20:51 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ uname -r
5.4.0-1015-raspi
ubuntu@ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
ubuntu@ubuntu:~$ 
```
don't have knowledge on c nor cpp, think cannot help much though but will try again this weekend

jasonwee, 2020-10-01:
i follow the instruction in https://github.com/doe300/VC4CL/blob/master/.circleci/config.yml
```
idein@1681b1c7be7f:~/VC4CL/build$ ls src/libVC4CL.so.0.4.1 
src/libVC4CL.so.0.4.1
idein@1681b1c7be7f:~/VC4CL/build$ ls test/TestVC4CL 
test/TestVC4CL
idein@1681b1c7be7f:~/VC4CL/build$ vc4cl-0.4-Linux.deb
bash: vc4cl-0.4-Linux.deb: command not found
idein@1681b1c7be7f:~/VC4CL/build$ ls vc4cl-0.4-Linux.deb
vc4cl-0.4-Linux.deb
idein@1681b1c7be7f:~/VC4CL/build$ 
```

jasonwee, 2020-10-01:
when i run this step in container
```
      - run: dpkg -i /tmp/vc4cl-stdlib.deb /tmp/vc4c.deb build/vc4cl-0.4-Linux.deb
```

```
root@82dfc1affaad:~/project# dpkg -i /tmp/vc4cl-stdlib.deb /tmp/vc4c.deb vc4cl-0.4-Linux.deb 
(Reading database ... 23334 files and directories currently installed.)
Preparing to unpack /tmp/vc4cl-stdlib.deb ...
Unpacking vc4cl-stdlib (0.4-2020-09-30) over (0.4-2020-09-30) ...
Preparing to unpack /tmp/vc4c.deb ...
Unpacking vc4c (0.4.2156-2020-09-30) over (0.4.2156-2020-09-30) ...
dpkg: error processing archive vc4cl-0.4-Linux.deb (--install):
 package architecture (armhf) does not match system (amd64)
Setting up vc4cl-stdlib (0.4-2020-09-30) ...
Setting up vc4c (0.4.2156-2020-09-30) ...
[VC4C] Extended library load path by: /usr/local/lib
[E] Thu Oct  1 14:28:21 2020: Errors in precompilation:
[E] Thu Oct  1 14:28:21 2020: sh: 1: /usr/bin/clang-7: not found

[E] Thu Oct  1 14:28:21 2020:  (1) /usr/local/lib/libVC4CC.so.1.2 : vc4c::CompilationError::CompilationError(vc4c::CompilationStep, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x121 [0x7f4c18acca37]
[E] Thu Oct  1 14:28:21 2020:  (2) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x1cf [0x7f4c18e8f3f6]
[E] Thu Oct  1 14:28:21 2020:  (3) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x2c6 [0x7f4c18ea7650]
[E] Thu Oct  1 14:28:21 2020:  (4) /usr/local/bin/vc4c : main+0xa63 [0x563c7422a8a2]
[E] Thu Oct  1 14:28:21 2020:  (5) /lib/x86_64-linux-gnu/libc.so.6 : __libc_start_main+0xeb [0x7f4c14146bbb]
[E] Thu Oct  1 14:28:21 2020:  (6) /usr/local/bin/vc4c : _start+0x2a [0x563c742285ea]
terminate called after throwing an instance of 'vc4c::CompilationError'
  what():  Pre-compilation: Error in precompilation: sh: 1: /usr/bin/clang-7: not found

[E] Thu Oct  1 14:28:21 2020: Received signal: SIGABRT
[E] Thu Oct  1 14:28:21 2020:  (1) /usr/local/lib/libVC4CC.so.1.2 : +0x93f9f4 [0x7f4c18b2c9f4]
[E] Thu Oct  1 14:28:21 2020:  (2) /lib/x86_64-linux-gnu/libpthread.so.0 : +0x13520 [0x7f4c181da520]
[E] Thu Oct  1 14:28:21 2020:  (3) /lib/x86_64-linux-gnu/libc.so.6 : gsignal+0x141 [0x7f4c14159f61]
[E] Thu Oct  1 14:28:21 2020:  (4) /lib/x86_64-linux-gnu/libc.so.6 : abort+0x121 [0x7f4c14145535]
[E] Thu Oct  1 14:28:21 2020:  (5) /usr/lib/x86_64-linux-gnu/libstdc++.so.6 : +0x9a80a [0x7f4c146d880a]
[E] Thu Oct  1 14:28:21 2020:  (6) /usr/lib/x86_64-linux-gnu/libstdc++.so.6 : +0xa5906 [0x7f4c146e3906]
[E] Thu Oct  1 14:28:21 2020:  (7) /usr/lib/x86_64-linux-gnu/libstdc++.so.6 : +0xa5971 [0x7f4c146e3971]
[E] Thu Oct  1 14:28:21 2020:  (8) /usr/lib/x86_64-linux-gnu/libstdc++.so.6 : +0xa5c15 [0x7f4c146e3c15]
[E] Thu Oct  1 14:28:21 2020:  (9) /usr/local/lib/libVC4CC.so.1.2 : vc4c::precompilation::runPrecompiler(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::istream*, std::ostream*)+0x20f [0x7f4c18e8f436]
[E] Thu Oct  1 14:28:21 2020:  (10) /usr/local/lib/libVC4CC.so.1.2 : vc4c::Precompiler::precompileStandardLibraryFiles(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0x2c6 [0x7f4c18ea7650]
[E] Thu Oct  1 14:28:21 2020:  (11) /usr/local/bin/vc4c : main+0xa63 [0x563c7422a8a2]
[E] Thu Oct  1 14:28:21 2020:  (12) /lib/x86_64-linux-gnu/libc.so.6 : __libc_start_main+0xeb [0x7f4c14146bbb]
[E] Thu Oct  1 14:28:21 2020:  (13) /usr/local/bin/vc4c : _start+0x2a [0x563c742285ea]
dpkg: error processing package vc4c (--install):
 subprocess installed post-installation script returned error exit status 6
Errors were encountered while processing:
 vc4cl-0.4-Linux.deb
 vc4c
root@82dfc1affaad:~/project# clang
clang++-3.9                   clang-apply-replacements-3.9  clang-cl-3.9                  clang-format-7                clang-include-fixer-3.9       clang-rename-3.9
clang-3.9                     clang-check-3.9               clang-format                  clang-format-diff-7           clang-query-3.9               clang-tblgen-3.9
root@82dfc1affaad:~/project# clang-7^C
root@82dfc1affaad:~/project# apt-cache search clang-7
root@82dfc1affaad:~/project# apt search
E: You must give at least one search pattern
root@82dfc1affaad:~/project# apt search foo
Sorting... Done
Full Text Search... Done
root@82dfc1affaad:~/project# apt-get update
Get:1 http://security.debian.org/debian-security stretch/updates InRelease [53.0 kB]
Get:2 http://security.debian.org/debian-security stretch/updates/main amd64 Packages [600 kB]
Ign:3 http://deb.debian.org/debian stretch InRelease                     
Get:4 http://ftp.debian.org/debian unstable InRelease [146 kB]
Get:5 http://deb.debian.org/debian stretch-updates InRelease [93.6 kB]
Get:6 http://deb.debian.org/debian stretch Release [118 kB]       
Get:7 http://deb.debian.org/debian stretch-updates/main amd64 Packages [2596 B]                                                                                                              
[... cut ...]
The following packages have unmet dependencies:
 libclang-common-7-dev : Depends: lib32gcc1 (>= 1:4.2) but it is not going to be installed
                         Depends: lib32stdc++6 (>= 4.1.1) but it is not going to be installed
                         Depends: libc6-i386 (>= 2.17) but it is not going to be installed
                         Depends: libllvm7 (= 1:7.0.1-8~deb9u4) but 1:7.0.1-10 is to be installed
E: Unable to correct problems, you have held broken packages.
root@82dfc1affaad:~/project# 

```

jasonwee, 2020-10-02:
never that failures, because this time I got a little bit further.

```
jason@localhost:~$ docker pull doe300/native
Using default tag: latest
latest: Pulling from doe300/native
57df1a1f1ad8: Pull complete 
ac9d4d5e2db5: Pull complete 
ee030873f186: Pull complete 
Digest: sha256:bfe38a2a13dae54a8b6465a5485fa446f92f5af66d002bafe4cac5977ae8583c
Status: Downloaded newer image for doe300/native:latest
docker.io/doe300/native:latest
jason@localhost:~$ docker run -it --rm doe300/native bash
root@f4bf62ff788d:/# git clone https://github.com/doe300/VC4CL.git
Cloning into 'VC4CL'...
remote: Enumerating objects: 152, done.
remote: Counting objects: 100% (152/152), done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 1700 (delta 85), reused 86 (delta 51), pack-reused 1548
Receiving objects: 100% (1700/1700), 1.00 MiB | 264.00 KiB/s, done.
Resolving deltas: 100% (1259/1259), done.
root@f4bf62ff788d:/# curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C?limit=100&filter=successful" --output /tmp/json
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  406k    0  406k    0     0  45990      0 --:--:--  0:00:09 --:--:-- 87025
root@f4bf62ff788d:/# #curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C/$(Vcircleci/build_num.py /tmp/json build)/artifacts" --output /tmp/dump
root@f4bf62ff788d:/# ls
VC4CL  boot  etc   lib	  lib64  mnt  proc  run   srv  tmp  var
bin    dev   home  lib32  media  opt  root  sbin  sys  usr
root@f4bf62ff788d:/# mv VC4CL/ root/
root@f4bf62ff788d:/# ls
bin   dev  home  lib32	media  opt   root  sbin  sys  usr
boot  etc  lib	 lib64	mnt    proc  run   srv	 tmp  var
root@f4bf62ff788d:/# cd root/
root@f4bf62ff788d:~# ls
VC4CL
root@f4bf62ff788d:~# cd VC4CL/
root@f4bf62ff788d:~/VC4CL# ls
CMakeLists.txt	LICENSE  Readme.md  cmake  src	test  tools
root@f4bf62ff788d:~/VC4CL# curl "https://circleci.com/api/v1.1/project/github/doe300/VC4C/$(.circleci/build_num.py /tmp/json build)/artifacts" --output /tmp/dump
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   599  100   599    0     0    460      0  0:00:01  0:00:01 --:--:--   460
root@f4bf62ff788d:~/VC4CL# apt-get update
Get:1 http://deb.debian.org/debian buster InRelease [121 kB]
Get:2 http://deb.debian.org/debian buster-updates InRelease [51.9 kB]
Get:3 http://deb.debian.org/debian buster/main amd64 Packages [7906 kB]
Get:4 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:5 http://security.debian.org/debian-security buster/updates/main amd64 Packages [234 kB]
Get:6 http://deb.debian.org/debian buster-updates/main amd64 Packages [7868 B]               
Fetched 8387 kB in 8s (1057 kB/s)                                                            
Reading package lists... Done
root@f4bf62ff788d:~/VC4CL# apt-get install vim
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libgpm2 vim-common vim-runtime xxd
Suggested packages:
  gpm ctags vim-doc vim-scripts
The following NEW packages will be installed:
  libgpm2 vim vim-common vim-runtime xxd
0 upgraded, 5 newly installed, 0 to remove and 10 not upgraded.
Need to get 7425 kB of archives.
After this operation, 33.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://deb.debian.org/debian buster/main amd64 xxd amd64 2:8.1.0875-5 [140 kB]
Get:2 http://deb.debian.org/debian buster/main amd64 vim-common all 2:8.1.0875-5 [195 kB]
Get:3 http://deb.debian.org/debian buster/main amd64 libgpm2 amd64 1.20.7-5 [35.1 kB]
Get:4 http://deb.debian.org/debian buster/main amd64 vim-runtime all 2:8.1.0875-5 [5775 kB]
[... 485 lines cut ...]
	Error: No such file or directory
Test 'TestProgram::testCompileProgram()' failed!
	Suite: TestProgram
	File: TestProgram.cpp
	Line: 148
	Failure: Got -44, expected 0
^C
root@f4bf62ff788d:~/project#
root@f4bf62ff788d:~/project# cat /etc/issue.net 
Debian GNU/Linux 10
root@f4bf62ff788d:~/project# uname -a           
Linux f4bf62ff788d 5.2.0-3-amd64 #1 SMP Debian 5.2.17-1 (2019-09-26) x86_64 GNU/Linux
root@f4bf62ff788d:~/project# uname -r
5.2.0-3-amd64
root@f4bf62ff788d:~/project# cat /etc/deb
debconf.conf    debian_version  
root@f4bf62ff788d:~/project# cat /etc/debian_version 
10.5
root@f4bf62ff788d:~/project# dpkg --print-architecture
amd64
root@f4bf62ff788d:~/project# 

```

now I just need to get the right debian os and figure out how to get the package install into rpi3