doe300/VC4CL #99: [System-Error] application hang when launching a simple example (HelloWorld)

sunneo, 2020-10-25:
https://github.com/rsnemmen/OpenCL-examples/tree/master/Hello_World
To make this program launch,
I change the SIZE to 16
and let local become 4 because the workgroupSize 12 cannot divide 1024
after build with `gcc -o hello.exe hello.c -lOpenCL`
the test program stuck at launch kernel.

[VC4CL](      hello.exe): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x5da5e4, const char* "clIcdGetPlatformIDsKHR")
[VC4CL](      hello.exe): get extension function address: clIcdGetPlatformIDsKHR
[VC4CL](      hello.exe): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x5da5e4, const char* "clGetPlatformInfo")
[VC4CL](      hello.exe): get extension function address: clGetPlatformInfo
[VC4CL](      hello.exe): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 0, cl_platform_id* 0, cl_uint* 0xbee72d48)
[VC4CL](      hello.exe): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 1, cl_platform_id* 0x5d56b8, cl_uint* 0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x5da5e4, cl_platform_info 2308, size_t 0, void* 0, size_t* 0xbee72ce0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x5da5e4, cl_platform_info 2308, size_t 226, void* 0x5e6458, size_t* 0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x5da5e4, cl_platform_info 2336, size_t 0, void* 0, size_t* 0xbee72ce0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x5da5e4, cl_platform_info 2336, size_t 6, void* 0x5da630, size_t* 0)
[VC4CL](      hello.exe): API call: cl_int clGetDeviceIDs(cl_platform_id 0x5da5e4, cl_device_type 4, cl_uint 1, cl_device_id* 0xbee73634, cl_uint* 0)
[VC4CL](      hello.exe): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0xbee73634, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0xbee72d18, void* 0, cl_int* 0xbee75640)
[VC4CL](      hello.exe): Tracking live-time of object: 0x5e587c (cl_context)
[VC4CL](      hello.exe): API call: cl_command_queue clCreateCommandQueue(cl_context 0x5e587c, cl_device_id 0x5da5f8, cl_command_queue_properties 0, cl_int* 0xbee75640)
[VC4CL](      hello.exe): Starting queue handler thread...
[VC4CL](      hello.exe): Tracking live-time of object: 0x5e441c (cl_command_queue)
[VC4CL](      hello.exe): API call: cl_program clCreateProgramWithSource(cl_context 0x5e587c, cl_uint 1, const char** 0x22080, const size_t* 0, cl_int* 0xbee75640)
[VC4CL](      hello.exe): Tracking live-time of object: 0x5a7744 (cl_program)
[VC4CL](      hello.exe): API call: cl_int clBuildProgram(cl_program 0x5a7744, cl_uint 0, const cl_device_id* 0, const char* (null), void(CL_CALLBACK*)(cl_program program, void* user_data) 0xbee72e00, void* 0)
[VC4CL](      hello.exe): Precompiling source with:
Dumping program sources to /tmp/vc4cl-source-386839851.cl
[VC4CL](      hello.exe): Dumping program IR to /tmp/vc4cl-ir-771476364.ll
[VC4CL](      hello.exe): Precompilation complete with status: 0
[VC4CL](      hello.exe): [VC4CL] base=0x3fc00000, mem=0xb6f54000
[VC4CL](      hello.exe): [VC4CL] V3D base: 0xb6f54000
[VC4CL](      hello.exe): Compiling source with:
[VC4CL](      hello.exe): Compilation complete with status: 0
Dumping program binaries to /tmp/vc4cl-binary-942724790.bin
[VC4CL](      hello.exe): API call: cl_kernel clCreateKernel(cl_program 0x5a7744, const char* "square", cl_int* 0xbee75640)
[VC4CL](      hello.exe): Tracking live-time of object: 0x5a8324 (cl_kernel)
[VC4CL](      hello.exe): API call: cl_mem clCreateBuffer(cl_context 0x5e587c, cl_mem_flags 4, size_t 4096, void* 0, cl_int* 0)
[VC4CL](      hello.exe): [VC4CL] Mailbox file descriptor opened: 4
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 80000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 38000000 08000000 00000000
[VC4CL](      hello.exe): Mailbox request: succeeded
[VC4CL](      hello.exe): Tracking live-time of object: 0x5da69c (cl_mem)
[VC4CL](      hello.exe): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00001000 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 00000004 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000004 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be8f8000 00000000 00000000
[VC4CL](      hello.exe): [VC4CL] base=0x3e8f8000, mem=0xb6f53000
[VC4CL](      hello.exe): Allocated 4096 bytes of buffer: handle 4, device address 0xbe8f8000, host address 0xb6f53000
[VC4CL](      hello.exe): API call: cl_mem clCreateBuffer(cl_context 0x5e587c, cl_mem_flags 2, size_t 4096, void* 0, cl_int* 0)
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 38000000 08000000 00000000
[VC4CL](      hello.exe): Mailbox request: succeeded
[VC4CL](      hello.exe): Tracking live-time of object: 0x5da97c (cl_mem)
[VC4CL](      hello.exe): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00001000 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 00000014 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000014 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be8f4000 00000000 00000000
[VC4CL](      hello.exe): [VC4CL] base=0x3e8f4000, mem=0xb6f52000
[VC4CL](      hello.exe): Allocated 4096 bytes of buffer: handle 20, device address 0xbe8f4000, host address 0xb6f52000
[VC4CL](      hello.exe): API call: cl_int clEnqueueWriteBuffer(cl_command_queue 0x5e441c, cl_mem 0x5da69c, cl_bool 1, size_t 0, size_t 4096, void* 0xbee74640, cl_uint 0, const cl_event* 0, cl_event* 0)
[VC4CL](      hello.exe): Tracking live-time of object: 0x5d08cc (cl_event)
[VC4CL](      hello.exe): Releasing live-time of object: 0x5d08cc (cl_event)
[VC4CL](      hello.exe): API call: cl_int clSetKernelArg(cl_kernel 0x5a8324, cl_uint 0, size_t 4, const void* 0xbee73630)
[VC4CL](      hello.exe): Set kernel arg 0 for kernel 'square' to 0xbee73630 (6137500) with size 4
[VC4CL](      hello.exe): Kernel arg 0 for kernel 'square' is float* 'input' with size 4
[VC4CL](      hello.exe): Setting kernel-argument 0 to pointer 0x0x5da690
[VC4CL](      hello.exe): API call: cl_int clSetKernelArg(cl_kernel 0x5a8324, cl_uint 1, size_t 4, const void* 0xbee7362c)
[... 79 lines cut ...]
[VC4CL](      hello.exe): Precompiling source with:
Dumping program sources to /tmp/vc4cl-source-1365180540.cl
[VC4CL](      hello.exe): Dumping program IR to /tmp/vc4cl-ir-1540383426.ll
[VC4CL](      hello.exe): Precompilation complete with status: 0
[VC4CL](      hello.exe): [VC4CL] base=0x3fc00000, mem=0xb6fcf000
[VC4CL](      hello.exe): [VC4CL] V3D base: 0xb6fcf000
[VC4CL](      hello.exe): Compiling source with:
[VC4CL](      hello.exe): Compilation complete with status: 0
Dumping program binaries to /tmp/vc4cl-binary-304089172.bin
[VC4CL](      hello.exe): API call: cl_kernel clCreateKernel(cl_program 0x1b9744, const char* "square", cl_int* 0xbe9fe640)
[VC4CL](      hello.exe): Tracking live-time of object: 0x1ba324 (cl_kernel)
[VC4CL](      hello.exe): API call: cl_mem clCreateBuffer(cl_context 0x1f787c, cl_mem_flags 4, size_t 64, void* 0, cl_int* 0)
[VC4CL](      hello.exe): [VC4CL] Mailbox file descriptor opened: 4
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 80000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[... cut ...]
[VC4CL](      hello.exe): Mailbox request: succeeded
[VC4CL](      hello.exe): Tracking live-time of object: 0x1ec69c (cl_mem)
[VC4CL](      hello.exe): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00000040 00001000 0000000c 00000000
[VC4CL](      hello.exe): ioctl_set_msg failed: -1
[VC4CL] Error in mbox_property: Connection timed out
terminate called after throwing an instance of 'std::system_error'
  what():  Failed to set mailbox property: Connection timed out
terminate called recursively

doe300, 2020-10-25:
>  I change the SIZE to 16

Which size did you set to 16?

Also, how long did you approximately wait before terminating the process?

sunneo, 2020-10-25:
> > I change the SIZE to 16
> 
> Which size did you set to 16?
#define DATA_SIZE (16)
> 
> Also, how long did you approximately wait before terminating the process?
it's about 1 minute, because it takes too long.

I've rebooted and re-run it, and I got execution failed
this time I didn't terminate it. and I just wait for execution finish.

```
[VC4CL](      hello.exe): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x194b5e4, const char* "clIcdGetPlatformIDsKHR")
[VC4CL](      hello.exe): get extension function address: clIcdGetPlatformIDsKHR
[VC4CL](      hello.exe): API call: void* clGetExtensionFunctionAddressForPlatform(cl_platform_id 0x194b5e4, const char* "clGetPlatformInfo")
[VC4CL](      hello.exe): get extension function address: clGetPlatformInfo
[VC4CL](      hello.exe): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 0, cl_platform_id* 0, cl_uint* 0xbeb57c88)
[VC4CL](      hello.exe): API call: cl_int clIcdGetPlatformIDsKHR(cl_uint 1, cl_platform_id* 0x19466b8, cl_uint* 0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x194b5e4, cl_platform_info 2308, size_t 0, void* 0, size_t* 0xbeb57c20)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x194b5e4, cl_platform_info 2308, size_t 226, void* 0x1957458, size_t* 0)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x194b5e4, cl_platform_info 2336, size_t 0, void* 0, size_t* 0xbeb57c20)
[VC4CL](      hello.exe): API call: cl_int clGetPlatformInfo(cl_platform_id 0x194b5e4, cl_platform_info 2336, size_t 6, void* 0x194b630, size_t* 0)
[VC4CL](      hello.exe): API call: cl_int clGetDeviceIDs(cl_platform_id 0x194b5e4, cl_device_type 4, cl_uint 1, cl_device_id* 0xbeb58574, cl_uint* 0)
[VC4CL](      hello.exe): API call: cl_context clCreateContext(const cl_context_properties* 0, cl_uint 1, const cl_device_id* 0xbeb58574, void(CL_CALLBACK*)(const char* errinfo, const void* private_info, size_t cb, void* user_data) 0xbeb57c58, void* 0, cl_int* 0xbeb58600)
[VC4CL](      hello.exe): Tracking live-time of object: 0x195687c (cl_context)
[VC4CL](      hello.exe): API call: cl_command_queue clCreateCommandQueue(cl_context 0x195687c, cl_device_id 0x194b5f8, cl_command_queue_properties 0, cl_int* 0xbeb58600)
[VC4CL](      hello.exe): Starting queue handler thread...
[VC4CL](      hello.exe): Tracking live-time of object: 0x195541c (cl_command_queue)
[VC4CL](      hello.exe): API call: cl_program clCreateProgramWithSource(cl_context 0x195687c, cl_uint 1, const char** 0x22080, const size_t* 0, cl_int* 0xbeb58600)
[VC4CL](      hello.exe): Tracking live-time of object: 0x1918744 (cl_program)
[VC4CL](      hello.exe): API call: cl_int clBuildProgram(cl_program 0x1918744, cl_uint 0, const cl_device_id* 0, const char* (null), void(CL_CALLBACK*)(cl_program program, void* user_data) 0xbeb57d40, void* 0)
[VC4CL](      hello.exe): Precompiling source with: 
Dumping program sources to /tmp/vc4cl-source-1365180540.cl
[VC4CL](      hello.exe): Dumping program IR to /tmp/vc4cl-ir-1540383426.ll
[VC4CL](      hello.exe): Precompilation complete with status: 0
[VC4CL](      hello.exe): [VC4CL] base=0x3fc00000, mem=0xb6f85000
[VC4CL](      hello.exe): [VC4CL] V3D base: 0xb6f85000
[VC4CL](      hello.exe): Compiling source with: 
[VC4CL](      hello.exe): Compilation complete with status: 0
Dumping program binaries to /tmp/vc4cl-binary-304089172.bin
[VC4CL](      hello.exe): API call: cl_kernel clCreateKernel(cl_program 0x1918744, const char* "square", cl_int* 0xbeb58600)
[VC4CL](      hello.exe): Tracking live-time of object: 0x1919324 (cl_kernel)
[VC4CL](      hello.exe): API call: cl_mem clCreateBuffer(cl_context 0x195687c, cl_mem_flags 4, size_t 64, void* 0, cl_int* 0)
[VC4CL](      hello.exe): [VC4CL] Mailbox file descriptor opened: 4
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000001 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 80000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 38000000 08000000 00000000
[VC4CL](      hello.exe): Mailbox request: succeeded
[VC4CL](      hello.exe): Tracking live-time of object: 0x194b69c (cl_mem)
[VC4CL](      hello.exe): Mailbox buffer before: 00000024 00000000 0003000c 0000000c 0000000c 00000040 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000024 80000000 0003000c 0000000c 80000004 00000012 00001000 0000000c 00000000
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 0003000d 00000008 00000004 00000012 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 0003000d 00000008 80000004 be6f9000 00000000 00000000
[VC4CL](      hello.exe): [VC4CL] base=0x3e6f9000, mem=0xb6f84000
[VC4CL](      hello.exe): Allocated 64 bytes of buffer: handle 18, device address 0xbe6f9000, host address 0xb6f84000
[VC4CL](      hello.exe): API call: cl_mem clCreateBuffer(cl_context 0x195687c, cl_mem_flags 2, size_t 64, void* 0, cl_int* 0)
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00010006 00000008 00000000 00000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00010006 00000008 80000008 38000000 08000000 00000000
[... cut ...]
[VC4CL](      hello.exe): Releasing live-time of object: 0x195687c (cl_context)
[VC4CL](      hello.exe): Mailbox buffer before: 00000020 00000000 00030012 00000008 00000004 00000000 00000000 00000000
[VC4CL](      hello.exe): Mailbox buffer after: 00000020 80000000 00030012 00000008 80000004 80000000 00000000 00000000
[VC4CL](      hello.exe): [VC4CL] Mailbox file descriptor closed: 4
[VC4CL](      hello.exe): Stopping queue handler thread...
[VC4CL](VC4CL Queue Han): Queue handler thread stopped

```

doe300, 2020-10-26:
It looks like the compiler generates invalid code here which hangs/runs in some infinte loop on the VC4 hardware. I will have to check the generated code to see, what goes wrong there.

sunneo, 2020-10-26:
[https://mega.nz/file/thlSkBrJ#liJUWSo5KGzlLh4XSnzmNRvQslaAhwKCxzg4xTOghIA](generated.tar.gz)

I've upload compressed file to mega, 
it includes generated files (ll, cl, bin, dot) comes from /tmp/vc4*

doe300, 2020-10-27:
I can't see anything wrong with the generated code...

Did you compile the `VC4CL` project with the CMake configuration `-DREGISTER_POKE_KERNELS=ON` or use the debian package emitted by the CI?

Also, do you use the open source VC4 KMS driver or the closed-source OpenGL stack? See also #51 and #60

sunneo, 2020-10-27:
YES, You got me, thanks

I realized that my raspberry pi was configured with raspi-config to use FKMS (Fake-KMS) for Gaming. 
After I switch it to legacy one (without GL), this problem won't occurred.

I'll close this issue

closed 2020-10-27 by sunneo